UpstreamData/pyasic #169: Antminer S19 pro hydro with Vnish Firmware

ppp-205, 2024-07-04:
Hello,

Many thanks for your project.
Could you, please add Antminer S19 pro hydro with Vnish Firmware? right now is reading incorrectly. Can see only 3 boards, incorrect total hashrate reading.
Many thanks)

closed 2024-07-12 by b-rowan

928driver, 2024-07-30:
@ppp-205 are you now able to read values from all 4 hash boards? Also, do you read water temperature and additional temp sensors in Vnish? I am using Braiins OS, but there is no water temperature visible there which I could see in the original Antminer OS. In addition, there seems to be a few more temp sensors on each board that I do also not see in BOS. Therefore I am curious about Vnish :)

b-rowan, 2024-07-30:
> @ppp-205 are you now able to read values from all 4 hash boards? Also, do you read water temperature and additional temp sensors in Vnish? I am using Braiins OS, but there is no water temperature [...]

Keep in mind there is no support for Vnish power limits until they move from discrete setpoints to contiguous.  I have been asking them about this for a while, but since they cater to mainly larger clients it could be a while before this is implemented.

PR #173 (2024-07-12): feature: add support for Vnish S19 Pro Hydro.
Adds support for S19 Pro Hydro running vnish.  Also some minor bug fixes.  Closes #169.