UpstreamData/pyasic #170: 0.58.0 possible Bug (@dataclass)

pixeldoc2000, 2024-07-05:
**Describe the bug**
Python Error while testing pyasic 0.58.0
```
(.venv) D:\git\pyasic2>python test_bitaxe_01.py
Traceback (most recent call last):
  File "D:\git\pyasic2\test_bitaxe_01.py", line 2, in <module>
    from pyasic import get_miner
  File "D:\git\pyasic2\.venv\Lib\site-packages\pyasic\__init__.py", line 18, in <module>
    from pyasic.data import MinerData
  File "D:\git\pyasic2\.venv\Lib\site-packages\pyasic\data\__init__.py", line 32, in <module>
    from pyasic.data.pools import PoolMetrics
  File "D:\git\pyasic2\.venv\Lib\site-packages\pyasic\data\pools.py", line 35, in <module>
    @dataclass
     ^^^^^^^^^
  File "C:\Users\xx\AppData\Local\Programs\Python\Python311\Lib\dataclasses.py", line 1232, in dataclass
    return wrap(cls)
           ^^^^^^^^^
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python311\Lib\dataclasses.py", line 1222, in wrap
    return _process_class(cls, init, repr, eq, order, unsafe_hash,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python311\Lib\dataclasses.py", line 1027, in _process_class
    _init_fn(all_init_fields,
  File "C:\Users\xxx\AppData\Local\Programs\Python\Python311\Lib\dataclasses.py", line 545, in _init_fn
    raise TypeError(f'non-default argument {f.name!r} '
TypeError: non-default argument 'url' follows default argument
```

```
(.venv) D:\git\pyasic2>python -V
Python 3.11.9

(.venv) D:\git\pyasic2>pip list
Package           Version
----------------- -----------
aiofiles          23.2.1
anyio             4.4.0
asyncssh          2.15.0
betterproto       2.0.0b6
certifi           2024.7.4
cffi              1.16.0
cryptography      42.0.8
grpclib           0.4.7
h11               0.14.0
h2                4.1.0
hpack             4.0.0
httpcore          1.0.5
httpx             0.27.0
hyperframe        6.0.1
idna              3.7
multidict         6.0.5
passlib           1.7.4
pip               24.0
pyaml             23.12.0
pyasic            0.58.0
pycparser         2.22
python-dateutil   2.9.0.post0
PyYAML            6.0.1
setuptools        65.5.0
six               1.16.0
sniffio           1.3.1
toml              0.10.2
typing_extensions 4.12.2
```

**To Reproduce**
Test source from pyasic doc with pyasic 0.58.0
```
import asyncio
from pyasic import get_miner

async def gather_miner_data():
    miner = await get_miner("192.168.0.202")
    if miner is not None:
        miner_data = await miner.get_data()
        print(miner_data)  # all data from the dataclass
        print(miner_data.hashrate)  # hashrate of the miner in TH/s

if __name__ == "__main__":
    asyncio.run(gather_miner_data())
```

Looks like this was broken?: https://github.com/UpstreamData/pyasic/blob/f405bbff4d37a4a6cf96e515f4bfb0f3fce63034/pyasic/data/pools.py#L35-L55

No error with pyasic 0.57.6

b-rowan, 2024-07-05:
@Ytemiloluwa

b-rowan, 2024-07-05:
Fixed in 0.58.1

closed 2024-07-05 by b-rowan

pixeldoc2000, 2024-07-05:
I can confirm, the bug is fixed in 0.58.1, thx @b-rowan