UpstreamData/pyasic #174: Two miners Innosilicon T3+ and Bluestar L1

s5aero, 2024-07-18:
Is there any chance this two miners Innosilicon T3+ and Bluestar L1 can be added to pyasic? i would love to get temperatures and controling performance via home assistant. 

the Bluestar L1 has very similar UI as Antminers (clone version) if this helps.

I tried connecting it and get cant connect to miner message.

s5aero, 2024-07-18:
This is primary reason i would like to add it to home assistant. they both have options for performance change in their GUI. Innosilicon T3+ has 17 predefined performance options from efficiency to performance and Bluestar L1 has 4 predefined options and one manual voltage/frequency option. i use predefined options daily (only two options actualy ether efficient or performance). home assistant only turns them on and off when there is surplus power available. i am considering also to get sort of script that will login via web interface and do steps i do to change performance state.

s5aero, 2024-07-26:
does anyone know how to get command tracing in SSH, i connected to miner via SSH and would like to trace what command it gets when i change performance option, so then i could make script to send this command via ssh for automation.

b-rowan, 2024-07-26:
More than likely it does it with a web API endpoint.  If you go to the web interface in a browser, CTRL+SHIFT+I for devtools, then go to the Network tab, you should be able to record what it does.

s5aero, 2024-07-28:
i went to devtools under network and in bluestar L1 i found out there are two files names that are beeing sent when i click save while changing performance settings to low power. those are set_autotune_conf.cgi and set_miner_conf.cgi
here are payloads of set_miner_conf.cgi when i set it to low power mode with fan speed set to 10%
`_bb_pool1url: 
stratum+tcp://<URL>
_bb_pool1user: 
<minername>
_bb_pool1pw: 
<password>
_bb_pool2url: 
stratum+tcp://<URL>
_bb_pool2user: 
<minername>
_bb_pool2pw: 
<password>
_bb_pool3url: 
stratum+tcp://<URL>
_bb_pool3user: 
<minername>
_bb_pool3pw: 
<password>
_bb_nobeeper: 
false
_bb_notempoverctrl: 
false
_bb_fan_customize_switch: 
true
_bb_fan_customize_value: 
10
_bb_freq: 
800
_bb_coin_type: 
ltc
_bb_voltage_customize_switch: 
false
_bb_voltage_customize_value: 
1200
_bb_ema: 
128
_bb_low_power_switch: 
true`
and there is only one file name showing when i change to Chip adaptation mode this is set_miner_conf.cgi here is payload when i switch it to chip adaptation mode (this is optimized power/performance mode) with fan speed to auto
`_bb_pool1url: 
stratum+tcp://<URL>
_bb_pool1user: 
<minername>
_bb_pool1pw: 
<password>
_bb_pool2url: 
stratum+tcp://<URL>
_bb_pool2user: 
<minername>
_bb_pool2pw: 
<password>
_bb_pool3url: 
stratum+tcp://<URL>
_bb_pool3user: 
<minername>
_bb_pool3pw: 
<password>
_bb_nobeeper: 
false
_bb_notempoverctrl: 
false
_bb_fan_customize_switch: 
false
_bb_fan_customize_value: 
_bb_freq: 
340
_bb_coin_type: 
ltc
_bb_voltage_customize_switch: 
false
_bb_voltage_customize_value: 
940
_bb_ema: 
128
_bb_low_power_switch: 
false`

it is strange only because it looks like freq and voltage settings are opposite as it is reading what was previusly there.
when the miner is normaly running and i just refresh page i can read minerConfiguration.cgi and the only difference i found in this file while id different power modes are:
when in Low power mode:
`                "fan-ctrl": true,
                "fan-pwm": "10",
                "use-vil": true,
                "freq": "340",
                "sram-voltage": "128",
                "coin-type": "ltc"
            }];
            bb_data_voltage = 940;
            bb_data_voltage_check = 0;
            bb_data_low_power_check = 1;
            bb_autotune_check = 0;`
and when in Chip adaptation mode:
`								
                "use-vil": true,
                "freq": "800",
                "sram-voltage": "128",
                "coin-type": "ltc"
            }];
            bb_data_voltage = 1200;
            bb_data_voltage_check = 0;
            bb_data_low_power_check = 0;
            bb_autotune_check = 2;`

do you think it is possible to do do it via API , i never thinkered with api before, so i have no clue if it can be writen or can only be read.

b-rowan, 2024-07-28:
It seems like they use that _bb_low_power_switch, but this looks extremely similar to what the pre-X19 antminers do.  I'll see what I can do.