UpstreamData/pyasic #177: ssl.SSLError: [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:1000)

smalldaemon, 2024-07-28:
Hello, 

I always get the same error when trying to get information from the miner

File "/home/warp/.local/lib/python3.12/site-packages/anyio/streams/tls.py", line 140, in _call_sslobject_method
    result = func(*args)
    .[36m         │     └ ().[m
    .[36m         └ <bound method SSLObject.do_handshake of <ssl.SSLObject object at 0xffffa9068860>>.[m
  File "/usr/local/lib/python3.12/ssl.py", line 917, in do_handshake
    self._sslobj.do_handshake()
    .[36m└ <ssl.SSLObject object at 0xffffa9068860>.[m
ssl.SSLError: [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:1000)

Full log in the attachment
[log.txt](https://github.com/user-attachments/files/16402490/log.txt)

Using python 3.12 and the latest version of pyasic

The code used is shown below

`netIpv4Address = ipaddress.ip_network('10.1.1.2')

    for i in netIpv4Address:
        print(f"Check IP: [{i}]\n")
        miner = await get_miner(i)

        if miner is not None:
            miner_data = await miner.get_data()
            print(f"Found miner on IP: [{i}], MINER: [{miner_data}\n")
`

I can see from the logs that the https protocol is being accessed, which causes an error

Can you help solve this problem?
Thanks

b-rowan, 2024-07-28:
Are you using a proxy or such?  Seems like most of the issues surrounding this are from people using proxies which do not support https, and are therefore responding to those requests incorrectly.

smalldaemon, 2024-07-28:
No, I don't use proxies

closed 2024-07-28 by smalldaemon