UpstreamData/pyasic #180: Whatsminer requests via 8889 port

VladTheJunior, 2024-07-30:
Hi, I found out this great repo when searched some info about Whatsminer 8889 port. I found out on sources that Upstream Data developed web api for unlocking Whatsminer privileged API: https://wmt.pyasic.org/docs which exactly uses this port. Does it have open source code or can you share details of how stage1 and stage 2 output is generated?  

My goal is not to unlock WM privileged api, but to simulate WM Tool scan request via 8889 port, which gives a bit more data and with only 1 request, than standard api.

b-rowan, 2024-07-30:
I definitely am not able to release anything on that.  I can add more to WMT.pyasic.org to allow you to generate requests (and the requests should be static for any given IP), but I believe they still arrive encrypted, so that may not be useful.

VladTheJunior, 2024-07-30:
> but I believe they still arrive encrypted, so that may not be useful.

Scan response data is not encrypted if you meant this, I checked it with Wireshark

> I definitely am not able to release anything on that.

:( I spent so much time, tried to find some common with this first responses to device. But okay, I will do more deap research since your team was able to do this

b-rowan, 2024-07-30:
Absolutely, and best of luck to you.  I would love to help you, but I'm sure you understand why I can publicize this.

closed 2024-07-30 by b-rowan