UpstreamData/pyasic #182: Missing temp sensor readings for Antminer S19 Pro+ Hyd 191T

928driver, 2024-07-30:
**Is your feature request related to a problem? Please describe.**
With the original Antminer OS, additional temperature sensors are visible on the dashboard which seems to be water inn, water out(?) and two sensors for chips on each board. But is not very clear what the individual sensors are though. After I installed Braiins OS, these additional sensors are no longer visible. I have sent a ticket to Braiins a few days ago and waiting for answers about this. It might be that these temperature sensors are logged in BOS but just not visible in the dashboard. Therefore I was hoping it would be possible to read these values in HA.

**Describe the solution you'd like**
It would be very useful to have all temperature sensors available through pyasic and HASS-Miner in HA. Especially the water in/out temp sensor is useful to both monitor cooling performance and make automations to control pumps, valves, cooling fans etc.

**Describe alternatives you've considered**
I am curious i Vnish OS have better support for Antminer hydro versions to be able to monitor all temperature sensors. But first I would like to give BOS a go as I am runnin BOS on air cooled S19 without issues.

**Additional context**
[image]

928driver, 2024-07-30:
I now received feedback from Braiins support team that the additional temperature sensors for S19 Pro+ Hydro is not displayed in Braiins OS at this point of time. He said that he will put it in as a feature request. I asked a follow up question if the temperatures are monitored in the background of BOS even though they are not shown on the dashboard, I am now awaiting reply.

b-rowan, 2024-07-30:
> I now received feedback from Braiins support team that the additional temperature sensors for S19 Pro+ Hydro is not displayed in Braiins OS at this point of time. He said that he will put it in as a [...]

Yeah makes sense.  There is full support in the next version I release for this miner (except things that arent in the API, which it seems like this isnt), but I just have to wait on some testing for the pool configuration and then ill push the version and update HASS-miner.

b-rowan, 2024-08-01:
> @b-rowan , I have a question partly related to this thread: Would it be possible to make HA monitor chip and board temperatures from BOS even though the miner is in idle mode? They now become "unknown" status when idling even though they are visible in the BOS dashboard. 
> 
> 
> 
> The reason why I am asking is that my hydro miner is blocking from starting to mine if the chip temp is below 20deg C. If I could monitor the temp in HA during idling, I could make automation to trigger heating, adjust fan, pumps etc to reach above 20deg to be able to start mining.

The problem here is that to get chip temp, something has to communicate with the chips, and the process that communicates with the chips is bosminer.  So if bosminer isn't running, nothing is communicating with the chips, meaning no temp info.

928driver, 2024-08-02:
@b-rowan , I see the point that temperatures are not monitored when the logic is in standby. But when I open the Braiins dashboard in my browser, I can see temperatures there when the BOS miner is stopped, what is monitoring the temperature then?

Totally unrelated, but the miner I was checking failed on me so I have limited possibility to test this further at the moment.