UpstreamData/pyasic #198: [Feature Request] Add Avalon Nano 3 Miner Support

hanleyp, 2024-09-01:
The new Avalon Nano 3 is running AvalonMiner software (Firmware Version24071801_42c628d). It is sometimes detected as "AvalonMiner (Stock)" and pyasic works. Other times, it is detected as "Unknown (VNish)" and get_data fails.

b-rowan, 2024-09-01:
> Other times, it is detected as "Unknown (VNish)" and get_data fails.

This should be fixed with 0.59.7.

> It is sometimes detected as "AvalonMiner (Stock)" and pyasic works

I would like to add actual support for it, so if you're up for it, could you run this script?

To use this, make a new text file, paste this data in, and change the "server" variable at the top to match the IP of your miner.  Once that's done, change the file extension of the file to .ps1 (if you cant see this, its in file explore top menu, view, then a checkbox that says file name extensions, then you can rename the file).  Right click, select "Run with powershell", then send the result in a text file here.

```powershell
$server = "192.168.1.5"
$port = 4028
$commands = @("devs", "stats", "devdetails", "version")

try {
    foreach ($cmd in $commands) {
        $client = New-Object System.Net.Sockets.TcpClient
        $client.Connect($server, $port)

        $stream = $client.GetStream()

        $command = '{"command": "' + $cmd + '"}'
        $writer = [System.Text.Encoding]::UTF8.GetBytes($command)

        $stream.Write($writer, 0, $writer.Length)
        $stream.Flush()

        # Read response
        $reader = New-Object System.IO.StreamReader($stream)
        $response = $reader.ReadToEnd()
        $reader.Close()

        # Display the response
        Write-Host "Response for command '$cmd':"
        Write-Host "-------------------------------------------------------------------------------------"
        Write-Host $response
        Write-Host ""

        # Clear the stream for the next command
        $stream.Flush()
        # Close the connection
        $client.Close()
    }

    # Wait for a key press
    Read-Host "Press Enter to exit..."
}
catch {
    Write-Host "An error occurred: $_"
    Read-Host "Press Enter to exit..."
}
```

hanleyp, 2024-09-01:
I don't have any windows boxes (but can borrow a laptop if the below doesn't help).

Here is the output of "socat":
phanley@tele ~ % echo -n "version" | socat -t 30 stdio tcp:192.168.2.131:4028,shut-none && echo
STATUS=S,When=1725213740,Code=22,Msg=CGMiner versions,Description=cgminer 4.11.1|VERSION,CGMiner=4.11.1,API=3.7,PROD=Avalonnano,MODEL=nano3,HWTYPE=PMMv1_X1,SWTYPE=MM318_X2,VERSION=24071801_42c628d,HVERSION=24071801_b906c52_6223725,UPAPI=2|
phanley@tele ~ % echo -n "devdetails" | socat -t 30 stdio tcp:192.168.2.131:4028,shut-none && echo
STATUS=S,When=1725213762,Code=69,Msg=Device Details,Description=cgminer 4.11.1|DEVDETAILS=0,Name=AVANANO,ID=0,Driver=avalonnano,Kernel=,Model=,Device Path=|
phanley@tele ~ % echo -n "stats" | socat -t 30 stdio tcp:192.168.2.131:4028,shut-none && echo
STATUS=S,When=1725213774,Code=70,Msg=CGMiner stats,Description=cgminer 4.11.1|STATS=0,ID=AVANANO0,Elapsed=334151,Calls=0,Wait=0.000000,Max=0.000000,Min=99999999.000000,MM ID0=HashStatus[1] Ver[nano3-24071801_42c628d] HVer[24071801_b906c52_6223725]  DNA[0000000000000000]  HDNA[020100009358c17d]  PFCnt[2] NETFAIL[1724892531 1724892546 1725120155 1725120156 0 0 0 0] Elapsed[334167] LW[8342299] MH[289] DHW[0] HW[289] DH[1.735%] Temp[71] OTemp[45] TMax[92] TAvg[90] TarT[90] Fan1[4830] FanR[60%] Vo[0] PS[0 0 0 4 2755 123 340] GHSspd[4125.40] DHspd[1.735%] GHSmm[4196.48] GHSavg[3974.32] WU[55520.58] Freq[0.00] Led[1] MGHS[3974.32] MTmax[92] MTavg[90] TA[10] Core[A3198S] PING[290] SoftOFF[0] ECHU[0] ECMM[0] PLL0[0 306 752 712] SF0[100 568 588 608] PVT_T0[ 89  91  92  91  87  86  91  91  92  87] PVT_V0[330 325 326 322 326 325 326 324 324 325] MW0[7689 7585 7494 7724 7609 7578 7659 7394 7325 7436] CRC[0] ATA0[60-80-318-100:293:313:333] ATA1[90-85-322-100:459:479:499] ATA2[125-90-344-100:568:588:608] WORKLEVEL[2] MPO[125] ADJ[1],MM Count=1,Nonce Mask=1|STATS=1,ID=POOL0,Elapsed=334151,Calls=0,Wait=0.000000,Max=0.000000,Min=99999999.000000,Pool Calls=0,Pool Attempts=0,Pool Wait=0.000000,Pool Max=0.000000,Pool Min=99999999.000000,Pool Av=0.000000,Work Had Roll Time=false,Work Can Roll=false,Work Had Expire=false,Work Roll Time=0,Work Diff=16384.00000000,Min Diff=16384.00000000,Max Diff=16384.00000000,Min Diff Count=544,Max Diff Count=544,Times Sent=18965,Bytes Sent=3686291,Times Recv=29230,Bytes Recv=24320077,Net Bytes Sent=3686291,Net Bytes Recv=24320077|
phanley@tele ~ % echo -n "devs" | socat -t 30 stdio tcp:192.168.2.131:4028,shut-none && echo
STATUS=S,When=1725213784,Code=9,Msg=1 ASC(s),Description=cgminer 4.11.1|ASC=0,Name=AVANANO,ID=0,Enabled=Y,Status=Alive,Temperature=0.00,MHS av=3963345.31,MHS 5s=8306339.27,MHS 1m=6859427.21,MHS 5m=5142265.64,MHS 15m=4579396.40,Accepted=18889,Rejected=58,Hardware Errors=289,Utility=3.39,Last Share Pool=0,Last Share Time=1725213760,Total MH=1324058287104.0000,Diff1 Work=309231616,Difficulty Accepted=309477376.00000000,Difficulty Rejected=950272.00000000,Last Share Difficulty=16384.00000000,Last Valid Work=1725213781,Device Hardware%=0.0001,Device Rejected%=0.3073,Device Elapsed=334076|
phanley@tele ~ %

b-rowan, 2024-09-01:
Ah perfect.  I usually expect people to just have windows devices at this point, nice to see you actually understood what that did haha.  Ill take a look later today, I think this is standard return stuff for the avalons though, so just need to create the device type and set chip count and model and it should be good.

b-rowan, 2024-09-01:
Done in 0.60.0.

closed 2024-09-01 by b-rowan