UpstreamData/pyasic #199: Avalon Nano 3 hash rate reported by pyasic much lower than reported by device itself

hanleyp, 2024-09-01:
**Describe the bug**
Realtime Hashrate on web interface of Avalon Nano 3:
4.12 THS

Miner Log:
{'STATUS':[{'STATUS':'S','When':1725228460,'Code':11,'Msg':'Summary','Description':'cgminer 4.11.1'}],'SUMMARY':[{'Elapsed':348845,'MHS av':3961115.58,'MHS 5s':2759123.01,'MHS 1m':2410313.07,'MHS 5m':3430034.79,'MHS 15m':3853993.49,'Found Blocks':0,'Getworks':10720,'Accepted':19703,'Rejected':60,'Hardware Errors':301,'Utility':3.39,'Discarded':0,'Stale':9,'Get Failures':2,'Local Work':159346,'Remote Failures':0,'Network Blocks':567,'Total MH':1381813433742.0000,'Work Utility':55505.22,'Difficulty Accepted':322813952.00000000,'Difficulty Rejected':983040.00000000,'Difficulty Stale':147456.00000000,'Best Share':352799466,'Device Hardware%':0.0001,'Device Rejected%':0.3046,'Pool Rejected%':0.3035,'Pool Stale%':0.0455,'Last getwork':0}
{'STATUS':[{'STATUS':'S','When':1725228460,'Code':70,'Msg':'CGMiner stats','Description':'cgminer 4.11.1'}],'STATS':[{'STATS':0,'ID':'AVANANO0','Elapsed':348845,'Calls':0,'Wait':0.000000,'Max':0.000000,'Min':99999999.000000,'MM ID0':'HashStatus[1] Ver[nano3-24071801_42c628d] HVer[24071801_b906c52_6223725] DNA[0000000000000000] HDNA[020100009358c17d] PFCnt[2] NETFAIL[1724892531 1724892546 1725120155 1725120156 0 0 0 0] Elapsed[348857] LW[8709099] MH[301] DHW[0] HW[301] DH[1.727%] Temp[69] OTemp[45] TMax[93] TAvg[90] TarT[90] Fan1[4470] FanR[58%] Vo[0] PS[0 0 0 4 2755 123 340] GHSspd[4123.47] DHspd[1.727%] GHSmm[4200.72] GHSavg[3973.08] WU[55503.28] Freq[0.00] Led[1] MGHS[3973.08] MTmax[93] MTavg[90] TA[10] Core[A3198S] PING[116] SoftOFF[0] ECHU[0] ECMM[0] PLL0[0 278 765 727] SF0[100 568 588 608] PVT_T0[ 89 91 93 91 87 87 91 91 92 87] PVT_V0[330 323 324 323 325 324 328 323 322 328] MW0[8052 7912 7825 8051 7923 7937 8012 7677 7647 7751] CRC[0] ATA0[60-80-318-100:293:313:333] ATA1[90-85-322-100:459:479:499] ATA2[125-90-344-100:568:588:608] WORKLEVEL[2] MPO[125] ADJ[1]','MM Count':1,'Nonce Mask':1}]}

phanley@tele pyasic % echo -n "stats" | socat -t 30 stdio tcp:192.168.2.131:4028,shut-none && echo
STATUS=S,When=1725228583,Code=70,Msg=CGMiner stats,Description=cgminer 4.11.1|STATS=0,ID=AVANANO0,Elapsed=348963,Calls=0,Wait=0.000000,Max=0.000000,Min=99999999.000000,MM ID0=HashStatus[1] Ver[nano3-24071801_42c628d] HVer[24071801_b906c52_6223725]  DNA[0000000000000000]  HDNA[020100009358c17d]  PFCnt[2] NETFAIL[1724892531 1724892546 1725120155 1725120156 0 0 0 0] Elapsed[348976] LW[8712083] MH[301] DHW[0] HW[301] DH[1.728%] Temp[69] OTemp[45] TMax[92] TAvg[90] TarT[90] Fan1[4740] FanR[58%] Vo[0] PS[0 0 0 4 2756 123 340] GHSspd[4125.99] DHspd[1.728%] GHSmm[4200.80] GHSavg[3973.24] WU[55505.47] Freq[0.00] Led[1] MGHS[3973.24] MTmax[92] MTavg[90] TA[10] Core[A3198S] PING[73] SoftOFF[0] ECHU[0] ECMM[0] PLL0[0 268 778 724] SF0[100 568 588 608] PVT_T0[ 89  91  92  91  88  87  91  91  92  87] PVT_V0[325 323 323 326 325 323 325 318 325 324] MW0[8054 7912 7834 8053 7925 7940 8017 7678 7651 7753] CRC[0] ATA0[60-80-318-100:293:313:333] ATA1[90-85-322-100:459:479:499] ATA2[125-90-344-100:568:588:608] WORKLEVEL[2] MPO[125] ADJ[1],MM Count=1,Nonce Mask=1|STATS=1,ID=POOL0,Elapsed=348963,Calls=0,Wait=0.000000,Max=0.000000,Min=99999999.000000,Pool Calls=0,Pool Attempts=0,Pool Wait=0.000000,Pool Max=0.000000,Pool Min=99999999.000000,Pool Av=0.000000,Work Had Roll Time=false,Work Can Roll=false,Work Had Expire=false,Work Roll Time=0,Work Diff=16384.00000000,Min Diff=16384.00000000,Max Diff=16384.00000000,Min Diff Count=568,Max Diff Count=568,Times Sent=19786,Bytes Sent=3846386,Times Recv=30507,Bytes Recv=25397640,Net Bytes Sent=3846386,Net Bytes Recv=25397640|

pyasic Miner response from get_data():
MinerData(ip='192.168.2.131', datetime='2024-09-01T17:07:40.276923-05:00', timestamp=1725228460, device_info=DeviceInfo(make=<MinerMake.AVALONMINER: 'AvalonMiner'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AvalonMiner', model=None, firmware='Stock', algo='SHA256', mac=None, api_ver='3.7', fw_ver='4.11.1', hostname=None, hashrate=0.019 TH/s, expected_hashrate=None, expected_hashboards=3, expected_chips=0, expected_fans=2, percent_expected_chips=0, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=4, env_temp=69.0, wattage=None, wattage_limit=125, voltage=None, fans=[Fan(speed=4560), Fan(speed=None)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=0.003 TH/s, temp=9, chip_temp=9, chips=10, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=0.009 TH/s, temp=0, chip_temp=2, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=0.007 TH/s, temp=None, chip_temp=None, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None)], total_chips=10, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://mine.ocean.xyz:3334', user='bc1pl2lxscrq8kamtwgescns6dy8hcmjsewju44mr9lzj7ctkahjclnqrwcs8r', password='x')], quota=1, name='F5SHLO')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=True, errors=[], is_mining=True, uptime=348838, efficiency=None, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='mine.ocean.xyz', port=3334, pubkey=None), accepted=19702, rejected=60, get_failures=2, remote_failures=0, active=True, alive=True, index=0, user='bc1pl2lxscrq8kamtwgescns6dy8hcmjsewju44mr9lzj7ctkahjclnqrwcs8r', pool_rejected_percent=0.3036129946361704, pool_stale_percent=0.010120433154539015)])

hashrate=0.019 TH/s with pyasic, but should be closer to 4 TH/s

hanleyp, 2024-09-01:
v0.60.0

b-rowan, 2024-09-01:
> v0.60.0

Interesting, its still not identifying it?

b-rowan, 2024-09-01:
There is a bug with the stats parser thats causing this, because they did their stats in the dumbest possible way...

b-rowan, 2024-09-01:
Parsing bug fixed in 0.60.1.  Still not sure why its not identifying...

b-rowan, 2024-09-01:
If you run `pyasic.miner_factory.get_miner_model_avalonminer(ip)` on that miner does it return `AVALONMINER NANO3`?

hanleyp, 2024-09-01:
AVALON0O

hanleyp, 2024-09-01:
Hashrate looks good now for Nano 3's with 0.60.1.

b-rowan, 2024-09-01:
> pyasic.miner_factory.get_miner_model_avalonminer(ip) = "AVALON0O"

Hmm...  It seems like RPC version and log version don't agree with each other...

Can you try `await miner.rpc.version()` and see what that gets?

hanleyp, 2024-09-01:
print(await miner.rpc.version()):
{'STATUS': [{'STATUS': 'S', 'When': 1725230592, 'Code': 22, 'Msg': 'CGMiner versions', 'Description': 'cgminer 4.11.1'}], 'VERSION': [{'CGMiner': '4.11.1', 'API': '3.7', 'PROD': 'Avalon0o', 'MODEL': '0o3', 'HWTYPE': 'PMMv1_X1', 'SWTYPE': 'MM318_X2', 'VERSION': '24071801_42c628d', 'HVERSION': '24071801_b906c52_6223725', 'UPAPI': '2'}], 'id': 1}

b-rowan, 2024-09-01:
Ah.  I see what the issue is.  Ill fix it, one sec.

hanleyp, 2024-09-01:
I also think you left a print statement somewhere when I call miner.get_data() for the Avalon.

b-rowan, 2024-09-01:
Should be fixed with 0.60.2 now, so many parsing issues.

> I also think you left a print statement somewhere when I call miner.get_data() for the Avalon. (Unless that was me.)

Yeah... Should really introduce linting for that...

b-rowan, 2024-09-01:
Fixed print issue with 0.60.3.

hanleyp, 2024-09-01:
All good now. Now I can see how much money I'm losing each day in realtime.

closed 2024-09-01 by b-rowan