UpstreamData/pyasic #209: Antminer KA3 model not detected

Combinator78, 2024-10-29:
MinerData(ip='172.16.57.55', datetime='2024-10-29T14:22:17.258884+03:00', timestamp=1730200937, device_info=**DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>**, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', **model=None**, firmware='Stock', algo='SHA256', mac=None, api_ver='3.1', fw_ver='Mon Feb 20 12:43:39 CST 2023', hostname=None, hashrate=167.8539782881 TH/s, expected_hashrate=168.16128 TH/s, expected_hashboards=3, expected_chips=0, expected_fans=2, percent_expected_chips=0, percent_expected_hashrate=100, percent_expected_wattage=None, temperature_avg=69, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=2400), Fan(speed=2520)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=55.9229362504 TH/s, temp=70, chip_temp=60, chips=92, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=55.9608485642 TH/s, temp=69, chip_temp=59, chips=92, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=55.970193473500004 TH/s, temp=68, chip_temp=58, chips=92, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=276, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://kda.antpool.com:8026', user='AlexeyAlexey.A028', password='x'), Pool(url='', user='', password='x'), Pool(url='', user='', password='x')], quota=1, name='N5JQK9')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=2182946, efficiency=None, pools=[])

b-rowan, 2024-10-29:
Fixed, fixing your other issues before I put out a version.

Combinator78, 2024-10-30:
Yes, it works. Thank you.

PS D:\pyasic-0.61.2> py test.py
MinerData(ip='172.16.57.55', datetime='2024-10-30T06:30:40.016501+03:00', timestamp=1730259040, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.KA3: 'KA3'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', model='KA3', firmware='Stock', algo='SHA256', mac='26:EA:2F:F5:69:0E', api_ver='3.1', fw_ver='Mon Feb 20 12:43:39 CST 2023', hostname='Antminer', hashrate=167.74074000000002 TH/s, expected_hashrate=168.16128 TH/s, expected_hashboards=3, expected_chips=276, expected_fans=2, percent_expected_chips=100, percent_expected_hashrate=100, percent_expected_wattage=None, temperature_avg=51, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=2520), Fan(speed=2640)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=55.594010000000004 TH/s, temp=52.0, chip_temp=62.0, chips=92, expected_chips=92, serial_number='JYZZYPFBCJHAB00Z2', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=56.01426 TH/s, temp=51.5, chip_temp=61.5, chips=92, expected_chips=92, serial_number='JYZZYPFBCJHAB00XG', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=56.13247 TH/s, temp=50.5, chip_temp=60.5, chips=92, expected_chips=92, serial_number='JYZZYPFBCJHAB00Z7', missing=False, tuned=None, active=None, voltage=None)], total_chips=276, nominal=True, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://kda.antpool.com:8026', user='AlexeyAlexey.A028', password='123'), Pool(url='', user='', password=''), Pool(url='', user='', password='')], quota=1, name='EJ2YS6')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=False, errors=[], is_mining=True, uptime=2241048, efficiency=None, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='kda.antpool.com', port=8026, pubkey=None), accepted=459518, rejected=6787, get_failures=12, remote_failures=4, active=True, alive=True, index=0, user='AlexeyAlexey.A028', pool_rejected_percent=1.4554851438436216, pool_stale_percent=0.0025734229742336023), PoolMetrics(url=None, accepted=0, rejected=0, get_failures=0, remote_failures=0, active=True, alive=False, index=1, user='', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=None, accepted=0, rejected=0, get_failures=0, remote_failures=0, active=True, alive=False, index=2, user='', pool_rejected_percent=0, pool_stale_percent=0)])
167.74074000000002 TH/s

closed 2024-10-30 by b-rowan