UpstreamData/pyasic #210: Antminer L3+ does not return data

Combinator78, 2024-10-29:
Starting with version 57.0 of the pyasic library, all data is not returned on the device Antminer L3+

MinerData(ip='172.16.57.41', datetime='2024-10-29T14:33:36.901186+03:00', timestamp=1730201616, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.L3Plus: 'L3+'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', model='L3+', firmware='Stock', algo='SHA256', mac='2C:6B:7D:AD:C6:56', api_ver='3.1', fw_ver='4.9.0 rwglr', **hostname=None, hashrate=None, expected_hashrate=None**, expected_hashboards=3, expected_chips=216, expected_fans=2, **percent_expected_chips=None, percent_expected_hashrate=None**, percent_expected_wattage=None, temperature_avg=None, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=0), Fan(speed=0)], fan_psu=None, hashboards=[], total_chips=None, nominal=None, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='ltc.trustpool.ru:3333', user='first1vadim.003', password='123'), Pool(url='ltc.trustpool.ru:25', user='first1vadim.003', password='123'), Pool(url='ltc.trustpool.ru:443', user='first1vadim.003', password='123')], quota=1, name='PJUJK8')]), fan_mode=FanModeNormal(mode='normal', 
minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=609702, efficiency=None, pools=[])

Although on the library version 56.0 the data was returned correctly:

MinerData(ip='172.16.57.41', datetime=datetime.datetime(2024, 10, 29, 14, 37, 34, 844438, tzinfo=datetime.timezone(datetime.timedelta(seconds=10800), 'RTZ 2 (зима)')), uptime=609940, mac='2C:6B:7D:AD:C6:56', model='L3+', make='AntMiner', api_ver='3.1', fw_ver='4.9.0 rwglr', hostname=None, hashrate=0.39, expected_hashrate=None, hashboards=[HashBoard(slot=0, hashrate=0.13, temp=48, chip_temp=41, chips=72, expected_chips=72, serial_number=None, missing=False), HashBoard(slot=1, hashrate=0.13, temp=45, chip_temp=37, chips=72, expected_chips=72, serial_number=None, missing=False), HashBoard(slot=2, hashrate=0.13, temp=44, chip_temp=36, chips=72, expected_chips=72, serial_number=None, missing=False)], expected_hashboards=3, temperature_avg=46, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=0), Fan(speed=0)], fan_psu=None, total_chips=216, expected_chips=216, percent_expected_chips=100, percent_expected_hashrate=None, percent_expected_wattage=None, nominal=True, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='ltc.trustpool.ru:3333', user='first1vadim.003', password='123'), Pool(url='ltc.trustpool.ru:25', user='first1vadim.003', password='123'), Pool(url='ltc.trustpool.ru:443', user='first1vadim.003', password='123')], quota=1, name='OEITNI')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal'), power_scaling=PowerScalingDisabled(mode='disabled')), errors=[], fault_light=None, efficiency=None, is_mining=True)
0.39
PS D:\pyasic-0.56.0>

b-rowan, 2024-10-29:
I can't find any reason these would be broken in the changes from then to now...  Could you try manually calling some of those functions (`await miner.get_hashrate()` etc)?

Combinator78, 2024-10-29:
Other devices L3+ are scanned normally.

b-rowan, 2024-10-29:
Fixed in latest commit, will try to fix your other issues before I push out a version though.

Combinator78, 2024-10-29:
Thanks a lot

Combinator78, 2024-10-29:
It doesn't return an error now, but the data is incomplete.
PS D:\pyasic-0.61.1> py test.py
MinerData(ip='172.16.57.41', datetime='2024-10-29T20:26:17.382110+03:00', timestamp=1730222777, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.L3Plus: 'L3+'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', model='L3+', firmware='Stock', algo='SHA256', mac='2C:6B:7D:AD:C6:56', api_ver='3.1', fw_ver='4.9.0 rwglr', hostname=None, hashrate=None, expected_hashrate=None, expected_hashboards=3, expected_chips=216, expected_fans=2, percent_expected_chips=None, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=None, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=0), Fan(speed=0)], fan_psu=None, hashboards=[], total_chips=None, nominal=None, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='ltc.trustpool.ru:3333', user='first1vadim.003', password='123'), Pool(url='ltc.trustpool.ru:25', user='first1vadim.003', password='123'), Pool(url='ltc.trustpool.ru:443', user='first1vadim.003', password='123')], quota=1, name='M5C2UG')]), fan_mode=FanModeNormal(mode='normal', 
minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=630863, efficiency=None, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=3333, pubkey=None), accepted=96101, rejected=204, get_failures=6, remote_failures=1, active=True, alive=True, index=0, user='first1vadim.003', pool_rejected_percent=0.2118270079435128, pool_stale_percent=0.006230206115985671), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=25, pubkey=None), accepted=85, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=1, user='first1vadim.003', pool_rejected_percent=0.0, pool_stale_percent=0.0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=443, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=2, user='first1vadim.003', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host=None, port=None, pubkey=None), accepted=10010, rejected=461, get_failures=261, remote_failures=2, active=True, alive=True, index=3, user='*', pool_rejected_percent=4.402635851399102, pool_stale_percent=2.4925986056728107)])
None

Combinator78, 2024-10-29:
Maybe you need this data to identify the cause.

http://172.16.57.41/cgi-bin/get_status_api.cgi

STATUS=S,When=1730223591,Code=11,Msg=Summary,Description=cgminer 4.9.0 rwglr|SUMMARY,Elapsed=631677,GHS 5s=514.86,GHS av=512.84,Found Blocks=0,Getworks=58895,Accepted=106289,Rejected=666,Hardware Errors=1471,Utility=10.10,Discarded=360973,Stale=14,Get Failures=268,Local Work=591778,Remote Failures=3,Network Blocks=4178,Total MH=323948036.0000,Work Utility=473386.09,Difficulty Accepted=4937949184.00000000,Difficulty Rejected=45834240.00000000,Difficulty Stale=0.00000000,Best Share=1833440750,Device Hardware%=0.0000,Device Rejected%=0.9197,Pool Rejected%=0.9197,Pool Stale%=0.0000,Last getwork=1730223590|
STATUS=S,When=1730223591,Code=70,Msg=CGMiner stats,Description=cgminer 4.9.0 rwglr|CGMiner=4.9.0 rwglr,Miner=1.0.1.3,CompileTime=Fri 11 Jun 15:32:42 MSK 2021,Type=Antminer L3+ (024)|STATS=0,ID=L30,Elapsed=631677,Calls=0,Wait=0.000000,Max=0.000000,Min=99999999.000000,GHS 5s=514.859,GHS av=512.84,miner_count=4,frequency=394,fan_num=2,frequency1=394,frequency2=394,frequency3=394,frequency4=394,fan1=3090,fan2=3630,temp_num=4,temp1=39,temp2=38,temp3=37,temp4=38,temp2_1=48,temp2_2=46,temp2_3=45,temp2_4=45,temp31=0,temp32=0,temp33=0,temp34=0,temp4_1=0,temp4_2=0,temp4_3=0,temp4_4=0,temp_max=39,Device Hardware%=0.0000,no_matching_work=1471,chain_acn1=72,chain_acn2=72,chain_acn3=72,chain_acn4=72,chain_acs1= oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo,chain_acs2= oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo,chain_acs3= oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo,chain_acs4= oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo,chain_hw1=0,chain_hw2=695,chain_hw3=386,chain_hw4=390,chain_rate1=128.66,chain_rate2=128.92,chain_rate3=128.82,chain_rate4=128.46,chain_power1=204.88,chain_power2=204.88,chain_power3=204.88,chain_power4=204.88,chain_power=819.52,voltage1=10.05,voltage2=10.05,voltage3=10.05,voltage4=10.05|
STATUS=S,When=1730223591,Code=7,Msg=4 Pool(s),Description=cgminer 4.9.0 rwglr|POOL=0,URL=stratum+tcp://ltc.trustpool.ru:3333,Status=Alive,Priority=0,Quota=1,Long Poll=N,Getworks=32232,Accepted=96194,Rejected=204,Discarded=353828,Stale=14,Get Failures=6,Remote Failures=1,User=first1vadim.003,Last Share Time=0:00:34,Diff=65.5K,Diff1 Shares=19005300,Proxy Type=,Proxy=,Difficulty Accepted=4836327424.00000000,Difficulty Rejected=9469952.00000000,Difficulty Stale=0.00000000,Last Share Difficulty=65536.00000000,Has Stratum=true,Stratum Active=true,Stratum URL=ltc.trustpool.ru,Has GBT=false,Best Share=1833440750,Pool Rejected%=0.1954,Pool Stale%=0.0000|POOL=1,URL=stratum+tcp://ltc.trustpool.ru:25,Status=Alive,Priority=1,Quota=1,Long Poll=N,Getworks=16,Accepted=85,Rejected=0,Discarded=187,Stale=0,Get Failures=0,Remote Failures=0,User=first1vadim.003,Last Share Time=164:14:48,Diff=32.8K,Diff1 Shares=10046,Proxy Type=,Proxy=,Difficulty Accepted=2850816.00000000,Difficulty Rejected=0.00000000,Difficulty Stale=0.00000000,Last Share Difficulty=65536.00000000,Has Stratum=true,Stratum Active=false,Stratum URL=,Has GBT=false,Best Share=2606492,Pool Rejected%=0.0000,Pool Stale%=0.0000|POOL=2,URL=stratum+tcp://ltc.trustpool.ru:443,Status=Alive,Priority=2,Quota=1,Long Poll=N,Getworks=1,Accepted=0,Rejected=0,Discarded=0,Stale=0,Get Failures=0,Remote Failures=0,User=first1vadim.003,Last Share Time=0,Diff=16.4K,Diff1 Shares=0,Proxy Type=,Proxy=,Difficulty Accepted=0.00000000,Difficulty Rejected=0.00000000,Difficulty Stale=0.00000000,Last Share Difficulty=0.00000000,Has Stratum=true,Stratum Active=false,Stratum URL=,Has GBT=false,Best Share=0,Pool Rejected%=0.0000,Pool Stale%=0.0000|POOL=3,URL=*,Status=Alive,Priority=999,Quota=1,Long Poll=N,Getworks=26646,Accepted=10010,Rejected=462,Discarded=6958,Stale=0,Get Failures=262,Remote Failures=2,User=*,Last Share Time=0:18:00,Diff=65.5K,Diff1 Shares=392851,Proxy Type=,Proxy=,Difficulty Accepted=98770944.00000000,Difficulty Rejected=36364288.00000000,Difficulty Stale=0.00000000,Last Share Difficulty=8192.00000000,Has Stratum=true,Stratum Active=true,Stratum URL=proto3.bytewall.net,Has GBT=false,Best Share=181483865,Pool Rejected%=26.9096,Pool Stale%=0.0000|

b-rowan, 2024-10-29:
Its failing to parse a couple of things.

Regarding the result above, thats the same as `rpc.stats()`, so might be helpful.

Combinator78, 2024-10-30:
Excellent! Everything works.

PS D:\pyasic-0.61.4> py test.py
MinerData(ip='172.16.57.41', datetime='2024-10-30T18:57:10.533629+03:00', timestamp=1730303830, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.L3Plus: 'L3+'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', model='L3+', firmware='Stock', algo='SHA256', mac='2C:6B:7D:AD:C6:56', api_ver='3.1', fw_ver='4.9.0 rwglr', hostname=None, hashrate=None, expected_hashrate=None, expected_hashboards=3, expected_chips=216, expected_fans=2, percent_expected_chips=None, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=None, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=0), Fan(speed=0)], fan_psu=None, hashboards=[], total_chips=None, nominal=None, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='ltc.trustpool.ru:3333', user='first1vadim.003', password='123'), Pool(url='ltc.trustpool.ru:25', user='first1vadim.003', password='123'), Pool(url='ltc.trustpool.ru:443', user='first1vadim.003', password='123')], quota=1, name='L1JXNS')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=711915, efficiency=None, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=3333, pubkey=None), accepted=107978, rejected=215, get_failures=6, remote_failures=1, active=True, alive=True, index=0, user='first1vadim.003', pool_rejected_percent=0.19871895593984823, pool_stale_percent=0.005545645282042277), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=25, pubkey=None), accepted=85, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=1, user='first1vadim.003', pool_rejected_percent=0.0, pool_stale_percent=0.0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=443, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=2, user='first1vadim.003', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host=None, port=None, pubkey=None), accepted=11332, rejected=509, get_failures=294, remote_failures=2, active=True, alive=True, index=3, user='*', pool_rejected_percent=4.298623427075416, pool_stale_percent=2.482898403851026)])
PS D:\pyasic-0.61.4>

Combinator78, 2024-10-30:
Sorry, I didn't notice right away, hashplate data is not being returned
percent_expected_chips=None, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=None, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=0), Fan(speed=0)], fan_psu=None, hashboards=[], total_chips=None

b-rowan, 2024-10-30:
Not sure about that fan status either, should those be 0?

Combinator78, 2024-10-30:
[image]

b-rowan, 2024-10-30:
Want to cross check stats with rpc stats, can you run this - 

```
import asyncio
from pyasic import miner_factory

async def gather_miner_data():
    miner = await pyasic.get_miner("172.16.57.41")
    print(await miner.rpc.stats())

if __name__ == "__main__":
    asyncio.run(gather_miner_data())
```

Combinator78, 2024-10-30:
PS D:\pyasic-0.61.10> py test.py   
Traceback (most recent call last):
  File "D:\pyasic-0.61.10\test.py", line 21, in <module>
    asyncio.run(gather_miner_data())
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.10\test.py", line 16, in gather_miner_data
    miner = await pyasic.get_miner("172.16.57.41")
                  ^^^^^^
NameError: name 'pyasic' is not defined
PS D:\pyasic-0.61.10>

b-rowan, 2024-10-30:
MB, fixed the comment.  Needed the import.

Combinator78, 2024-10-30:
PS D:\pyasic-0.61.10> py test.py
{'STATUS': [{'STATUS': 'S', 'When': 1730318758, 'Code': 70, 'Msg': 'CGMiner stats', 'Description': 'cgminer 4.9.0 rwglr'}], 'STATS': [{'CGMiner': '4.9.0 rwglr', 'Miner': '1.0.1.3', 'CompileTime': 'Fri 11 Jun 15:32:42 MSK 2021', 'Type': 'Antminer L3+ (024)'}, {'STATS': 0, 'ID': 'L30', 'Elapsed': 726845, 'Calls': 0, 'Wait': 0.0, 'Max': 0.0, 'Min': 99999999.0, 'GHS 5s': '515.179', 'GHS av': 512.87, 'miner_count': 4, 'frequency': '394', 'fan_num': 2, 'frequency1': '394', 'frequency2': '394', 'frequency3': '394', 'frequency4': '394', 'fan1': 3150, 'fan2': 3750, 'temp_num': 4, 'temp1': 41, 'temp2': 38, 'temp3': 37, 'temp4': 38, 'temp2_1': 52, 'temp2_2': 46, 'temp2_3': 45, 'temp2_4': 45, 'temp31': 0, 'temp32': 0, 'temp33': 0, 'temp34': 0, 'temp4_1': 0, 'temp4_2': 0, 'temp4_3': 0, 'temp4_4': 0, 'temp_max': 42, 'Device Hardware%': 0.0, 'no_matching_work': 1685, 'chain_acn1': 72, 'chain_acn2': 72, 'chain_acn3': 72, 'chain_acn4': 72, 'chain_acs1': ' oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo', 'chain_acs2': ' oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo', 'chain_acs3': ' oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo', 'chain_acs4': ' oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo', 'chain_hw1': 0, 'chain_hw2': 816, 'chain_hw3': 440, 'chain_hw4': 429, 'chain_rate1': '129.14', 'chain_rate2': '128.79', 'chain_rate3': '128.66', 'chain_rate4': '128.59', 'chain_power1': 204.88, 'chain_power2': 204.88, 'chain_power3': 204.88, 'chain_power4': 204.88, 'chain_power': 819.52, 'voltage1': 10.05, 'voltage2': 10.05, 'voltage3': 10.05, 'voltage4': 10.05}], 'id': 1}
PS D:\pyasic-0.61.10>

b-rowan, 2024-10-30:
Ok, was a minor parsing error, can you try with 0.61.12

Combinator78, 2024-10-30:
I think everything is fine!

PS D:\pyasic-0.61.12> py test.py
MinerData(ip='172.16.57.41', datetime='2024-10-30T23:41:25.186300+03:00', timestamp=1730320885, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.L3Plus: 'L3+'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', model='L3+', firmware='Stock', algo='SHA256', mac='2C:6B:7D:AD:C6:56', api_ver='3.1', fw_ver='4.9.0 rwglr', hostname=None, hashrate=0.38559 TH/s, expected_hashrate=None, expected_hashboards=3, expected_chips=216, expected_fans=2, percent_expected_chips=100, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=46, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=0), Fan(speed=0)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=0.12835 TH/s, temp=48, chip_temp=41, chips=72, expected_chips=72, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=0.12863 TH/s, temp=45, chip_temp=38, chips=72, expected_chips=72, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=0.12861 TH/s, temp=44, chip_temp=36, chips=72, expected_chips=72, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=216, nominal=True, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='ltc.trustpool.ru:3333', user='first1vadim.003', password='123'), Pool(url='ltc.trustpool.ru:25', user='first1vadim.003', password='123'), Pool(url='ltc.trustpool.ru:443', user='first1vadim.003', password='123')], quota=1, name='JVGNXB')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=728970, efficiency=None, 
pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=3333, pubkey=None), accepted=110592, rejected=215, get_failures=6, remote_failures=1, active=True, alive=True, index=0, user='first1vadim.003', pool_rejected_percent=0.19403106301948433, pool_stale_percent=0.005414820363334447), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=25, pubkey=None), accepted=85, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=1, user='first1vadim.003', pool_rejected_percent=0.0, pool_stale_percent=0.0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=443, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=2, user='first1vadim.003', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host=None, port=None, pubkey=None), accepted=11564, rejected=520, get_failures=303, remote_failures=2, active=True, alive=True, index=3, user='*', pool_rejected_percent=4.303210857332009, pool_stale_percent=2.5074478649453824)])
PS D:\pyasic-0.61.12>

closed 2024-10-30 by b-rowan