UpstreamData/pyasic #211: Error Innosilicon A11MX

Combinator78, 2024-10-29:
[image]
PS D:\pyasic-0.60.10> py test.py
Traceback (most recent call last):
  File "D:\pyasic-0.60.10\test.py", line 18, in <module>
    asyncio.run(gather_miner_data())
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "D:\pyasic-0.60.10\test.py", line 13, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.60.10\pyasic\miners\base.py", line 533, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.60.10\pyasic\miners\base.py", line 460, in _get_data
    await asyncio.gather(rpc_command_task, web_command_task)
  File "D:\pyasic-0.60.10\pyasic\web\vnish.py", line 102, in multicommand
    data[command] = await self.send_command(command)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.60.10\pyasic\web\vnish.py", line 64, in send_command
    await self.auth()
  File "D:\pyasic-0.60.10\pyasic\web\vnish.py", line 51, in auth
    self.token = json_auth["token"]
                 ~~~~~~~~~^^^^^^^^^
KeyError: 'token'
PS D:\pyasic-0.60.10>

b-rowan, 2024-10-29:
It appears that the device is being identified as a vnish miner...

Could you send the result of the rpc commands `devdetails` and `version`?

```
print(await miner.rpc.devdetails())
print(await miner.rpc.version())
```

Combinator78, 2024-10-29:
Where should I write this code?
I'm testing in this request
import asyncio
from pyasic import get_miner

async def gather_miner_data():
    miner = await get_miner("172.16.57.41")
    if miner is not None:
        miner_data = await miner.get_data()
        print(miner_data)  # all data from the dataclass
        print(miner_data.hashrate)  # hashrate of the miner in TH/s

if __name__ == "__main__":
    asyncio.run(gather_miner_data())

b-rowan, 2024-10-29:
```
import asyncio
from pyasic import get_miner

async def gather_miner_data():
    miner = await get_miner("172.16.57.41")
    if miner is not None:
        print(await miner.rpc.devdetails())
        print(await miner.rpc.version())
        # miner_data = await miner.get_data()
        # print(miner_data) # all data from the dataclass
        # print(miner_data.hashrate) # hashrate of the miner in TH/s

if __name__ == "__main__":
    asyncio.run(gather_miner_data())
```

Combinator78, 2024-10-29:
PS D:\pyasic-0.60.10> py test.py
{'STATUS': [{'STATUS': 'S', 'When': 1730214577, 'Code': 69, 'Msg': 'Device Details', 'Description': 'innominer a10-1.1.0'}], 'DEVDETAILS': [{'DEVDETAILS': 0, 'Name': 'HLT', 'ID': 2, 'Driver': 'HLT_Co0lex', 'Kernel': '', 'Model': 'ETH_A10.SingleChain', 'Device Path': ''}, {'DEVDETAILS': 1, 'Name': 'HLT', 'ID': 0, 'Driver': 'HLT_Co0lex', 'Kernel': '', 'Model': 'ETH_A10.SingleChain', 'Device Path': ''}, {'DEVDETAILS': 2, 'Name': 'HLT', 'ID': 1, 'Driver': 'HLT_Co0lex', 'Kernel': '', 'Model': 'ETH_A10.SingleChain', 'Device Path': ''}, {'DEVDETAILS': 3, 'Name': 'HLT', 'ID': 3, 'Driver': 'HLT_Co0lex', 'Kernel': '', 'Model': 'ETH_A10.SingleChain', 'Device Path': ''}], 'id': 1}
{'STATUS': [{'STATUS': 'S', 'When': 1730214577, 'Code': 22, 'Msg': 'CGMiner versions', 'Description': 'innominer a10-1.1.0'}], 'VERSION': [{'CGMiner': 'a10-1.1.0', 'API': '3.4'}], 'id': 1}
PS D:\pyasic-0.60.10>

b-rowan, 2024-10-29:
Ok, almost done, do you know the chip counts per board/board count?

Combinator78, 2024-10-29:
[image]
I don't see the number of chips in the web interface. I'll look for more now.

b-rowan, 2024-10-29:
Should show at `http://{miner_ip}/api/getAll`

Combinator78, 2024-10-29:
http://172.16.57.32/api/summary

{
    "success": true,
    "DEVS": [
        {
            "ASC": 0,
            "Name": "HLT",
            "ID": 0,
            "Enabled": "Y",
            "Status": "Alive",
            "Temperature": 39,
            "MHS av": 406895570.64,
            "MHS 5s": 538875721.88,
            "MHS 1m": 470821977.34,
            "MHS 5m": 470821977.34,
            "MHS 15m": 470821977.34,
            "Accepted": 23741,
            "Rejected": 17,
            "Hardware Errors": 0,
            "Utility": 2.84,
            "Last Share Pool": 0,
            "Last Share Time": 1730216156,
            "Total MH": 204079616786402,
            "Difficulty Accepted": 101966818550595,
            "Difficulty Rejected": 73014444015,
            "Last Share Difficulty": 4294967295,
            "Last Valid Work": 1730216157,
            "Device Hardware%": 0,
            "Device Rejected%": 4689.4899,
            "Device Elapsed": 501553,
            "Tune hash": 399643968,
            "Hash Rate": 470.82197734,
            "Unit": "MH\/s",
            "Hash Rate H": 470821977.34
        },
        {
            "ASC": 1,
            "Name": "HLT",
            "ID": 1,
            "Enabled": "Y",
            "Status": "Alive",
            "Temperature": 40,
            "MHS av": 403864155.96,
            "MHS 5s": 472515201,
            "MHS 1m": 394390888.82,
            "MHS 5m": 394390888.82,
            "MHS 15m": 394390888.82,
            "Accepted": 23474,
            "Rejected": 107,
            "Hardware Errors": 0,
            "Utility": 2.81,
            "Last Share Pool": 0,
            "Last Share Time": 1730216164,
            "Total MH": 202559198730539,
            "Difficulty Accepted": 100820062282830,
            "Difficulty Rejected": 459561500565,
            "Last Share Difficulty": 4294967295,
            "Last Valid Work": 1730216165,
            "Device Hardware%": 0,
            "Device Rejected%": 29737.7507,
            "Device Elapsed": 501553,
            "Tune hash": 411614560,
            "Hash Rate": 394.39088882,
            "Unit": "MH\/s",
            "Hash Rate H": 394390888.82
        },
        {
            "ASC": 2,
            "Name": "HLT",
            "ID": 2,
            "Enabled": "Y",
            "Status": "Alive",
            "Temperature": 39,
            "MHS av": 410817524.72,
            "MHS 5s": 399520086.84,
            "MHS 1m": 391447083.01,
            "MHS 5m": 391447083.01,
            "MHS 15m": 391447083.01,
            "Accepted": 23986,
            "Rejected": 1,
            "Hardware Errors": 0,
            "Utility": 2.87,
            "Last Share Pool": 0,
            "Last Share Time": 1730216087,
            "Total MH": 206046711333253,
            "Difficulty Accepted": 103019085537870,
            "Difficulty Rejected": 4294967295,
            "Last Share Difficulty": 4294967295,
            "Last Valid Work": 1730216092,
            "Device Hardware%": 0,
            "Device Rejected%": 273.2188,
            "Device Elapsed": 501553,
            "Tune hash": 462900960,
            "Hash Rate": 391.44708301000003,
            "Unit": "MH\/s",
            "Hash Rate H": 391447083.01
        },
        {
            "ASC": 3,
            "Name": "HLT",
            "ID": 3,
            "Enabled": "Y",
            "Status": "Alive",
            "Temperature": 36,
            "MHS av": 403264760.38,
            "MHS 5s": 481704400.27,
            "MHS 1m": 407056215.29,
            "MHS 5m": 407056215.29,
            "MHS 15m": 407056215.29,
            "Accepted": 23538,
            "Rejected": 8,
            "Hardware Errors": 0,
            "Utility": 2.82,
            "Last Share Pool": 0,
            "Last Share Time": 1730216169,
            "Total MH": 202258551092374,
            "Difficulty Accepted": 101094940189710,
            "Difficulty Rejected": 34359738360,
            "Last Share Difficulty": 4294967295,
            "Last Valid Work": 1730216169,
            "Device Hardware%": 0,
            "Device Rejected%": 2226.6882,
            "Device Elapsed": 501553,
            "Tune hash": 443813120,
            "Hash Rate": 407.05621529,
            "Unit": "MH\/s",
            "Hash Rate H": 407056215.29
        }
    ],
    "POOLS": [
        {
            "POOL": 0,
            "URL": "stratum+tcp:\/\/etc.2miners.com:1010",
            "User": "0x0b4F31563c722b3aD59Afd920200d76EE4e0cd5C.A11_02",
            "Status": "Alive",
            "Priority": 0,
            "Quota": 1,
            "Long Poll": "N",
            "Getworks": 57002,
            "Accepted": 94739,
            "Rejected": 133,
            "Works": 5533805,
            "Discarded": 79124729,
            "Stale": 0,
            "Get Failures": 0,
            "Remote Failures": 0,
            "Last Share Time": 1730216169,
            "Proxy Type": "",
            "Proxy": "",
            "Difficulty Accepted": 406900906561005,
            "Difficulty Rejected": 571230650235,
            "Difficulty Stale": 0,
            "Last Share Difficulty": 4294967295,
            "Has Stratum": true,
            "Stratum Active": true,
            "Stratum URL": "etc.2miners.com",
            "Has GBT": false,
            "Pool Rejected%": 0.1402,
            "Pool Stale%": 0
        }
    ],
    "HARDWARE": {
        "Fan duty": 65
    },
    "tuning": false,
    "hashrates": [],
    "TotalHash": {
        "Hash Rate": 1.6637161644599998,
        "Unit": "GH\/s"
    }
}

b-rowan, 2024-10-29:
Try http://172.16.57.32/api/getAll

b-rowan, 2024-10-29:
> {"success":false,"message":"invalid request"}

Figures.  So they changed a bunch of stuff then.  Ill push a version so at least you can test stuff.

Combinator78, 2024-10-30:
D:\pyasic-0.61.2\pyasic\miners\base.py:550: UserWarning: Unknown chip count for miner type A11MX, please open an issue on GitHub (https://github.com/UpstreamData/pyasic).
  warnings.warn(
Traceback (most recent call last):
  File "D:\pyasic-0.61.2\pyasic\miners\base.py", line 496, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.2\pyasic\miners\backends\innosilicon.py", line 242, in _get_hashboards  
    hashboards[idx].chips = chips
    ~~~~~~~~~~^^^^^
IndexError: list index out of range

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "D:\pyasic-0.61.2\test.py", line 25, in <module>
    asyncio.run(gather_miner_data())
  File "C:\Program Files\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.2\test.py", line 20, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.2\pyasic\miners\base.py", line 535, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.2\pyasic\miners\base.py", line 498, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call hashboards on A11MX (Stock): 172.16.57.32 while getting data.
PS D:\pyasic-0.61.2>

Combinator78, 2024-10-30:
If I changed some code in the file pyasic\miners\base.py

        except Exception as e:
            raise APIError(
                f"Failed to call {data_name} on {self} while getting data."
            ) from e
    return miner_data
    
changed to

        except Exception as e:
                pass
    return miner_data    

Then I get an acceptable answer:

PS D:\pyasic-0.61.2> py test.py
D:\pyasic-0.61.2\pyasic\miners\base.py:548: UserWarning: Unknown chip count for miner type A11MX, please open an issue on GitHub (https://github.com/UpstreamData/pyasic).
  warnings.warn(
MinerData(ip='172.16.57.32', datetime='2024-10-30T06:37:25.742953+03:00', timestamp=1730259445, device_info=DeviceInfo(make=<MinerMake.INNOSILICON: 'Innosilicon'>, model=<InnosiliconModels.A11MX: 'A11MX'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='Innosilicon', model='A11MX', firmware='Stock', algo='SHA256', mac='A0:B0:45:3E:6C:C2', api_ver='3.4', fw_ver='a10-1.1.0', hostname=None, hashrate=0.00171601756991 TH/s, expected_hashrate=None, expected_hashboards=3, expected_chips=0, expected_fans=2, percent_expected_chips=None, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=None, env_temp=None, wattage=None, wattage_limit=1750, voltage=None, fans=[Fan(speed=85), Fan(speed=85)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None)], total_chips=None, nominal=None, config=None, fault_light=None, errors=[], is_mining=True, uptime=544821, efficiency=None, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='etc.2miners.com', port=1010, pubkey=None), accepted=102904, rejected=140, get_failures=0, remote_failures=0, active=True, alive=True, index=0, user='0x0b4F31563c722b3aD59Afd920200d76EE4e0cd5C.A11_02', pool_rejected_percent=0.13586429098249292, pool_stale_percent=0.0)])
0.00171601756991 TH/s
PS D:\pyasic-0.61.2>

b-rowan, 2024-10-30:
> If I changed some code in the file pyasic\miners\base.py

This isn't super useful, since it just ignores the issue and you get garbage data.  I'll try to take a look at it more closely a bit later.

Combinator78, 2024-10-30:
You answer so fast. Do you ever sleep? :)

b-rowan, 2024-10-30:
> You answer so fast. Do you ever sleep? :)

I pretty much get up in the morning and answer questions, then go to bed after answering questions lol

b-rowan, 2024-10-30:
> http://172.16.57.32/api/summary

Is this the only location you were able to find on this?  Are you able to find other API endpoints?  I'm trying to replace that `getAll` command, which is on older miners and held all the data...

b-rowan, 2024-10-30:
Ok, seems like the board count might be 4?  Can you check out `/api/type` and see what it returns there?

Combinator78, 2024-10-30:
yes, there are 4 of them
[image]

Combinator78, 2024-10-30:
{"success":false,"message":"invalid request"}

Combinator78, 2024-10-30:
http://172.16.57.32/overview
{"success":true,"type":"A11MX","hardware":{"status":"16:28:02 up 6 days, 20:19,  load average: 0.46, 0.38, 0.31","memUsed":74448,"memFree":165936,"memTotal":240384,"cacheUsed":42476,"cacheFree":177720,"cacheTotal":220196},"network":{"dhcp":"dhcp","ipaddress":"172.16.57.32","netmask":"255.255.255.0","gateway":"172.16.57.1","dns1":"8.8.8.8","dns2":"8.8.4.4"},"version":{"hwver":"g19","ethaddr":"a0:b0:45:3e:6c:c2","build_date":"14th of September 2022 07:38 AM","platform_v":"a11mx_20220914_073812"}}

b-rowan, 2024-10-30:
Try 0.61.5?

Combinator78, 2024-10-30:
Let's

Combinator78, 2024-10-30:
I need to step away for an hour. I'll be back.

Combinator78, 2024-10-30:
Haven't you finished version 0.61.5 yet?

b-rowan, 2024-10-30:
> Haven't you finished version 0.61.5 yet?

Should be coming here in 1 minute, tag didnt push so release didnt get triggered.

Combinator78, 2024-10-30:
PS D:\pyasic-0.61.5> py test.py
D:\pyasic-0.61.5\pyasic\miners\base.py:550: UserWarning: Unknown chip count for miner type A11MX, please open an issue on GitHub (https://github.com/UpstreamData/pyasic).
  warnings.warn(
Traceback (most recent call last):
  File "D:\pyasic-0.61.5\pyasic\miners\base.py", line 496, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.5\pyasic\miners\backends\innosilicon.py", line 119, in get_config
    self.config = MinerConfig.from_inno([pools])
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.5\pyasic\config\__init__.py", line 188, in from_inno
    return cls(pools=PoolConfig.from_inno(web_pools))
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.5\pyasic\config\pools.py", line 536, in from_inno
    return cls([PoolGroup.from_inno(web_pools)])
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.5\pyasic\config\pools.py", line 366, in from_inno
    return cls([Pool.from_inno(p) for p in web_pools])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.5\pyasic\config\pools.py", line 366, in <listcomp>
    return cls([Pool.from_inno(p) for p in web_pools])
                ^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.5\pyasic\config\pools.py", line 181, in from_inno
    url=web_pool["url"], user=web_pool["user"], password=web_pool["pass"]
        ~~~~~~~~^^^^^^^
KeyError: 'url'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "D:\pyasic-0.61.5\test.py", line 29, in <module>
    asyncio.run(gather_miner_data())
  File "C:\Program Files\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.5\test.py", line 23, in gather_miner_data
    all_miner_data = await asyncio.gather(*[miner.get_data() for miner in miners])
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.5\pyasic\miners\base.py", line 535, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.5\pyasic\miners\base.py", line 498, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call config on A11MX (Stock): 172.16.57.32 while getting data.
PS D:\pyasic-0.61.5>

b-rowan, 2024-10-30:
Ok, progress.

b-rowan, 2024-10-30:
Is there anything at http://172.16.57.32/pools or http://172.16.57.32/api/pools?

Im wondering if they maybe removed the /api prefix, maybe http://172.16.57.32/getAll works?

Combinator78, 2024-10-30:
http://172.16.57.32/pools
opens pool settings in web interface

http://172.16.57.32/api/pools
{"success":false,"message":"invalid request"}

 http://172.16.57.32/getAll
an empty web interface of ASIC opens

POST /api/summary HTTP/1.1
Accept: application/json, text/plain, */*
Accept-Encoding: gzip, deflate
Accept-Language: ru-RU,ru;q=0.9,en-US;q=0.8,en;q=0.7,ca;q=0.6,it;q=0.5,de;q=0.4
Authorization: Bearer eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJBc2ljTWluZXIiLCJpYXQiOjE3MzAzMDM0MDIsImV4cCI6MTczMDMyNTAwMiwidXNlciI6ImFkbWluIn0.1C0_oT_AVKLSvBsL3x4PiDsGf5NqABsOhaCaNyx3HAI
Connection: keep-alive
Content-Length: 2
Content-Type: application/json;charset=UTF-8
Host: 172.16.57.32
Origin: http://172.16.57.32
Referer: http://172.16.57.32/
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36

{"success":true,"DEVS":[{"ASC":0,"Name":"HLT","ID":0,"Enabled":"Y","Status":"Alive","Temperature":37,"MHS av":330741493.72,"MHS 5s":462532819.04,"MHS 1m":468054751.08,"MHS 5m":468054751.08,"MHS 15m":468054751.08,"Accepted":261,"Rejected":35,"Hardware Errors":0,"Utility":2.04,"Last Share Pool":0,"Last Share Time":1730310550,"Total MH":2542620025624,"Difficulty Accepted":1120986463995,"Difficulty Rejected":150323855325,"Last Share Difficulty":4294967295,"Last Valid Work":1730310550,"Device Hardware%":0,"Device Rejected%":774930.7432,"Device Elapsed":7688,"Tune hash":346769728,"Hash Rate":468.05475107999996,"Unit":"MH\/s","Hash Rate H":468054751.08},{"ASC":1,"Name":"HLT","ID":1,"Enabled":"Y","Status":"Alive","Temperature":40,"MHS av":384377415.58,"MHS 5s":339438871.61,"MHS 1m":362030745.76,"MHS 5m":362030745.76,"MHS 15m":362030745.76,"Accepted":326,"Rejected":18,"Hardware Errors":0,"Utility":2.54,"Last Share Pool":0,"Last Share Time":1730310537,"Total MH":2954936786536,"Difficulty Accepted":1400159338170,"Difficulty Rejected":77309411310,"Last Share Difficulty":4294967295,"Last Valid Work":1730310537,"Device Hardware%":0,"Device Rejected%":342926.1628,"Device Elapsed":7688,"Tune hash":386430400,"Hash Rate":362.03074575999995,"Unit":"MH\/s","Hash Rate H":362030745.76},{"ASC":2,"Name":"HLT","ID":2,"Enabled":"Y","Status":"Alive","Temperature":38,"MHS av":387730248.85,"MHS 5s":533549762.38,"MHS 1m":457159446.97,"MHS 5m":457159446.97,"MHS 15m":457159446.97,"Accepted":346,"Rejected":1,"Hardware Errors":0,"Utility":2.7,"Last Share Pool":0,"Last Share Time":1730310586,"Total MH":2980706584093,"Difficulty Accepted":1486058684070,"Difficulty Rejected":4294967295,"Last Share Difficulty":4294967295,"Last Valid Work":1730310586,"Device Hardware%":0,"Device Rejected%":18886.7435,"Device Elapsed":7688,"Tune hash":408024768,"Hash Rate":457.15944697000003,"Unit":"MH\/s","Hash Rate H":457159446.97},{"ASC":3,"Name":"HLT","ID":3,"Enabled":"Y","Status":"Alive","Temperature":35,"MHS av":358677923.58,"MHS 5s":441598317.58,"MHS 1m":410585593.67,"MHS 5m":410585593.67,"MHS 15m":410585593.67,"Accepted":321,"Rejected":0,"Hardware Errors":0,"Utility":2.51,"Last Share Pool":0,"Last Share Time":1730310585,"Total MH":2757368338599,"Difficulty Accepted":1378684501695,"Difficulty Rejected":0,"Last Share Difficulty":4294967295,"Last Valid Work":1730310585,"Device Hardware%":0,"Device Rejected%":0,"Device Elapsed":7688,"Tune hash":380581376,"Hash Rate":410.58559367000004,"Unit":"MH\/s","Hash Rate H":410585593.67}],"POOLS":[{"POOL":0,"URL":"stratum+tcp:\/\/etc.2miners.com:1010","User":"0x0b4F31563c722b3aD59Afd920200d76EE4e0cd5C.A11_02","Status":"Alive","Priority":0,"Quota":1,"Long Poll":"N","Getworks":1025,"Accepted":1254,"Rejected":54,"Works":87095,"Discarded":111492,"Stale":0,"Get Failures":0,"Remote Failures":0,"Last Share Time":1730310586,"Proxy Type":"","Proxy":"","Difficulty Accepted":5385888987930,"Difficulty Rejected":231928233930,"Difficulty Stale":0,"Last Share Difficulty":4294967295,"Has Stratum":true,"Stratum Active":true,"Stratum URL":"etc.2miners.com","Has GBT":false,"Pool Rejected%":4.1284,"Pool Stale%":0}],"HARDWARE":{"Fan duty":70},"tuning":false,"hashrates":[],"TotalHash":{"Hash Rate":1.69783053748,"Unit":"GH\/s"}}

Combinator78, 2024-10-30:
http://172.16.57.32/api/summary
{"success":false,"message":"invalid request"}

Combinator78, 2024-10-30:
POST /api/pools HTTP/1.1
Accept: application/json, text/plain, */*
Accept-Encoding: gzip, deflate
Accept-Language: ru-RU,ru;q=0.9,en-US;q=0.8,en;q=0.7,ca;q=0.6,it;q=0.5,de;q=0.4
Authorization: Bearer eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJBc2ljTWluZXIiLCJpYXQiOjE3MzAzMDM0MDIsImV4cCI6MTczMDMyNTAwMiwidXNlciI6ImFkbWluIn0.1C0_oT_AVKLSvBsL3x4PiDsGf5NqABsOhaCaNyx3HAI
Connection: keep-alive
Content-Length: 2
Content-Type: application/json;charset=UTF-8
Host: 172.16.57.32
Origin: http://172.16.57.32
Referer: http://172.16.57.32/pools
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/130.0.0.0 Safari/537.36

{"success" : true,"pools" : [{"url" : "stratum+tcp://etc.2miners.com:1010","user" : "0x0b4F31563c722b3aD59Afd920200d76EE4e0cd5C.A11_02","pass" : "x"}]}

b-rowan, 2024-10-30:
Ok, I think I fixed that config error in 0.61.6, try that?

Combinator78, 2024-10-30:
just in case, this is how it responds when 3 pools are filled

{
    "success": true,
    "pools": [
        {
            "url": "stratum+tcp://etc.2miners.com:1010",
            "user": "0x0b4F31563c722b3aD59Afd920200d76EE4e0cd5C.A11_02",
            "pass": "x"
        },
        {
            "url": "stratum+tcp://etc.2miners.com:1010",
            "user": "0x0b4F31563c722b3aD59Afd920200d76EE4e0cd5C.A11_02",
            "pass": "y"
        },
        {
            "url": "stratum+tcp://etc.2miners.com:1010",
            "user": "0x0b4F31563c722b3aD59Afd920200d76EE4e0cd5C.A11_02",
            "pass": "z"
        }
    ]
}

Combinator78, 2024-10-30:
D:\pyasic-0.61.6\pyasic\miners\base.py:550: UserWarning: Unknown chip count for miner type A11MX, please open an issue on GitHub (https://github.com/UpstreamData/pyasic).
  warnings.warn(
MinerData(ip='172.16.57.153', datetime='2024-10-30T21:22:40.581737+03:00', timestamp=1730312560, device_info=DeviceInfo(make=<MinerMake.INNOSILICON: 'Innosilicon'>, model=<InnosiliconModels.A11MX: 'A11MX'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='Innosilicon', model='A11MX', firmware='Stock', algo='SHA256', mac='A0:B0:45:3E:64:07', api_ver='3.4', fw_ver='a10-1.1.0', hostname=None, hashrate=0.0013054269104300002 TH/s, expected_hashrate=None, expected_hashboards=4, expected_chips=0, expected_fans=2, percent_expected_chips=0, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=33, env_temp=None, wattage=None, wattage_limit=1750, voltage=None, fans=[Fan(speed=80), Fan(speed=80)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=0.00039375401126 TH/s, temp=31, chip_temp=41, chips=8, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=0.00048441173404000003 TH/s, temp=35, chip_temp=44, chips=8, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=3, hashrate=0.00042726116513 TH/s, temp=33, chip_temp=43, chips=8, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=24, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://etc.2miners.com:1010', user='0x0b4F31563c722b3aD59Afd920200d76EE4e0cd5C.A11_01', password='x'), Pool(url='stratum+tcp://etc.2miners.com:1010', user='0x0b4F31563c722b3aD59Afd920200d76EE4e0cd5C.A11_01', password='x')], quota=1, name='2Q49DO')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[InnosiliconError(error_code=21, error_message='The PLUG signal of the hash board is not 
detected.')], is_mining=True, uptime=9307, efficiency=None, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='etc.2miners.com', port=1010, pubkey=None), accepted=1273, rejected=7, get_failures=0, remote_failures=0, active=True, alive=True, index=0, user='0x0b4F31563c722b3aD59Afd920200d76EE4e0cd5C.A11_01', pool_rejected_percent=0.546875, pool_stale_percent=0.0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='etc.2miners.com', port=1010, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=1, user='0x0b4F31563c722b3aD59Afd920200d76EE4e0cd5C.A11_01', pool_rejected_percent=0, pool_stale_percent=0)])PS D:\pyasic-0.61.6>

Combinator78, 2024-10-30:
Much better!

b-rowan, 2024-10-30:
Nice.  Just finished fixing that chip count warning in 0.61.8, should be good to go now!