UpstreamData/pyasic #212: pool poolbinance changes to poolbi0ce

Combinator78, 2024-10-29:
in the name of the pool **binance** there is an entry NAN, which changes the name to 0

PS D:\pyasic-0.60.10> py test.py
MinerData(ip='172.16.20.28', datetime='2024-10-29T18:21:57.138586+03:00', timestamp=1730215317, device_info=DeviceInfo(make=<MinerMake.WHATSMINER: 'WhatsMiner'>, model=<WhatsminerModels.M50VH10: 'M50 VH10'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='WhatsMiner', model='M50 VH10', firmware='Stock', algo='SHA256', mac='CA:09:1C:00:47:3B', api_ver=None, fw_ver='20230207.17.REL', hostname='WhatsMiner', hashrate=117.25353012 TH/s, expected_hashrate=111.97 TH/s, expected_hashboards=3, expected_chips=258, expected_fans=2, percent_expected_chips=100, percent_expected_hashrate=105, percent_expected_wattage=99, temperature_avg=70, env_temp=19.5, wattage=3467, wattage_limit=3500, voltage=None, fans=[Fan(speed=3720), Fan(speed=3690)], fan_psu=6176, hashboards=[HashBoard(slot=0, hashrate=38.21151107 TH/s, temp=67, chip_temp=79, chips=86, expected_chips=86, serial_number='HHM32X56103208K43455', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=39.31463205 TH/s, temp=70, chip_temp=81, chips=86, expected_chips=86, serial_number='HHM32X56103208K43467', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=39.727387 TH/s, temp=72, chip_temp=85, chips=86, expected_chips=86, serial_number='HHM32X56103208K43345', missing=False, tuned=None, active=None, voltage=None)], total_chips=258, nominal=True, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://sha256.**poolbi0ce**.com:8888', user='KIV0906.M50-110-03A', password='x'), Pool(url='stratum+tcp://bs.**poolbi0ce**.com:3333', user='KIV0906.M50-110-03A', password='x'), Pool(url='stratum+tcp://sha256.**poolbi0ce**.com:443', user='KIV0906.M50-110-03A', password='x')], quota=1, name='IWAK7Y')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModePowerTune(mode='power_tuning', power=3500, algo=StandardTuneAlgo(mode='standard'), scaling=None)), fault_light=False, errors=[], is_mining=True, uptime=735887, efficiency=30, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='sha256.poolbi0ce.com', port=8888, pubkey=None), accepted=50450, rejected=13, get_failures=7, remote_failures=0, active=True, alive=True, index=1, user='KIV0906.M50-110-03A', pool_rejected_percent=0.025761448982422764, pool_stale_percent=0.013871549452073797), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='bs.poolbi0ce.com', port=3333, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=2, user='KIV0906.M50-110-03A', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='sha256.poolbi0ce.com', port=443, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=3, user='KIV0906.M50-110-03A', pool_rejected_percent=0, pool_stale_percent=0)])
117.25353012 TH/s
PS D:\pyasic-0.60.10>

b-rowan, 2024-10-29:
Might be time to swap JSON parsing to something that supports inf and nan...

Combinator78, 2024-10-29:
What do I need to do?

b-rowan, 2024-10-29:
> What do I need to do?

Naw, this is an internal issue.  I changed the way the parsing is handled, should fix it.

Combinator78, 2024-10-29:
Is it possible to get the serial number of the device in addition to the serial numbers of the hashboards?
It is displayed in the web interface. For S19k pro, S21, T21
[image]

b-rowan, 2024-10-29:
> Is it possible to get the serial number of the device in addition to the serial numbers of the hashboards?
> 
> It is displayed in the web interface. For S19k pro, S21, T21
> 
> [image]
> 
> 

Possibly?  Technically that is the serial number of the control board, and I've considered it in the past, but it's pretty much only for stock antminers that would work, all other miners tend to not expose that...

Combinator78, 2024-10-29:
Yes, it is only in new devices. Will it not be too difficult for you to get it for those devices that return it?

b-rowan, 2024-10-29:
> Yes, it is only in new devices. Will it not be too difficult for you to get it for those devices that return it?
> 
> This is convenient for identifying the device, since the same serial number is pasted on the back of the device.

Agreed, but if you want to identify devices, I would recommend you use the built in `miner.fault_light_on`, which will flash the red light on the miner for identification.

I'll see what I can do regarding the SN, but pyasic tries to support a lot of miners, so if something can't be done easily for most of them it can be a bit of a problem to implement.

Combinator78, 2024-10-29:
I see, thanks for the answer

b-rowan, 2024-10-30:
Is this working as expected now?  If you want the serial numbers, I'm open to a new issue for that feature, but want to make sure this was actually fixed.

Combinator78, 2024-10-30:
Yes, it works correctly. Thank you!

PS C:\pyasic-0.61.2> py test.py
MinerData(ip='10.1.2.80', datetime='2024-10-30T08:22:59.156411+03:00', timestamp=1730265779, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.S19KPro: 'S19K Pro'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', model='S19K Pro', firmware='Stock', algo='SHA256', mac='02:56:F8:89:58:4B', api_ver='3.1', fw_ver='Fri Feb  9 13:11:38 CST 2024', hostname='Antminer', hashrate=116.43829000000001 TH/s, expected_hashrate=115.0 TH/s, expected_hashboards=3, expected_chips=231, expected_fans=4, percent_expected_chips=100, percent_expected_hashrate=101, percent_expected_wattage=None, temperature_avg=44, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=2340), Fan(speed=2340), Fan(speed=2850), Fan(speed=2820)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=39.450480000000006 TH/s, temp=43.0, chip_temp=48.0, chips=77, expected_chips=77, serial_number='JYZZYR6BCJIAF0MYW', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=38.52689 TH/s, temp=44.0, chip_temp=49.0, chips=77, expected_chips=77, serial_number='JYZZYR6BCJIAF0HDS', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=38.46092 TH/s, temp=45.5, chip_temp=50.5, chips=77, expected_chips=77, serial_number='JYZZYR6BCJIAF0HAC', missing=False, tuned=None, active=None, voltage=None)], total_chips=231, nominal=True, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='sha256.poolbinance.com:443', user='udomlya2024.88x172', password='123'), Pool(url='btc.poolbinance.com:1800', user='udomlya2024.88x172', password='123'), Pool(url='bs.poolbinance.com:3333', user='udomlya2024.88x172', password='123')], quota=1, name='E4ZYYC')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=False, errors=[], is_mining=True, uptime=756125, efficiency=None, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='sha256.poolbinance.com', port=443, pubkey=None), accepted=63379, rejected=15, get_failures=20, remote_failures=0, active=True, alive=True, index=0, user='udomlya2024.88x172', pool_rejected_percent=0.023661545256648896, pool_stale_percent=0.03154872700886519), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='btc.poolbinance.com', port=1800, pubkey=None), accepted=7, rejected=2, get_failures=90, remote_failures=0, active=True, alive=True, index=1, user='udomlya2024.88x172', pool_rejected_percent=22.22222222222222, pool_stale_percent=1000.0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='bs.poolbinance.com', port=3333, pubkey=None), accepted=18, rejected=1, get_failures=1205, remote_failures=0, active=True, alive=True, index=2, user='udomlya2024.88x172', pool_rejected_percent=5.263157894736842, pool_stale_percent=6342.105263157894)])
PS C:\pyasic-0.61.2>

closed 2024-10-30 by b-rowan