UpstreamData/pyasic #215: Error WhatsMiner M60 VK20

Combinator78, 2024-10-30:
**C:\pyasic-0.61.2\pyasic\miners\base.py:550: UserWarning: Unknown chip count for miner type M60 VK20, please open an issue on GitHub (https://github.com/UpstreamData/pyasic).**
  warnings.warn(
MinerData(ip='10.1.16.125', datetime='2024-10-30T07:40:44.429496+03:00', timestamp=1730263244, device_info=DeviceInfo(make=<MinerMake.WHATSMINER: 'WhatsMiner'>, model=<WhatsminerModels.M60VK20: 'M60 VK20'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='WhatsMiner', model='M60 VK20', firmware='Stock', algo='SHA256', mac='CE:01:19:00:A4:3C', api_ver=None, fw_ver='20240322.11.REL', hostname='WhatsMiner', hashrate=180.72077214 TH/s, expected_hashrate=177.613 TH/s, expected_hashboards=3, expected_chips=0, expected_fans=2, percent_expected_chips=0, percent_expected_hashrate=102, percent_expected_wattage=100, temperature_avg=74, env_temp=8.75, wattage=3591, wattage_limit=3600, voltage=None, fans=[Fan(speed=2880), Fan(speed=2820)], fan_psu=6128, hashboards=[HashBoard(slot=0, hashrate=60.44263929 TH/s, temp=73, chip_temp=68, chips=172, expected_chips=None, serial_number='BKM3CXAC204229X43506', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=57.12158234 TH/s, temp=69, chip_temp=63, chips=172, expected_chips=None, serial_number='BKM3CXAC204229X43867', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=63.156550509999995 TH/s, temp=80, chip_temp=75, chips=172, expected_chips=None, serial_number='BKM3CXAC204314X44770', missing=False, tuned=None, active=None, voltage=None)], total_chips=516, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://bitcoin.viabtc.com:3333', user='PKFmining.1x19x102', password='x'), Pool(url='stratum+tcp://bitcoin.viabtc.com:3333', user='PKFmining.1x19x102', password='x'), Pool(url='stratum+tcp://bitcoin.viabtc.com:3333', user='PKFmining.1x19x102', password='x')], quota=1, name='TBZKZ2')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModePowerTune(mode='power_tuning', power=3600, algo=StandardTuneAlgo(mode='standard'), scaling=None)), fault_light=False, errors=[], is_mining=True, uptime=1886120, efficiency=20, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='bitcoin.viabtc.com', port=3333, pubkey=None), accepted=4820099, rejected=979, get_failures=5, remote_failures=1, active=True, alive=True, index=1, user='PKFmining.1x19x102', pool_rejected_percent=0.020306661705120722, pool_stale_percent=0.00010371124466353792), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='bitcoin.viabtc.com', port=3333, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=2, user='PKFmining.1x19x102', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='bitcoin.viabtc.com', port=3333, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=3, user='PKFmining.1x19x102', pool_rejected_percent=0, pool_stale_percent=0)])
PS C:\pyasic-0.61.2>

b-rowan, 2024-10-30:
Will have a fix for this in the morning, won't take long.

b-rowan, 2024-10-30:
Done in 4dc5b1a5410a5f2df173126c6249b1c676898242

closed 2024-10-30 by b-rowan