UpstreamData/pyasic #216: Antminer L7 error

Combinator78, 2024-10-30:
PS C:\pyasic-0.61.2> py test.py
Traceback (most recent call last):
  File "C:\pyasic-0.61.2\pyasic\miners\base.py", line 496, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\pyasic-0.61.2\pyasic\miners\backends\antminer.py", line 411, in _get_pools
    pool_url = PoolUrl.from_str(url) if url else None
               ^^^^^^^^^^^^^^^^^^^^^
  File "C:\pyasic-0.61.2\pyasic\data\pools.py", line 29, in from_str
    scheme = Scheme(parsed_url.scheme)
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\enum.py", line 714, in __call__
    return cls.__new__(cls, value)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\enum.py", line 1130, in __new__
    raise ve_exc
ValueError: 'stratum+ssl' is not a valid Scheme

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\pyasic-0.61.2\test.py", line 16, in <module>
    asyncio.run(gather_miner_data())
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\pyasic-0.61.2\test.py", line 10, in gather_miner_data
    all_miner_data = await asyncio.gather(*[miner.get_data() for miner in miners])
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\pyasic-0.61.2\pyasic\miners\base.py", line 535, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "C:\pyasic-0.61.2\pyasic\miners\base.py", line 498, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call pools on L7 (Stock): 10.1.21.207 while getting data.
PS C:\pyasic-0.61.2> 
[image]

b-rowan, 2024-10-30:
I didn't consider the option of using SSL when creating the pool handling, I'll have to reconsider that.  I'll sleep on it.

b-rowan, 2024-10-30:
Added naive support for `stratun+ssl` in ddd3e867f9c57e31862d1697d7a91e09946d6562

closed 2024-10-30 by b-rowan