UpstreamData/pyasic #218: Antminer KS3 and Antminer KS5 Pro model not detected

Combinator78, 2024-10-30:
**Antminer KS3:**

PS C:\pyasic-0.61.2> py test.py
MinerData(ip='172.16.22.70', datetime='2024-10-30T11:04:52.279815+03:00', timestamp=1730275492, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', model=None, firmware='Stock', algo='SHA256', mac=None, api_ver='3.1', fw_ver='Fri Nov 17 09:46:18 CST 2023', hostname=None, hashrate=9.62320085812 TH/s, expected_hashrate=9.6255 TH/s, expected_hashboards=3, expected_chips=0, expected_fans=2, percent_expected_chips=0, percent_expected_hashrate=100, percent_expected_wattage=None, temperature_avg=70, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=2930), Fan(speed=2960)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=3.20876314624 TH/s, temp=71, chip_temp=56, chips=92, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=3.2063603343400002 TH/s, temp=69, chip_temp=54, chips=92, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=3.20807737754 TH/s, 
temp=70, chip_temp=55, chips=92, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=276, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://kheavyhash.auto.nicehash.com:9200', user='NHbRTqpfMtoZZ3uNiR5MLKzi9f3J6B3Su5BP.Z001', password='x'), Pool(url='stratum+ssl://kheavyhash.auto.nicehash.com:443', user='NHbRTqpfMtoZZ3uNiR5MLKzi9f3J6B3Su5BP.Z001', password='x'), Pool(url='', user='', password='x')], quota=1, name='XKM0BC')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=445698, efficiency=None, pools=[])

**Antminer KS5 Pro**

PS C:\pyasic-0.61.2> py test.py
MinerData(ip='172.16.26.13', datetime='2024-10-30T11:05:11.492419+03:00', timestamp=1730275511, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', model=None, firmware='Stock', algo='SHA256', mac=None, api_ver='3.1', fw_ver='Mon Aug  5 17:28:48 CST 2024', hostname=None, hashrate=21.61478362727 TH/s, expected_hashrate=21.0 TH/s, expected_hashboards=3, 
expected_chips=0, expected_fans=2, percent_expected_chips=0, percent_expected_hashrate=103, percent_expected_wattage=None, temperature_avg=79, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=2990), Fan(speed=2940)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=7.19700413645 TH/s, temp=80, chip_temp=58, chips=92, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=7.19216443392 TH/s, temp=77, chip_temp=59, chips=92, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=7.2256150569 TH/s, temp=79, chip_temp=57, chips=92, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=276, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://kheavyhash.auto.nicehash.com:9200', user='NHbF4CGtUNXMdcqQ6kwbUEdknzTSVL7uZGdg.z001', password='x'), Pool(url='stratum+ssl://kheavyhash.auto.nicehash.com:443', user='NHbF4CGtUNXMdcqQ6kwbUEdknzTSVL7uZGdg.z001', password='x'), Pool(url='', user='', password='x')], quota=1, name='QYHKM8')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=744671, efficiency=None, pools=[])
PS C:\pyasic-0.61.2>

b-rowan, 2024-10-30:
Perfect, should be able to work with this.  Do you mind sending the devdetails and version command results for both of these?  Just want confirmation the version string is consistent.

`miner.rpc.devdetails()` and `miner.rpc.versions()`

b-rowan, 2024-10-30:
> Where should I write this code?

You can use this code snippet - https://github.com/UpstreamData/pyasic/issues/211#issuecomment-2444549687

Combinator78, 2024-10-30:
Antminer KS3 and Antminer KS5 Pro
They return the same answer

PS C:\pyasic-0.61.2> 
 *  History restored 

Traceback (most recent call last):
  File "C:\pyasic-0.61.2\test.py", line 14, in <module>
    asyncio.run(gather_miner_data())
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\pyasic-0.61.2\test.py", line 7, in gather_miner_data
    print(await miner.rpc.devdetails())
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\pyasic-0.61.2\pyasic\rpc\cgminer.py", line 347, in devdetails
    return await self.send_command("devdetails")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\pyasic-0.61.2\pyasic\rpc\base.py", line 96, in send_command
    raise APIError(f"{command}: {validation[1]}")
pyasic.errors.APIError: devdetails: Invalid command
PS C:\pyasic-0.61.2>

b-rowan, 2024-10-30:
I'm not even sure they support the RPC api, or at least it hasn't been implemented properly.

Combinator78, 2024-10-30:
[image]
[image]

b-rowan, 2024-10-30:
Seems like just a KS5, not a pro?

Try this snippet instead for the API, i forgot they don't respond to devdetails.

```
import asyncio
from pyasic import get_miner

async def gather_miner_data():
    miner = await get_miner("172.16.57.41")
    if miner is not None:
        print(await miner.rpc.version())
        print(await miner.rpc.stats())

if __name__ == "__main__":
    asyncio.run(gather_miner_data())
```

Combinator78, 2024-10-30:
PS C:\pyasic-0.61.2> py test.py
{'STATUS': [{'STATUS': 'S', 'When': 1730299049, 'Code': 22, 'Msg': 'CGMiner versions', 'Description': 'jansson 2.12'}], 'VERSION': [{'BMMiner': '2.12', 'API': '3.1', 'Miner': '0.0-2.0.0', 'CompileTime': 'Mon Aug  5 17:28:48 CST 2024', 'Type': 'Antminer KS5'}], 'id': 1}
{'STATUS': [{'STATUS': 'S', 'When': 1730299049, 'Code': 70, 'Msg': 'CGMiner stats', 'Description': 'jansson 2.12'}], 'STATS': [{'BMMiner': '2.12', 'Miner': '0.0-2.0.0', 'CompileTime': 'Mon Aug  5 17:28:48 CST 2024', 'Type': 'Antminer KS5'}, {'STATS': 3, 'ID': 'BTM_SOC3', 'Elapsed': 6954, 'GHS 5s': 19956.13, 'GHS av': 21595.21, 'rate_30m': 21634.68, 'total_rateideal': 21000.0, 'Mode': 2, 'miner_count': 3, 'frequency': 625, 'fan_num': 4, 'fan1': 2910, 'fan2': 2880, 'fan3': 2960, 'fan4': 2910, 'temp_num': 6, 'temp1': 58, 'temp2_1': 80, 'temp2': 59, 'temp2_2': 77, 'temp3': 57, 'temp2_3': 79, 'temp_in_pcb_1': '43', 'temp_in_pcb_2': '41', 'temp_in_pcb_3': '40', 'temp_out_pcb_1': '58', 'temp_out_pcb_2': '59', 'temp_out_pcb_3': '57', 'temp_in_chip_1': '63', 'temp_in_chip_2': '61', 'temp_in_chip_3': '60', 'temp_out_chip_1': '58', 'temp_out_chip_2': '59', 'temp_out_chip_3': '57', 'temp_max': 80, 'chain_acn1': 92, 'chain_acn2': 92, 'chain_acn3': 92, 'chain_acs1': 'ooooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo ooo', 'chain_acs2': 'ooooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo ooo', 'chain_acs3': 'ooooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo oooooooo ooo', 'chain_hw1': 3044, 'chain_hw2': 3307, 'chain_hw3': 1099, 'no_matching_work': 7450, 'chain_rate1': 7183.14668032, 'chain_rate2': 7178.35311514, 'chain_rate3': 7233.71268506, 'CHAIN AVG HASHRATE1': '7183.15 GH/s', 'CHAIN AVG HASHRATE2': '7178.35 GH/s', 'CHAIN AVG HASHRATE3': '7233.71 GH/s', 'freq1': 625, 'freq2': 625, 'freq3': 625, 'miner_version': '0.0-2.0.0'}], 'id': 1}
PS C:\pyasic-0.61.2> 
[image]

b-rowan, 2024-10-30:
Ok, cool.  Both done in the latest commit then, ill release a version in a bit.