UpstreamData/pyasic #219: Iceriver miner does not return data

Combinator78, 2024-10-30:
PS C:\pyasic-0.61.2> py test.py
MinerData(ip='172.16.22.3', datetime='2024-10-30T11:22:02.792375+03:00', timestamp=1730276522, device_info=DeviceInfo(make=<MinerMake.BITAXE: 'BitAxe'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='BitAxe', model=None, firmware='Stock', algo='SHA256', mac=None, api_ver=None, fw_ver=None, hostname=None, hashrate=None, expected_hashrate=None, expected_hashboards=3, expected_chips=0, expected_fans=2, percent_expected_chips=None, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=None, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=None, 
temp=None, chip_temp=None, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None)], total_chips=None, nominal=None, config=MinerConfig(pools=PoolConfig(groups=[]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=False, errors=[], is_mining=True, uptime=None, efficiency=None, pools=[])
PS C:\pyasic-0.61.2> 
[image]

b-rowan, 2024-10-30:
Iceriver is going to be a struggle, I don't have access to one for testing, and AFAIU they are quite non-standard with the way they do things.  I pretty much put in initial identification based on a comment a while ago by someone and didn't go much past that.

I'll look into it and see what I can do, but this might be a lot of back and forth on different responses from different endpoints.

Combinator78, 2024-10-30:
Maybe I can give you some information?

Combinator78, 2024-10-30:
POST /user/userpanel HTTP/1.1
Host: 172.16.26.4
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:132.0) Gecko/20100101 Firefox/132.0
Accept: application/json, text/javascript, */*; q=0.01
Accept-Language: ru-RU,ru;q=0.8,en-US;q=0.5,en;q=0.3
Accept-Encoding: gzip, deflate
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
X-Requested-With: XMLHttpRequest
Content-Length: 6
Origin: http://172.16.26.4
Connection: keep-alive
Referer: http://172.16.26.4/
Cookie: language=en; ctime=1

HTTP/1.1 200 OK
Vary: Accept-Encoding
X-Frame-Options: SAMEORIGIN
X-Content-Type-Options: nosniff
Date: Tue, 28 Mar 2023 10:31:07 GMT
Content-Length: 2704
X-XSS-Protection: 1; mode=block
Connection: Keep-Alive
Accept-Ranges: bytes

{"error":0,"data":{"model":"none","algo":"none","online":true,"firmver1":"BOOT_2_0.BIN","firmver2":"image_1.0","softver1":"ICM168_2_1_0_ks3m_miner","softver2":"ICM168_2_1_0_ks3m_bg","firmtype":"Factory","nic":"eth0","mac":"00:0a:39:00:3c:68","ip":"172.16.26.4","netmask":"255.255.255.0","host":"petalinux_config13","dhcp":false,"gateway":"172.16.26.1","dns":"8.8.8.8","locate":false,"rtpow":"6435G","avgpow":"6284G","reject":0.412360,"runtime":"03:01:39:05","unit":"G","pows": {"board1":[6542,6263,6003,6215,6237,6300,6182,6325,6314,6252,6457,6135,6259,6289,6146,6208,6399,6318,6230,6252,6252,6219,6435]},"pows_x":["0 mins","5 mins","10 mins","15 mins","20 mins","25 mins","30 mins","35 mins","40 mins","45 mins","50 mins","55 mins","60 mins","65 mins","70 mins","75 mins","80 mins","85 mins","90 mins","95 mins","100 mins","105 mins","110 mins"],"powstate":true,"netstate":true,"fanstate":true,"tempstate":true,"fans":[5140,5229,5185,5140],"pools":[{"no":1.000000,"addr":"stratum+tcp://kheavyhash.auto.nicehash.com:9200","user":"NHbRTqpfMtoZZ3uNiR5MLKzi9f3J6B3Su5BP.zIce4","pass":"x","connect":true,"diff":"17592.19 G","priority":1.000000,"accepted":89681.000000,"rejected":372.000000,"diffa":0.000000,"diffr":0.000000,"state":1.000000,"lsdiff":0.000000,"lstime":"00:00:00"},{"no":2.000000,"addr":"","user":"","pass":"","connect":false,"diff":"0.00 G","priority":2.000000,"accepted":0.000000,"rejected":0.000000,"diffa":0.000000,"diffr":0.000000,"state":0.000000,"lsdiff":0.000000,"lstime":"00:00:00"},{"no":3.000000,"addr":"","user":"","pass":"","connect":false,"diff":"0.00 G","priority":3.000000,"accepted":0.000000,"rejected":0.000000,"diffa":0.000000,"diffr":0.000000,"state":0.000000,"lsdiff":0.000000,"lstime":"00:00:00"}],"boards":[{"no":1.000000,"chipnum":18.000000,"chipsuc":0.000000,"error":0.000000,"freq":875.000000,"rtpow":"2074.41G","avgpow":"2094.57G","idealpow":"0.00G","tempnum":"(null)","pcbtemp":"0.00-0.00-0.00-0.00","intmp":18.000000,"outtmp":38.000000,"state":true,"false":[1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18]},{"no":2.000000,"chipnum":18.000000,"chipsuc":0.000000,"error":0.000000,"freq":875.000000,"rtpow":"2228.34G","avgpow":"2089.68G","idealpow":"0.00G","tempnum":"(null)","pcbtemp":"0.00-0.00-0.00-0.00","intmp":18.000000,"outtmp":39.000000,"state":true,"false":[1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18]},{"no":3.000000,"chipnum":18.000000,"chipsuc":0.000000,"error":0.000000,"freq":875.000000,"rtpow":"2133.05G","avgpow":"2100.68G","idealpow":"0.00G","tempnum":"(null)","pcbtemp":"0.00-0.00-0.00-0.00","intmp":17.000000,"outtmp":40.000000,"state":true,"false":[1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18]}],"refTime":"2023-03-28 10:31:07 UTC"},"message":""}

b-rowan, 2024-10-30:
Should be fixed in 0.61.4, let me know.

Combinator78, 2024-10-30:
C:\pyasic-0.61.4\pyasic\miners\factory.py:850: UserWarning: Partially supported miner found: None, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(
MinerData(ip='172.16.26.4', datetime='2024-10-30T18:32:36.833619+03:00', timestamp=1730302356, device_info=DeviceInfo(make=<MinerMake.ICERIVER: 'IceRiver'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='IceRiver', model=None, firmware='Stock', algo='SHA256', mac=None, api_ver=None, fw_ver=None, hostname=None, hashrate=None, expected_hashrate=None, expected_hashboards=3, expected_chips=0, expected_fans=2, percent_expected_chips=None, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=None, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None)], total_chips=None, nominal=None, config=MinerConfig(pools=PoolConfig(groups=[]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=False, errors=[], is_mining=True, uptime=None, efficiency=None, pools=[])
PS C:\pyasic-0.61.4>

b-rowan, 2024-10-30:
Is the password on this miner custom?

Combinator78, 2024-10-30:
admin
12345678

b-rowan, 2024-10-30:
Can you try this, trying to see if the identification is failing for some reason - 
```
import asyncio
from pyasic import miner_factory

async def gather_miner_data():
    print(await miner_factory.get_miner_model_iceriver("172.16.26.4"))

if __name__ == "__main__":
    asyncio.run(gather_miner_data())
```

Combinator78, 2024-10-30:
PS C:\pyasic-0.61.4> py test.py
KS3M
PS C:\pyasic-0.61.4>

b-rowan, 2024-10-30:
Whoops, forgot to actually register the parser into the factory.  Should be identifying properly as of 0.61.7.

Combinator78, 2024-10-30:
PS C:\pyasic-0.61.8> py test.py
KS3M
PS C:\pyasic-0.61.8>

Combinator78, 2024-10-30:
KS3M - Answers correctly
PS C:\pyasic-0.61.8> py test.py
KS3M

Antminer KS5 Pro - 
PS C:\pyasic-0.61.8> py test.py
None

Combinator78, 2024-10-30:
oops, i'm checking the wrong devices. sorry

Combinator78, 2024-10-30:
PS C:\pyasic-0.61.8> py test.py
MinerData(ip='172.16.26.4', datetime='2024-10-30T21:41:54.578729+03:00', timestamp=1730313714, device_info=DeviceInfo(make=<MinerMake.ICERIVER: 'IceRiver'>, model=<IceRiverModels.KS3M: 'KS3M'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='IceRiver', model='KS3M', firmware='Stock', algo='SHA256', mac=None, api_ver=None, fw_ver=None, hostname=None, hashrate=None, expected_hashrate=None, expected_hashboards=3, expected_chips=54, expected_fans=4, percent_expected_chips=None, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=None, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=18, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=None, temp=None, chip_temp=None, 
chips=None, expected_chips=18, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=18, serial_number=None, missing=True, tuned=None, active=None, voltage=None)], total_chips=None, nominal=None, config=MinerConfig(pools=PoolConfig(groups=[]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=False, errors=[], is_mining=True, uptime=None, efficiency=None, pools=[])
PS C:\pyasic-0.61.8>

b-rowan, 2024-10-30:
Ok, its identifying properly now at least.  Can you test out `miner.web.userpanel()`?

b-rowan, 2024-10-30:
```
import asyncio
import pyasic

async def gather_miner_data():
    miner = await pyasic.get_miner "172.16.26.4")
    print(await miner.web.userpanel())

if __name__ == "__main__":
    asyncio.run(gather_miner_data())
```

b-rowan, 2024-10-30:
Ok, somethings going wrong with the multicommand then, 1 sec.

b-rowan, 2024-10-30:
I built a local simulator just to find out I was missing 2 keys in the parsing :l

Should be better in 0.61.9

Combinator78, 2024-10-30:
PS C:\pyasic-0.61.9> py test.py
{'error': 0, 'data': {'model': 'none', 'algo': 'none', 'online': True, 'firmver1': 'BOOT_2_0.BIN', 'firmver2': 'image_1.0', 'softver1': 'ICM168_2_1_0_ks3m_miner', 'softver2': 'ICM168_2_1_0_ks3m_bg', 'firmtype': 'Factory', 'nic': 'eth0', 'mac': '00:0a:39:00:3c:68', 'ip': '172.16.26.4', 'netmask': '255.255.255.0', 'host': 'petalinux_config13', 'dhcp': False, 'gateway': '172.16.26.1', 'dns': '8.8.8.8', 'locate': False, 'rtpow': '6278G', 'avgpow': '6286G', 'reject': 0.415377, 'runtime': '03:06:54:57', 'unit': 'G', 'pows': {'board1': [6322, 6285, 6237, 6344, 6047, 6113, 6006, 6094, 6318, 6102, 6083, 6230, 6058, 6248, 6259, 5941, 6358, 6043, 6307, 6516, 6373, 6201, 6278]}, 'pows_x': ['0 mins', '5 mins', '10 mins', '15 mins', '20 mins', '25 mins', '30 mins', '35 mins', '40 mins', '45 mins', '50 mins', '55 mins', '60 mins', '65 mins', '70 mins', '75 mins', '80 mins', '85 mins', '90 mins', '95 mins', '100 mins', '105 mins', '110 mins'], 'powstate': True, 'netstate': True, 'fanstate': True, 'tempstate': True, 'fans': [5140, 5185, 5185, 5140], 'pools': [{'no': 1.0, 'addr': 'stratum+tcp://kheavyhash.auto.nicehash.com:9200', 
'user': 'NHbRTqpfMtoZZ3uNiR5MLKzi9f3J6B3Su5BP.zIce4', 'pass': 'x', 'connect': True, 'diff': '17592.19 G', 'priority': 1.0, 'accepted': 95947.0, 'rejected': 400.0, 'diffa': 0.0, 'diffr': 0.0, 'state': 1.0, 'lsdiff': 0.0, 'lstime': '00:00:00'}, {'no': 2.0, 'addr': '', 'user': '', 'pass': '', 'connect': False, 'diff': '0.00 G', 'priority': 2.0, 'accepted': 0.0, 'rejected': 0.0, 'diffa': 0.0, 'diffr': 0.0, 'state': 0.0, 'lsdiff': 0.0, 'lstime': '00:00:00'}, {'no': 3.0, 'addr': '', 'user': '', 'pass': '', 'connect': False, 'diff': '0.00 G', 'priority': 3.0, 'accepted': 0.0, 'rejected': 0.0, 'diffa': 0.0, 'diffr': 0.0, 'state': 0.0, 'lsdiff': 0.0, 'lstime': '00:00:00'}], 'boards': [{'no': 1.0, 'chipnum': 18.0, 'chipsuc': 0.0, 'error': 0.0, 'freq': 875.0, 'rtpow': '2001.11G', 'avgpow': '2059.75G', 'idealpow': '0.00G', 'tempnum': '(null)', 'pcbtemp': '0.00-0.00-0.00-0.00', 'intmp': 15.0, 'outtmp': 35.0, 'state': True, 'false': [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18]}, {'no': 2.0, 'chipnum': 18.0, 'chipsuc': 0.0, 'error': 0.0, 'freq': 875.0, 'rtpow': '2129.39G', 'avgpow': '2130.00G', 'idealpow': '0.00G', 'tempnum': '(null)', 'pcbtemp': '0.00-0.00-0.00-0.00', 'intmp': 15.0, 'outtmp': 35.0, 'state': True, 'false': [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18]}, {'no': 3.0, 'chipnum': 18.0, 'chipsuc': 0.0, 'error': 0.0, 'freq': 875.0, 'rtpow': '2147.71G', 'avgpow': '2097.01G', 'idealpow': '0.00G', 'tempnum': '(null)', 'pcbtemp': '0.00-0.00-0.00-0.00', 'intmp': 14.0, 'outtmp': 36.0, 'state': True, 'false': [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18]}], 'refTime': '2023-03-28 15:46:58 UTC'}, 'message': ''}
PS C:\pyasic-0.61.9>

Combinator78, 2024-10-30:
PS C:\pyasic-0.61.9> py test.py
MinerData(ip='172.16.26.4', datetime='2024-10-30T22:24:32.179646+03:00', timestamp=1730316272, device_info=DeviceInfo(make=<MinerMake.ICERIVER: 'IceRiver'>, model=<IceRiverModels.KS3M: 'KS3M'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='IceRiver', model='KS3M', firmware='Stock', algo='SHA256', mac='00:0A:39:00:3C:68', api_ver=None, fw_ver=None, hostname='petalinux_config13', hashrate=6.5384199999999995 TH/s, expected_hashrate=None, expected_hashboards=3, expected_chips=54, expected_fans=4, percent_expected_chips=100, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=15, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=5140), Fan(speed=5185), Fan(speed=5140), Fan(speed=5095)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=2.08174 TH/s, temp=15, chip_temp=35, chips=18, expected_chips=18, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=2.15504 TH/s, temp=15, chip_temp=35, chips=18, expected_chips=18, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=2.30164 TH/s, temp=14, chip_temp=36, chips=18, expected_chips=18, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=54, nominal=True, config=MinerConfig(pools=PoolConfig(groups=[]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=False, errors=[], is_mining=True, uptime=284354, efficiency=None, pools=[])
PS C:\pyasic-0.61.9>

Combinator78, 2024-10-30:
pools do not return

b-rowan, 2024-10-30:
Pools added in 0.61.10

Combinator78, 2024-10-30:
Will you update the version?

Combinator78, 2024-10-30:
PS C:\pyasic-0.61.10> py test.py
Traceback (most recent call last):
  File "C:\pyasic-0.61.10\test.py", line 16, in <module>
    asyncio.run(gather_miner_data())    
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\pyasic-0.61.10\test.py", line 13, in gather_miner_data
    print(miner_data)    # print out all the data one by one
    ^^^^^^^^^^^^^^^^^
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\dataclasses.py", line 240, in wrapper
    result = user_function(self)
             ^^^^^^^^^^^^^^^^^^^
  File "<string>", line 3, in __repr__
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\dataclasses.py", line 240, in wrapper
    result = user_function(self)
             ^^^^^^^^^^^^^^^^^^^
  File "<string>", line 3, in __repr__
  File "C:\pyasic-0.61.10\pyasic\data\pools.py", line 81, in pool_stale_percent
    return self._calculate_percentage(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\pyasic-0.61.10\pyasic\data\pools.py", line 94, in _calculate_percentage
    return (value / total) * 100
            ~~~~~~^~~~~~~
TypeError: unsupported operand type(s) for /: 'NoneType' and 'float'
PS C:\pyasic-0.61.10>

b-rowan, 2024-10-30:
Fixed, 0.61.11

Combinator78, 2024-10-30:
please update the version

Combinator78, 2024-10-30:
config=MinerConfig(pools=PoolConfig(groups=[])

b-rowan, 2024-10-30:
> no data has been added here yet. config=MinerConfig(pools=PoolConfig(groups=[]) This is what the antminer returns: config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://ss.antpool.com:3333', user='ezhek888.Z09', password='x'), Pool(url='stratum+tcp://ss.antpool.com:443', user='ezhek888.Z09', password='x'), Pool(url='stratum+tcp://ss.antpool.com:25', user='ezhek888.Z09', password='x')], quota=1, name='AO2MA0')])

Yeah, I was doing the pools query, not the config query.  Not really sure how I'm going to handle actually sending the config to that miner, so for now I am going to likely leave that functionality out, but I will work on gathering it.

Combinator78, 2024-10-30:
Then we close the task? Or I can help if you need it in testing

b-rowan, 2024-10-30:
I think I got the parsing done, should be there in 0.60.13

Combinator78, 2024-10-30:
PS C:\pyasic-0.61.13> py test.py
MinerData(ip='172.16.26.4', datetime='2024-10-30T23:52:39.993288+03:00', timestamp=1730321559, device_info=DeviceInfo(make=<MinerMake.ICERIVER: 'IceRiver'>, model=<IceRiverModels.KS3M: 'KS3M'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='IceRiver', model='KS3M', firmware='Stock', algo='SHA256', mac='00:0A:39:00:3C:68', api_ver=None, fw_ver=None, hostname='petalinux_config13', hashrate=6.1426099999999995 TH/s, expected_hashrate=None, expected_hashboards=3, expected_chips=54, expected_fans=4, percent_expected_chips=100, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=13, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=5140), Fan(speed=5185), Fan(speed=5140), Fan(speed=5095)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=1.90216 TH/s, temp=13, chip_temp=33, chips=18, expected_chips=18, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=2.1696999999999997 TH/s, temp=13, chip_temp=33, chips=18, expected_chips=18, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=2.07075 TH/s, temp=13, chip_temp=34, chips=18, expected_chips=18, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=54, nominal=True, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://kheavyhash.auto.nicehash.com:9200', user='NHbRTqpfMtoZZ3uNiR5MLKzi9f3J6B3Su5BP.zIce4', password='x'), Pool(url='', user='', password=''), Pool(url='', user='', password='')], quota=1, name='37WDUA')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=False, errors=[], is_mining=True, uptime=289589, efficiency=None, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='kheavyhash.auto.nicehash.com', port=9200, pubkey=None), accepted=97811.0, rejected=405.0, get_failures=None, remote_failures=None, active=True, alive=True, index=1, user='NHbRTqpfMtoZZ3uNiR5MLKzi9f3J6B3Su5BP.zIce4', pool_rejected_percent=0.41235643886943063, pool_stale_percent=0)])
PS C:\pyasic-0.61.13>

Combinator78, 2024-10-30:
Thank you very much! You did a great job and so quickly, I am very grateful!

closed 2024-10-30 by b-rowan