UpstreamData/pyasic #222: Error E9Pro (Stock)

Combinator78, 2024-10-31:
PS D:\pyasic-0.61.13> py test.py
Traceback (most recent call last):
  File "D:\pyasic-0.61.13\pyasic\miners\base.py", line 496, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.13\pyasic\miners\backends\antminer.py", line 113, in get_config
    self.config = MinerConfig.from_am_modern(data)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.13\pyasic\config\__init__.py", line 170, in from_am_modern
    pools=PoolConfig.from_am_modern(web_conf),
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.13\pyasic\config\pools.py", line 543, in from_am_modern
    pool_data = web_conf["pools"]
                ~~~~~~~~^^^^^^^^^
KeyError: 'pools'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "D:\pyasic-0.61.13\test.py", line 26, in <module>
    asyncio.run(gather_miner_data())
  File "C:\Program Files\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.13\test.py", line 20, in gather_miner_data
    all_miner_data = await asyncio.gather(*[miner.get_data() for miner in miners])
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.13\pyasic\miners\base.py", line 535, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "D:\pyasic-0.61.13\pyasic\miners\base.py", line 498, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call config on E9Pro (Stock): 172.16.43.43 while getting data.
PS D:\pyasic-0.61.13>

b-rowan, 2024-11-04:
What is the result of `miner.web.get_miner_conf()` for that device?

Combinator78, 2024-11-05:
there is no error in version 0.61.16. Thanks

MinerData(ip='172.16.43.43', datetime='2024-11-05T18:06:49.284579+03:00', timestamp=1730819209, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.E9Pro: 'E9Pro'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', model='E9Pro', firmware='Stock', algo='SHA256', mac='82:DE:FD:F3:75:56', api_ver='3.1', fw_ver='Tue Aug  1 09:32:21 CST 2023', hostname='Antminer', hashrate=3.85827 TH/s, expected_hashrate=3.75 TH/s, expected_hashboards=2, expected_chips=16, expected_fans=4, percent_expected_chips=100, percent_expected_hashrate=103, percent_expected_wattage=None, temperature_avg=50, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=4440), Fan(speed=4440), Fan(speed=4440), Fan(speed=4440)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=1.90169 TH/s, temp=49.125, chip_temp=66.5, chips=8, expected_chips=8, serial_number='JYZZYPWBCJCBE07WR', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=1.95658 TH/s, temp=50.0, chip_temp=67.0, chips=8, expected_chips=8, serial_number='JYZZYPWBCJCBE080G', missing=False, tuned=None, active=None, voltage=None)], total_chips=16, nominal=True, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='solo-etc.2miners.com:5050', user='0xd9a82D723E155608FCEFfeA36035f7055621ba48.Denis9E002', password='x'), Pool(url='stratum+tcp://solo-etc.2miners.com:5151', user='0xd9a82D723E155608FCEFfeA36035f7055621ba48.Denis9E002', password='x'), Pool(url='', user='', password='')], quota=1, name='7QXDPI')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=False, errors=[], is_mining=True, uptime=5198150, efficiency=None, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='solo-etc.2miners.com', port=5050, pubkey=None), accepted=1205943, rejected=4750, get_failures=16, remote_failures=0, active=True, alive=True, index=0, user='0xd9a82D723E155608FCEFfeA36035f7055621ba48.Denis9E002', pool_rejected_percent=0.39233728121001776, pool_stale_percent=0.0013215571577600597), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='solo-etc.2miners.com', port=5151, pubkey=None), accepted=23, rejected=1, get_failures=2928, remote_failures=0, active=True, alive=True, index=1, user='0xd9a82D723E155608FCEFfeA36035f7055621ba48.Denis9E002', pool_rejected_percent=4.166666666666666, pool_stale_percent=12200.0), PoolMetrics(url=None, accepted=0, rejected=0, get_failures=0, remote_failures=0, active=True, alive=False, index=2, user='', pool_rejected_percent=0, pool_stale_percent=0)])
PS C:\vekus\agent_new>

closed 2024-11-05 by b-rowan