UpstreamData/pyasic #223: S19VNish incomplete data pools=[]

Combinator78, 2024-10-31:
could you please add the missing data in pools=[]

MinerData(ip='10.1.16.16', datetime='2024-11-01T02:45:22.101559+03:00', timestamp=1730418322, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.S19: 'S19'>, 
firmware=<MinerFirmware.VNISH: 'VNish'>, algo=SHA256Algo), make='AntMiner', model='S19', firmware='VNish', algo='SHA256', mac='02:FE:D8:A0:DE:7E', api_ver=None, fw_ver='1.2.1', hostname=None, hashrate=88.783357 TH/s, expected_hashrate=92.768266 TH/s, expected_hashboards=3, expected_chips=228, expected_fans=4, percent_expected_chips=166, percent_expected_hashrate=96, percent_expected_wattage=82, temperature_avg=74, env_temp=None, wattage=2030, wattage_limit=2475, voltage=None, fans=[Fan(speed=2220), Fan(speed=2070), Fan(speed=2220), Fan(speed=2250)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=29.99487 TH/s, temp=75, chip_temp=60, chips=126, expected_chips=76, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=29.368383 TH/s, temp=72, chip_temp=57, chips=126, expected_chips=76, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=29.420104 TH/s, temp=74, chip_temp=59, chips=126, expected_chips=76, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=378, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='ss.antpool.com:3333', user='Denyak.03', password='123'), Pool(url='btc.poolbinance.com:1800', user='Denyak.03', password='123'), Pool(url='bs.poolbinance.com:3333', user='Denyak.03', password='123')], quota=1, name='PTO8NF')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=4, minimum_speed=10), temperature=TemperatureConfig(target=75, hot=None, danger=None), mining_mode=MiningModePowerTune(mode='power_tuning', power=2475, algo=StandardTuneAlgo(mode='standard'), scaling=None)), fault_light=None, errors=[], is_mining=True, uptime=144330, efficiency=23, **pools=[]**)

b-rowan, 2024-11-04:
Any ideas on where to source this?  I am getting the pool settings currently from `web.settings()` but that wont give share information...

Combinator78, 2024-11-06:
you can close the ticket

closed 2024-11-06 by b-rowan