UpstreamData/pyasic #225: Vnish is_mining

olegudff, 2024-11-02:
Hi Developers!

I hope this message finds you well.

I would like to inquire if it's possible to modify the Vnish model's `is_mining` attribute so that when the device's API returns "miner_state": "failure" in the `/status` endpoint, is_mining would return False instead of True.

If this adjustment isn't feasible, could you please advise on how to determine whether the device is in sleep mode?

The issue I'm encountering is that I need to put devices into sleep mode and confirm that the transition was successful. Currently, I use the is_mining method to verify this. However, there are instances where the device is in an error state and is_mining returns True, causing my script to repeatedly attempt to retrieve the status or reboot the device.

Alternatively, it would be helpful to add an additional parameter that indicates the device's state (e.g., active, sleep, or error).

Thank you for your assistance.

Best regards,
Oleg Duff

b-rowan, 2024-11-02:
Hey Oleg,

This certainly should be feasible, I wasn't even aware of that value.  I don't personally use Vnish (or really have any good way to test it), so I haven't been super on top of keeping the Vnish backend up to date.  If you could send over the full JSON result from that endpoint, I'll see what I can do.

Regarding sleep/normal mode, that would normally show up in the `config` attribute (or using `miner.get_config`), but I don't have a parser set up for the Vnish config yet.  If you have ideas on what endpoints would need to be called for that, I will see what I can do.

If you need more information on either of these, let me know.  I assume you are at least partially familiar with this stuff, considering you found the right endpoint to check the status.

olegudff, 2024-11-05:
Thank you for your prompt response!

To adjust the `is_mining` status to return False when `miner_state` is in a failure condition, we can modify the `_is_mining` method by including "failure" in the list of states that indicate non-mining status.

Here’s the adjusted code snippet in `/miners/backends/vnish.py`:
```python
async def _is_mining(self, web_summary: dict = None) -> Optional[bool]:
    if web_summary is None:
        web_summary = await self.web.summary()

    if web_summary is not None:
        try:
            is_mining = not web_summary["miner"]["miner_status"]["miner_state"] in [
                "stopped",
                "shutting-down",
                "failure",
            ]
            return is_mining
        except LookupError:
            pass
```

I’m not entirely sure if the device will actually stop mining in every failure condition, as there may be certain errors where it continues to operate. In my specific case, the errors are related to the power supply, which means the device is definitely not working in this state.

Best regards,
Oleg Duff

closed 2024-11-05 by UpstreamData

b-rowan, 2024-11-05:
Have a couple other things to fix, but will push out a version with this fix shortly.

b-rowan, 2024-11-05:
Released in 0.62.0

commit c7b7fe86 (2024-11-05): feature: improve vnish is_mining functionality

Fixes #225