UpstreamData/pyasic #226: syntax error in MinerData

Combinator78, 2024-11-03:
Please correct the error in the code.  At the very end there are curly braces }, they should be simple)

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 14">
<link id=Main-File rel=Main-File
href="file:///C:\Users\Artem\AppData\Local\Temp\msohtmlclip1\01\clip.htm">
<link rel=File-List
href="file:///C:\Users\Artem\AppData\Local\Temp\msohtmlclip1\01\clip_filelist.xml">
<style>

</style>
</head>

<body link=blue vlink=purple>

MinerData(
--
ip='10.1.16.16',
datetime='2024-11-03T06:12:40.019973+03:00',
timestamp=1730603560,
device_info=DeviceInfo(
make=<MinerMake.ANTMINER: 'AntMiner'>,
model=<AntminerModels.S19: 'S19'>,
firmware=<MinerFirmware.VNISH: 'VNish'>,
algo=SHA256Algo
),
make='AntMiner',
model='S19',
firmware='VNish',
algo='SHA256',
mac='02:FE:D8:A0:DE:7E',
api_ver=None,
fw_ver='1.2.1',
hostname=None,
hashrate=89.78628499999999,  #   TH/s
expected_hashrate=92.768266,  #   TH/s
expected_hashboards=3,
expected_chips=228,
expected_fans=4,
percent_expected_chips=166,
percent_expected_hashrate=97,
percent_expected_wattage=81,
temperature_avg=72,
env_temp=None,
wattage=2015,
wattage_limit=2475,
voltage=None,
fans=[
Fan(speed=1620),
Fan(speed=1530),
Fan(speed=1590),
Fan(speed=1560)
],
fan_psu=None,
hashboards=[
HashBoard(slot=0, hashrate=29.284479, temp=74, chip_temp=59,   chips=126, expected_chips=76),
HashBoard(slot=1, hashrate=29.86129, temp=71, chip_temp=56, chips=126,   expected_chips=76),
HashBoard(slot=2, hashrate=30.640516, temp=70, chip_temp=55,   chips=126, expected_chips=76)
],
total_chips=378,
nominal=False,
config=MinerConfig(
pools=PoolConfig(groups=[
PoolGroup(pools=[
Pool(url='ss.antpool.com:3333', user='Denyak.03', password='123'),
Pool(url='btc.poolbinance.com:1800', user='Denyak.03',   password='123'),
Pool(url='bs.poolbinance.com:3333', user='Denyak.03', password='123')
])
]),
fan_mode=FanModeNormal(mode='normal', minimum_fans=4,   minimum_speed=10),
temperature=TemperatureConfig(target=75, hot=None, danger=None),
mining_mode=MiningModePowerTune(mode='power_tuning', power=2475,   algo=StandardTuneAlgo(mode='standard'))
),
fault_light=None,
errors=[],
is_mining=True,
uptime=136999,
efficiency=22,
pools=[]
)
"nominal": true,
"pool_split":   "0",
"pool_1_url":   "ru.vmcpool.io:3333",
"pool_1_user":   "Sabakakakapaka1.120x18x243",
"pool_2_url":   "eu.vmcpool.io:3333",
"pool_2_user":   "Sabakakakapaka1.120x18x243",
"pool_3_url":   "kz.vmcpool.io:3333",
"pool_3_user":   "Sabakakakapaka1.120x18x243",
"pool_4_url":   "",
"pool_4_user":   "",
"errors": [],
"fault_light": false,
"efficiency": null,
"is_mining": true
**}**
]
**}**

</body>

</html>

b-rowan, 2024-11-03:
I'm a bit confused?  Do you have more context?

Combinator78, 2024-11-03:
Sorry, it's my mistake. Please close the ticket.

closed 2024-11-03 by b-rowan