UpstreamData/pyasic #228: default password

Combinator78, 2024-11-05:
There are ASICs with different passwords on the hosting. Is it possible to add the ability to specify in the settings what passwords, in addition to the standard one, to connect to the ASIC if the standard password fails to connect?
For example:

"default_antminer_web_password": "root",
"additional_antminer_web_password": "root2",
"additional_antminer_web_password": "root3",

olegudff, 2024-11-05:
Yep

```python
from pyasic import settings

settings.update("default_antminer_web_password", password)
```

Combinator78, 2024-11-05:
Can I specify multiple lines with such parameters?
settings.update("default_antminer_web_password", password1)
settings.update("default_antminer_web_password", password2)
settings.update("default_antminer_web_password", password3)
settings.update("default_antminer_web_password", password4)

olegudff, 2024-11-05:
I think if you do it in a loop then yes.
But you will have to initialize the class again in a loop each time

Combinator78, 2024-11-05:
I think it would be nice to add such functionality to the library. Now it turns out you need to get data from the entire subnet, remember the IP addresses of devices that did not return the pool and user, then scan these IP addresses one by one with a different password.

b-rowan, 2024-11-05:
I'm not opposed to this idea, but it is a bit tricky to implement in a way that makes sense that would do it automatically.

My recommendation would be to keep a dict of IP -> password mappings, or even IP -> {web -> web_pwd, -> rpc -> rpc_pwd}, then you can just set the passwords for those devices using a custom get_miner function, in which you get the miner, then check the IP address against that dict.

closed 2025-01-23 by Combinator78