UpstreamData/pyasic #229: Partially supported miner found: Antminer Z15 Pro

Combinator78, 2024-11-05:
\pyasic\miners\factory.py:851: UserWarning: Partially supported miner found: Antminer Z15 Pro, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(

b-rowan, 2024-11-05:
Just unknown types.  I'll get those added here shortly.

b-rowan, 2024-11-05:
The innosilicon A11 doesn't seem to be getting valid data, may have to fix that in a separate issue.

b-rowan, 2024-11-05:
Added all in 0.62.0.  A11 will still have data gathering issues, can open a separate issue if you want for that.

Combinator78, 2024-11-06:
\pyasic\miners\base.py", line 498, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call pools on S19j Pro (Stock): 10.1.11.35 while getting data.

MinerData(ip='10.1.11.35', datetime='2024-11-06T05:58:22.851861+03:00', timestamp=1730861902, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.S19jPro: 'S19j Pro'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', model='S19j Pro', firmware='Stock', algo='SHA256', mac='02:E6:B4:D9:8C:C4', api_ver='3.1', fw_ver='Mon Dec 26 17:10:01 CST 2022', hostname='Antminer', hashrate=0.0 TH/s, expected_hashrate=0.0 TH/s, expected_hashboards=3, expected_chips=378, expected_fans=4, percent_expected_chips=None, percent_expected_hashrate=0, percent_expected_wattage=None, temperature_avg=None, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=3270), Fan(speed=3390), Fan(speed=4020), Fan(speed=4080)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=126, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=126, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=126, serial_number=None, missing=True, tuned=None, active=None, voltage=None)], total_chips=None, nominal=None, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='btc.viabtc.io:25', user='VMC2.ViktoriaVxG021', password='123'), Pool(url='btc.viabtc.io:25f', user='ViktoriaV.G021', password=''), Pool(url='', user='', password='')], quota=1, name='2RC4WU')]), fan_mode=FanModeManual(mode='manual', speed=90, minimum_fans=1), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=False, errors=[X19Error(error_message='Rate is lower than ideal rate', error_code=0)], is_mining=True, uptime=32991, efficiency=None, pools=[])

Combinator78, 2024-11-06:
pyasic\miners\factory.py:851: UserWarning: Partially supported miner found: KS5L, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(
MinerData(ip='172.16.57.180', datetime='2024-11-06T18:40:12.901556+03:00', timestamp=1730907612, device_info=DeviceInfo(make=<MinerMake.ICERIVER: 'IceRiver'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='IceRiver', model=None, firmware='Stock', algo='SHA256', mac='00:0A:52:30:4E:2C', api_ver=None, fw_ver=None, hostname='ICERIVER', hashrate=12.725010000000001 TH/s, expected_hashrate=None, expected_hashboards=3, expected_chips=0, expected_fans=2, percent_expected_chips=0, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=60, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=3263), Fan(speed=3218), Fan(speed=3173), Fan(speed=3173)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=4.22212 TH/s, temp=61, chip_temp=66, chips=18, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=4.33941 TH/s, temp=58, chip_temp=63, chips=18, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=4.16348 TH/s, temp=60, chip_temp=70, chips=18, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=54, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://eu.kaspa.k1pool.com:23222', user='KrUJZiTNo4ZaiE3rcXXraW5SQByMpfEEDjv.river', password='x'), Pool(url='stratum+tcp://eu.kaspa.k1pool.com:23222', user='KrUJZiTNo4ZaiE3rcXXraW5SQByMpfEEDjv.river', password='x'), Pool(url='', user='', password='')], quota=1, name='3MTG7K')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=0, errors=[], is_mining=True, uptime=5233873, efficiency=None, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='eu.kaspa.k1pool.com', port=23222, pubkey=None), accepted=916374.0, rejected=119.0, get_failures=None, remote_failures=None, active=1.0, alive=True, index=1, user='KrUJZiTNo4ZaiE3rcXXraW5SQByMpfEEDjv.river', pool_rejected_percent=0.01298427811232601, pool_stale_percent=0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='eu.kaspa.k1pool.com', port=23222, pubkey=None), accepted=0.0, rejected=0.0, get_failures=None, remote_failures=None, active=-1.0, alive=False, index=2, user='KrUJZiTNo4ZaiE3rcXXraW5SQByMpfEEDjv.river', pool_rejected_percent=0, pool_stale_percent=0)])

\pyasic\miners\factory.py:851: UserWarning: Partially supported miner found: Antminer K7, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(
MinerData(ip='172.16.16.225', datetime='2024-11-06T18:41:39.205782+03:00', timestamp=1730907699, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', model=None, firmware='Stock', algo='SHA256', mac=None, api_ver='3.1', fw_ver='Fri Dec 23 17:11:26 CST 2022', hostname=None, hashrate=42.784329301999996 TH/s, expected_hashrate=43.07072 TH/s, expected_hashboards=3, expected_chips=0, expected_fans=2, percent_expected_chips=0, percent_expected_hashrate=99, percent_expected_wattage=None, temperature_avg=68, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=4800), Fan(speed=4800)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=21.2997138022 TH/s, temp=70, chip_temp=58, chips=92, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=None, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=21.484615499799997 TH/s, temp=65, chip_temp=53, chips=92, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=184, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://mining.viabtc.io:3001', user='VMC2.ilyamurxA001', password='x'), Pool(url='stratum+tcp://mining.viabtc.io:301', user='VMC2.ilyamurxA001', password='x'), Pool(url='ыstratum+tcp://ckb-asia.f2pool.com:43009797', user='ыilyamur.A001', password='x')], quota=1, name='X73WZW')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=6739, efficiency=None, pools=[])

b-rowan, 2024-11-06:
Antminer K7 is supported in latest (0.62.1) - `"ANTMINER K7": BMMinerK7,`

I have to do that KS5L though.

Combinator78, 2024-11-06:
I tested it on the latest version 0.62.1

b-rowan, 2024-11-06:
> I tested it on the latest version 0.62.1

Ok, can you try with 0.62.2?  IceRiver KS5L support should work, and I think the antminer K7 should work too, if not we can dig deeper.

Combinator78, 2024-11-06:
Now everything is detected, but other errors have appeared
pyasic\miners\base.py:548: UserWarning: Unknown chip count for miner type A11, please open an issue on GitHub (https://github.com/UpstreamData/pyasic).
  warnings.warn(
MinerData(ip='172.16.23.13', datetime='2024-11-06T21:03:23.646241+03:00', timestamp=1730916203, device_info=DeviceInfo(make=<MinerMake.INNOSILICON: 'Innosilicon'>, model=<InnosiliconModels.A11: 'A11'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='Innosilicon', model='A11', firmware='Stock', algo='SHA256', mac='A0:B0:45:3E:21:58', api_ver='3.4', fw_ver='a10-1.1.0', hostname=None, hashrate=0.0014619337807 TH/s, expected_hashrate=None, expected_hashboards=4, expected_chips=0, expected_fans=2, percent_expected_chips=0, percent_expected_hashrate=None, percent_expected_wattage=None, temperature_avg=42, env_temp=None, wattage=None, wattage_limit=1750, voltage=None, fans=[Fan(speed=100), Fan(speed=100)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=0.00044296245168 TH/s, temp=37, chip_temp=50, chips=8, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=0.00030429920446 TH/s, temp=40, chip_temp=49, chips=8, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=0.00033420171737 TH/s, temp=45, chip_temp=56, chips=8, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=3, hashrate=0.00038047040719 TH/s, temp=46, chip_temp=57, chips=8, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=32, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://solo-etc.2miners.com:5050', user='0x5A13A3049DFB0E4e9fb6b0EA93C10690891C86af.001', password='123'), Pool(url='stratum+tcp://solo-etc.2miners.com:5151', user='0x5A13A3049DFB0E4e9fb6b0EA93C10690891C86af', password='123'), Pool(url='stratum+tcp://etc.poolbinance.com:1800', user='EtcA11Voronov.001', password='123')], quota=1, name='609222')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=89348, efficiency=None, pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='solo-etc.2miners.com', port=5050, pubkey=None), accepted=7925, rejected=321, get_failures=0, remote_failures=0, active=True, alive=True, index=0, user='0x5A13A3049DFB0E4e9fb6b0EA93C10690891C86af.001', pool_rejected_percent=3.892796507397526, pool_stale_percent=0.0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='solo-etc.2miners.com', port=5151, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=1, user='0x5A13A3049DFB0E4e9fb6b0EA93C10690891C86af', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='etc.poolbinance.com', port=1800, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=2, user='EtcA11Voronov.001', pool_rejected_percent=0, pool_stale_percent=0)])

pyasic\miners\base.py:548: UserWarning: Unknown chip count for miner type KS5, please open an issue on GitHub (https://github.com/UpstreamData/pyasic).
  warnings.warn(
MinerData(ip='172.16.15.220', datetime='2024-11-06T21:09:13.202092+03:00', timestamp=1730916553, device_info=DeviceInfo(make=<MinerMake.ICERIVER: 'IceRiver'>, model=<IceRiverModels.KS5: 'KS5'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='IceRiver', model='KS5', firmware='Stock', algo='SHA256', mac='EA:55:E4:5A:F2:45', api_ver='3.1', fw_ver='Mon Aug  5 17:28:48 CST 2024', hostname='Antminer', hashrate=21.60903 TH/s, expected_hashrate=21.0 TH/s, expected_hashboards=3, expected_chips=0, expected_fans=4, percent_expected_chips=0, percent_expected_hashrate=103, percent_expected_wattage=None, temperature_avg=46, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=3350), Fan(speed=3360), Fan(speed=3410), Fan(speed=3400)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=7.20685 TH/s, temp=46.0, chip_temp=66.0, chips=92, expected_chips=None, serial_number='JYZZYRRBDJHAJ08AX', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=7.22342 TH/s, temp=48.0, chip_temp=70.0, chips=92, expected_chips=None, serial_number='JYZZYRRBDJHAJ08Y1', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=7.1787600000000005 TH/s, temp=45.5, chip_temp=62.5, chips=92, expected_chips=None, serial_number='JYZZYRRBDJHAJ08A4', missing=False, tuned=None, active=None, voltage=None)], total_chips=276, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://kas.kryptex.network:7777', user='kaspa:qqnuvcmvk0tyzk79nawr94vraqlqjjfxw8gk0juh2rg9luqz4w08swefseqse.bitmain', password='123'), Pool(url='', user='', password=''), Pool(url='', user='', password='')], quota=1, name='RBKYSJ')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=False, errors=[], is_mining=True, uptime=2953036 [... cut ...] jfxw8gk0juh2rg9luqz4w08swefseqse.bitmain', pool_rejected_percent=0.020362715701999108, pool_stale_percent=0.00035869282401895444), PoolMetrics(url=None, accepted=0, rejected=0, get_failures=0, remote_failures=0, active=True, alive=False, index=1, user='', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=None, accepted=0, rejected=0, get_failures=0, remote_failures=0, active=True, alive=False, index=2, user='', pool_rejected_percent=0, pool_stale_percent=0)])
PS C:\vekus\agent_new>

Combinator78, 2024-11-08:
pyasic\miners\factory.py:857: UserWarning: Partially supported miner found: Antminer D9, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(
MinerData(ip='172.20.162.244', datetime='2024-11-08T14:43:46.546889+03:00', timestamp=1731066226, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=SHA256Algo), make='AntMiner', model=None, firmware='Stock', algo='SHA256', mac=None, api_ver='3.1', fw_ver='Mon Feb 20 12:40:22 CST 2023', hostname=None, hashrate=1.7839907799039998 TH/s, expected_hashrate=1.8144 TH/s, expected_hashboards=3, expected_chips=0, expected_fans=2, percent_expected_chips=0, percent_expected_hashrate=98, percent_expected_wattage=None, temperature_avg=69, env_temp=None, wattage=None, wattage_limit=None, voltage=None, fans=[Fan(speed=2400), Fan(speed=2400)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=0.593814618112 TH/s, temp=70, chip_temp=58, chips=126, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=0.59625930752 TH/s, temp=70, chip_temp=58, chips=126, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=0.593916854272 TH/s, temp=67, chip_temp=55, chips=126, expected_chips=None, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=378, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://btc.viabtc.io:3333', user='VMC2.43390xFworker1', password='x'), Pool(url='stratum+tcp://moscow.mining-dutch.nl:9998', user='43390.Fworker1', password='x'), Pool(url='', user='', password='x')], quota=1, name='4MMC1Y')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')), fault_light=None, errors=[], is_mining=True, uptime=6061136, efficiency=None, pools=[])

b-rowan, 2024-11-12:
Should have those 3 fixed in 0.62.4.

Combinator78, 2024-11-18:
it's fixed, thanks

closed 2024-11-18 by b-rowan