UpstreamData/pyasic #234: Read timeout expired

Combinator78, 2024-11-14:
Sometimes, if a device freezes, the program constantly tries to reach it, and then does not continue working. Is it possible to make it so that if the device does not respond once by timeout, the program continues scanning the next IPs?

stderr: C:agent_new\local_libs\pyasic\rpc\base.py:228 in _send_bytes
[WARNING][11/14/24 05:50:08](root) - AntminerRPCAPI: 172.16.14.241 - ([Hidden] Send Bytes) - Read timeout expired.

stderr: C:\agent_new\local_libs\pyasic\rpc\base.py:228 in _send_bytes
[WARNING][11/14/24 05:51:48](root) - AntminerRPCAPI: 172.16.14.241 - ([Hidden] Send Bytes) - Read timeout expired.

stderr: C:\agent_new\local_libs\pyasic\rpc\base.py:228 in _send_bytes
[WARNING][11/14/24 05:53:28](root) - AntminerRPCAPI: 172.16.14.241 - ([Hidden] Send Bytes) - Read timeout expired.

stderr: C:\agent_new\local_libs\pyasic\rpc\base.py:228 in _send_bytes
[WARNING][11/14/24 05:55:08](root) - AntminerRPCAPI: 172.16.14.241 - ([Hidden] Send Bytes) - Read timeout expired.

stderr: C:\agent_new\local_libs\pyasic\rpc\base.py:228 in _send_bytes
[WARNING][11/14/24 05:56:48](root) - AntminerRPCAPI: 172.16.14.241 - ([Hidden] Send Bytes) - Read timeout expired.

b-rowan, 2024-11-14:
I sort of see what you mean, but not sure I understand what the solution you want here is?  Are you gathering all the tasks together and this one is slowing it down?

Combinator78, 2024-11-14:
I collect data from a subnet, but when the program can't reach the IP, it ends with an error (updated above)

b-rowan, 2024-11-14:
> I collect data from a subnet, but when the program can't reach the IP, it ends with an error (updated above)

I think you can use `return_exceptions=True` in the gather to handle for that, then you just can log it if any item in the list returned is a subclass of `Exception`?

Combinator78, 2024-11-14:
do you need to add anything to this piece of code?
async def gather_miner_data(ip_ranges_str):
    all_subnets = ip_range_to_cidr(ip_ranges_str)
    global ip_num
    all_json_results = []

    for subnet in all_subnets:
        network = MinerNetwork.from_subnet(subnet)
        miners = await network.scan()
        
        all_results = await asyncio.gather(*[miner.get_data() for miner in miners])
        
        json_results = [format_miner_data_to_json(miner_data) for miner_data in all_results]
        all_json_results.extend(json_results)
        # Format and print the results
        json_results = [format_miner_data_to_json(miner_data) for miner_data in all_results]

        if json_results:
            print(json.dumps(json_results, indent=4))

        for result in json_results:
            if result.get('ip'):
                ip_num += 1

b-rowan, 2024-11-14:
Try this:

```python3
async def gather_miner_data(ip_ranges_str):
    all_subnets = ip_range_to_cidr(ip_ranges_str)
    global ip_num
    all_json_results = []

    for subnet in all_subnets:
        network = MinerNetwork.from_subnet(subnet)
        miners = await network.scan()

        all_results = await asyncio.gather(
            *[miner.get_data() for miner in miners], return_exceptions=True
        )

        safe_results = []
        for item in all_results:
            if isinstance(item, Exception):
                print(f"Error: {item}")
            else:
                safe_results.append(item)

        json_results = [
            format_miner_data_to_json(miner_data) for miner_data in safe_results
        ]
        all_json_results.extend(json_results)

        if json_results:
            print(json.dumps(json_results, indent=4))

        for result in json_results:
            if result.get("ip"):
                ip_num += 1

```

Combinator78, 2024-11-14:
Thank you! This helped to catch all the problematic devices

Combinator78, 2024-11-14:
File "C:\agent_new\local_libs\pyasic\miners\base.py", line 498, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call pools on Avalon 1246 (Stock): 172.16.43.27 while getting data.

please tell me how to handle this error? it occurs when incorrect pools are specified, but I still need to write this data to all_json_results.

b-rowan, 2024-11-14:
> File "C:\agent_new\local_libs\pyasic\miners\base.py", line 498, in _get_data raise APIError( pyasic.errors.APIError: Failed to call pools on Avalon 1246 (Stock): 172.16.43.27 while getting data. [...]

This is an issue with only the pools, but the rest of the data is valid?  So the idea is you want the rest of the data to write, but to not write the invalid pools?

Combinator78, 2024-11-14:
I want the data to be written even if the pool is incorrect. We have a large hosting platform, and when a client has not paid for the hosting, the technicians deliberately make changes to the pool, for example stratum+tcp://eu.emcd.io:333333333333
And this causes an error

b-rowan, 2024-11-14:
Ah.  Yeah something isn't being handled properly downstream then, likely in the pools function. Could you change that `pass` to `raise e` and let me know what that raises?

Combinator78, 2024-11-14:
Traceback (most recent call last):
  File "C:\agent_new\test.py", line 18, in <module>
    asyncio.run(gather_miner_data())
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User1\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\User1\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User1\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\agent_new\test.py", line 12, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\agent_new\local_libs\pyasic\miners\base.py", line 533, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "C:\agent_new\local_libs\pyasic\miners\base.py", line 498, in _get_data
    raise e
  File "C:\agent_new\local_libs\pyasic\miners\base.py", line 496, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\agent_new\local_libs\pyasic\miners\backends\cgminer.py", line 158, in _get_pools
    pool_url = PoolUrl.from_str(url) if url else None
               ^^^^^^^^^^^^^^^^^^^^^
  File "C:\agent_new\local_libs\pyasic\data\pools.py", line 34, in from_str
    port = parsed_url.port
           ^^^^^^^^^^^^^^^
  File "C:\Users\User1\AppData\Local\Programs\Python\Python311\Lib\urllib\parse.py", line 184, in port
    raise ValueError("Port out of range 0-65535")
ValueError: Port out of range 0-65535

b-rowan, 2024-11-14:
Lol.  Maybe tell your techs not to use an invalid port like that, its much better to prefix it with an invalid schema, such as `fail://stratum+tcp://...`.

I could fix that in pyasic I suppose, but its really not a pyasic issue...

b-rowan, 2024-11-14:
> its much better to prefix it with an invalid schema, such as `fail://stratum+tcp://...`.

Even this isnt correct now that I think about it, pyasic tries to parse the stratum type too, so from my side the best thing is just to prefix the actual path, EG `stratum+tcp://hasnt-payed.their-pool.whatever:1234`

Combinator78, 2024-11-14:
Okay, I'll instruct the technicians, thank you!