UpstreamData/pyasic #235: Is it possible to add support for ElphaPexMiner DG1+ ?

Combinator78, 2024-11-14:
api on port 4028 is missing

[image]

http://172.19.203.183/cgi-bin/get_system_info.cgi

{"ipaddress":"172.19.203.183","system_mode":"GNU\/Linux","netmask":"255.255.255.0","gateway":"","Algorithm":"Scrypt","system_kernel_version":"4.4.194 #1 SMP Sat Sep 7 16:59:20 CST 2024","system_filesystem_version":"DG1+_SW_V1.0.2","nettype":"DHCP","dnsservers":"","netdevice":"eth0","minertype":"DG1+","macaddr":"b8:4c:87:e0:1a:95","firmware_type":"Release","hostname":"DG1+"}

http://172.19.203.183/cgi-bin/summary.cgi

{"STATUS":{"STATUS":"S","when":2557706,"timestamp":1731569527,"api_version":"1.0.0","Msg":"summary"},"SUMMARY":[{"rate_unit":"MH\/s","elapsed":357357,"rate_30m":0,"rate_5s":14920.940000000001,"bestshare":0,"rate_ideal":14229,"status":[{"status":"s","type":"rate","msg":"","code":0},{"status":"s","type":"network","msg":"","code":0},{"status":"s","type":"fans","msg":"","code":0},{"status":"s","type":"temp","msg":"","code":0}],"hw_all":14199.040000000001,"rate_avg":14199.040000000001,"rate_15m":14415}],"INFO":{"miner_version":"DG1+_SW_V1.0.2","CompileTime":"","dev_sn":"28HY245192N000245C23B","type":"DG1+","hw_version":"DG1+_HW_V1.0"}}

http://172.19.203.183/cgi-bin/stats.cgi

{"STATUS":{"STATUS":"S","when":2557700,"timestamp":1731569521,"api_version":"1.0.0","Msg":"stats"},"INFO":{"miner_version":"DG1+_SW_V1.0.2","CompileTime":"","dev_sn":"28HY245192N000245C23B","type":"DG1+","hw_version":"DG1+_HW_V1.0"},"STATS":[{"rate_unit":"MH\/s","elapsed":357352,"rate_30m":0,"rate_5s":11531.879999999999,"hwp_total":0.11550000000000001,"rate_ideal":14229,"chain":[{"freq_avg":62000,"index":0,"sn":"13HY245156N000581H11JB52","temp_chip":["47125","50500","",""],"eeprom_loaded":true,"rate_15m":3507,"hw":204,"temp_pcb":[47,46,67,66],"failrate":0.029999999999999999,"asic":"ooooooooo oooooooo oooooooo oooooooo oooo","rate_real":3553.5,"asic_num":204,"temp_pic":[47,46,67,66],"rate_ideal":3557.25,"hashrate":3278.5999999999999},{"freq_avg":62000,"index":1,"sn":"13HY245156N000579H11JB52","temp_chip":["52812","56937","",""],"eeprom_loaded":true,"rate_15m":3736,"hw":204,"temp_pcb":[47,46,67,66],"failrate":0.02,"asic":"ooooooooo oooooooo oooooooo oooooooo oooo","rate_real":3550.1100000000001,"asic_num":204,"temp_pic":[47,46,67,66],"rate_ideal":3557.25,"hashrate":3491.8400000000001},{"freq_avg":62000,"index":2,"sn":"13HY245156N000810H11JB52","temp_chip":["48312","51687","",""],"eeprom_loaded":true,"rate_15m":3531,"hw":204,"temp_pcb":[47,46,67,66],"failrate":0.51000000000000001,"asic":"ooooooooo oooooooo oooooooo oooooooo oooo","rate_real":3551.8000000000002,"asic_num":204,"temp_pic":[47,46,67,66],"rate_ideal":3557.25,"hashrate":3408.6999999999998},{"freq_avg":62000,"index":3,"sn":"13HY245156N000587H11JB52","temp_chip":["46500","49062","",""],"eeprom_loaded":true,"rate_15m":3641,"hw":204,"temp_pcb":[47,46,67,66],"failrate":0.029999999999999999,"asic":"ooooooooo oooooooo oooooooo oooooooo oooo","rate_real":3543.6300000000001,"asic_num":204,"temp_pic":[47,46,67,66],"rate_ideal":3557.25,"hashrate":3463.6799999999998}],"rate_15m":14415,"chain_num":4,"fan":["5340","5400","5400","5400"],"rate_avg":14199.040000000001,"fan_num":4}]}

http://172.19.203.183/cgi-bin/get_blink_status.cgi

{"blink":false}

http://172.19.203.183/cgi-bin/get_miner_conf.cgi

{"pools":[{"url":"stratum+tcp:\/\/ltc.trustpool.ru:3333","pass":"123","user":"Nikita9231.fworker"},{"url":"stratum+tcp:\/\/ltc.trustpool.ru:443","pass":"123","user":"Nikita9231.fworker"},{"url":"stratum+tcp:\/\/ltc.trustpool.ru:25","pass":"123","user":"Nikita9231.fworker"}],"fc-voltage":"1470","fc-fan-ctrl":false,"fc-freq-level":"100","fc-fan-pwm":"80","algo":"ltc","fc-work-mode":0,"fc-freq":"1850"}

b-rowan, 2025-02-03:
Finally getting around to this, are you able to send the raw response from `http:{ip}/`?  Need to see if there is anything I can use to id this since the 4028 API isnt there.

Combinator78, 2025-02-03:
http://172.19.153.217/ - The link redirects to the address
http://172.19.153.217/#/dashboard

b-rowan, 2025-02-03:
Does this require logging in somehow?  If you hit the endpoint from curl does it show something different?

`curl http://172.19.153.217/`

Combinator78, 2025-02-04:
curl http://172.19.153.217/
Authorization Required

b-rowan, 2025-02-04:
> **curl http://172.19.153.217/**
> Authorization Required

Ok, so it is refusing connection there.  Can you send the result of this?  Need to check if there is anything I can identify with in the headers...

```
import asyncio
import httpx

async def main():
    async with httpx.AsyncClient() as c:
        resp = await c.get("http://172.19.153.217/")
    print("status code:", resp.status_code)
    print("headers:", resp.headers)

if __name__ == "__main__":
    asyncio.run(main())
```

Combinator78, 2025-02-04:
status code: 401
headers: Headers({'connection': 'Keep-Alive', 'transfer-encoding': 'chunked', 'keep-alive': 'timeout=20', 'www-authenticate': 'Basic realm="Daoge Configuration"', 'content-type': 'text/plain'})

b-rowan, 2025-02-04:
Perfect.  Will have this done in a bit hopefully.

Combinator78, 2025-02-05:
\pyasic\miners\factory.py:1081: UserWarning: Partially supported miner found: DG1+, type: MinerTypes.ELPHAPEX, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(
ip='172.19.153.217' device_info=DeviceInfo(make=<MinerMake.ELPHAPEX: 'Elphapex'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.base.GenericAlgo'>) mac='b8:4c:87:e0:1a:95' api_ver='1.0.0' fw_ver='1.0.2' hostname='DG1+' sticker_hashrate=None expected_hashrate=None expected_hashboards=None expected_chips=0 expected_fans=None env_temp=None wattage=None voltage=None fans=[] fan_psu=None hashboards=[] config=MinerConfig(pools=PoolConfig(groups=[]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=False errors=[] is_mining=True uptime=None pools=[] hashrate=None wattage_limit=None total_chips=None nominal=None percent_expected_chips=None percent_expected_hashrate=None percent_expected_wattage=None temperature_avg=None efficiency=None datetime='2025-02-05T08:39:11.806136+03:00' timestamp=1738733951 make='Elphapex' model=None firmware='Stock' algo='Generic (Unknown)'

jpcomps, 2025-02-05:
@b-rowan might have missed this in the review, should fix it 

https://github.com/UpstreamData/pyasic/blob/master/pyasic/miners/factory.py#L671
`"DG1+1": ElphapexDG1Plus,`

Combinator78, 2025-02-05:
Pools and workers not defined

ip='172.19.153.217' device_info=DeviceInfo(make=<MinerMake.ELPHAPEX: 'Elphapex'>, model=<ElphapexModels.DG1Plus: 'DG1+'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.scrypt.ScryptAlgo'>) mac='b8:4c:87:e0:1a:95' api_ver='1.0.0' fw_ver='1.0.2' hostname='DG1+' sticker_hashrate=None expected_hashrate=None expected_hashboards=4 expected_chips=816 expected_fans=4 env_temp=None wattage=None voltage=None fans=[Fan(speed=3960), Fan(speed=4020), Fan(speed=4020), Fan(speed=4020)] fan_psu=None hashboards=[HashBoard(slot=0, hashrate=3.545 GH/s, temp=56.5, chip_temp=37.4685, chips=204, expected_chips=204, serial_number='13HY245156N000581H11JB52', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=3.5445300000000004 GH/s, temp=56.5, chip_temp=40.7495, chips=204, expected_chips=204, serial_number='13HY245156N000579H11JB52', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=3.52554 GH/s, temp=56.5, chip_temp=37.781, chips=204, expected_chips=204, serial_number='13HY245156N000810H11JB52', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=3, hashrate=3.5478899999999998 GH/s, temp=56.5, chip_temp=37.031, chips=204, expected_chips=204, serial_number='13HY245156N000587H11JB52', missing=False, tuned=None, active=None, voltage=None)] config=MinerConfig(pools=PoolConfig(groups=[]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=False errors=[] is_mining=True uptime=None pools=[] hashrate=14.162959999999998 GH/s wattage_limit=None total_chips=816 nominal=True percent_expected_chips=100 percent_expected_hashrate=None percent_expected_wattage=None temperature_avg=56 efficiency=None datetime='2025-02-05T18:11:11.448381+03:00' timestamp=1738768271 make='Elphapex' model='ElphapexModels.DG1Plus' firmware='Stock' algo='Scrypt'
unit=GH/s rate=14.162959999999998

b-rowan, 2025-02-05:
Does the `pools.cgi` endpoint exist?  I know conf does but I think that one should exist for pools metrics.

Combinator78, 2025-02-05:
{"STATUS":{"STATUS":"S","when":5411762,"timestamp":1738768594,"api_version":"1.0.0","Msg":"pools"},"Device Total Rejected":8888,"POOLS":[{"diffs":0,"diffr":524288,"index":0,"user":"Nikita9231.fworker","lsdiff":524288,"lstime":"00:00:18","diffa":524288,"accepted":798704,"diff1":0,"stale":0,"diff":"","rejected":3320,"status":"Unreachable","getworks":802024,"priority":0,"url":"stratum+tcp:\/\/ltc.trustpool.ru:3333"},{"diffs":0,"diffr":524288,"index":1,"user":"Nikita9231.fworker","lsdiff":524288,"lstime":"00:00:00","diffa":524288,"accepted":604803,"diff1":0,"stale":0,"diff":"","rejected":2492,"status":"Alive","getworks":607295,"priority":1,"url":"stratum+tcp:\/\/ltc.trustpool.ru:443"},{"diffs":0,"diffr":524288,"index":2,"user":"Nikita9231.fworker","lsdiff":524288,"lstime":"00:00:05","diffa":524288,"accepted":691522,"diff1":0,"stale":0,"diff":"","rejected":3076,"status":"Unreachable","getworks":694598,"priority":2,"url":"stratum+tcp:\/\/ltc.trustpool.ru:25"}],"Device Rejected%":0.41999999999999998,"Device Total Work":2103917,"INFO":{"miner_version":"DG1+_SW_V1.0.2","CompileTime":"","dev_sn":"28HY245192N000245C23B","type":"DG1+","hw_version":"DG1+_HW_V1.0"}}

b-rowan, 2025-02-05:
Ok, try with 0.71.6?

Combinator78, 2025-02-05:
Hashrate is not defined

ip='172.19.153.217' device_info=DeviceInfo(make=<MinerMake.ELPHAPEX: 'Elphapex'>, model=<ElphapexModels.DG1Plus: 'DG1+'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.scrypt.ScryptAlgo'>) mac='b8:4c:87:e0:1a:95' api_ver=None fw_ver=None hostname=None sticker_hashrate=None expected_hashrate=None expected_hashboards=4 expected_chips=816 expected_fans=4 env_temp=None wattage=None voltage=None fans=[Fan(speed=None), Fan(speed=None), Fan(speed=None), Fan(speed=None)] fan_psu=None hashboards=[HashBoard(slot=0, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=204, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=204, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=204, serial_number=None, missing=True, tuned=None, active=None, voltage=None), HashBoard(slot=3, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=204, serial_number=None, missing=True, tuned=None, active=None, voltage=None)] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://ltc.trustpool.ru:3333', user='Nikita9231.fworker', password='123'), Pool(url='stratum+tcp://ltc.trustpool.ru:443', user='Nikita9231.fworker', password='123'), Pool(url='stratum+tcp://ltc.trustpool.ru:25', user='Nikita9231.fworker', password='123')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=None errors=[] is_mining=True uptime=None pools=[] hashrate=None wattage_limit=None total_chips=None nominal=None percent_expected_chips=None percent_expected_hashrate=None percent_expected_wattage=None temperature_avg=None efficiency=None datetime='2025-02-05T20:57:24.900769+03:00' timestamp=1738778244 make='Elphapex' model='ElphapexModels.DG1Plus' firmware='Stock' algo='Scrypt'

Combinator78, 2025-02-05:
There are no errors. I just updated the library, nothing else changed.

Combinator78, 2025-02-06:
I don't know what it was yesterday, I checked it now - everything is in order. I apologize if I took up your time looking for the problem...

ip='172.19.153.217' device_info=DeviceInfo(make=<MinerMake.ELPHAPEX: 'Elphapex'>, model=<ElphapexModels.DG1Plus: 'DG1+'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.scrypt.ScryptAlgo'>) mac='b8:4c:87:e0:1a:95' api_ver='1.0.0' fw_ver='1.0.2' hostname='DG1+' sticker_hashrate=None expected_hashrate=None expected_hashboards=4 expected_chips=816 expected_fans=4 env_temp=None wattage=None voltage=None fans=[Fan(speed=4020), Fan(speed=4020), Fan(speed=4020), Fan(speed=4020)] fan_psu=None hashboards=[HashBoard(slot=0, hashrate=3.5445900000000004 GH/s, temp=56.5, chip_temp=37.1875, chips=204, expected_chips=204, serial_number='13HY245156N000581H11JB52', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=3.54407 GH/s, temp=56.5, chip_temp=42.281, chips=204, expected_chips=204, serial_number='13HY245156N000579H11JB52', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=3.5254499999999998 GH/s, temp=56.5, chip_temp=37.8435, chips=204, expected_chips=204, serial_number='13HY245156N000810H11JB52', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=3, hashrate=3.54759 GH/s, temp=56.5, chip_temp=36.281, chips=204, expected_chips=204, serial_number='13HY245156N000587H11JB52', missing=False, tuned=None, active=None, voltage=None)] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://ltc.trustpool.ru:3333', user='Nikita9231.fworker', password='123'), Pool(url='stratum+tcp://ltc.trustpool.ru:443', user='Nikita9231.fworker', password='123'), Pool(url='stratum+tcp://ltc.trustpool.ru:25', user='Nikita9231.fworker', password='123')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=False errors=[] is_mining=True uptime=None pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=3333, pubkey=None), accepted=815217, rejected=3401, get_failures=0, remote_failures=None, active=False, alive=False, index=0, user='Nikita9231.fworker', pool_rejected_percent=0.415456293411579, pool_stale_percent=0.0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=443, pubkey=None), accepted=613528, rejected=2526, get_failures=0, remote_failures=None, active=False, alive=False, index=1, user='Nikita9231.fworker', pool_rejected_percent=0.4100289909650777, pool_stale_percent=0.0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=25, pubkey=None), accepted=704094, rejected=3137, get_failures=0, remote_failures=None, active=True, alive=True, index=2, user='Nikita9231.fworker', pool_rejected_percent=0.4435608733214466, pool_stale_percent=0.0)] hashrate=14.1617 GH/s wattage_limit=None total_chips=816 nominal=True percent_expected_chips=100 percent_expected_hashrate=None percent_expected_wattage=None temperature_avg=56 efficiency=None datetime='2025-02-06T18:25:38.241770+03:00' timestamp=1738855538 make='Elphapex' model='ElphapexModels.DG1Plus' firmware='Stock' algo='Scrypt'
unit=GH/s rate=14.1617

b-rowan, 2025-02-06:
Nice, glad to hear it works!

PR #290 (2025-02-04): feature: add support for Elphapex DG1+
Closes: #235
closed 2025-02-06 by b-rowan