UpstreamData/pyasic #241: After updating today, when trying to scan a network of miners (mostly L3+), I get so many warnings

smalldaemon, 2024-11-22:
/home/user/.local/lib/python3.12/site-packages/pydantic/_internal/_generate_schema.py:1918: UserWarning: Pydantic serializer warnings:
  Expected `int` but got `float` with value `31.0` - serialized value may not be as expected
  Expected `int` but got `float` with value `36.0` - serialized value may not be as expected
  Expected `int` but got `float` with value `31.75` - serialized value may not be as expected
  Expected `int` but got `float` with value `36.75` - serialized value may not be as expected
  Expected `int` but got `float` with value `31.5` - serialized value may not be as expected
  Expected `int` but got `float` with value `36.5` - serialized value may not be as expected
  lambda x, h: h(x), schema=core_schema.any_schema()

smalldaemon, 2024-11-22:
I think it's about these values

            {
                "slot": 2,
                "hashrate": 34.37073,
                "temp": 31.75,
                "chip_temp": 36.75,
                "chips": 126,
                "expected_chips": 126,
                "serial_number": "HYDTYN3BAJHBD02LD",
                "missing": false,
                "tuned": null,
                "active": null,
                "voltage": null
            }

smalldaemon, 2024-11-22:
Now when I try to run my code with scanning the network and getting the hashrate value, I get an error

`  File "/home/warp/scanner.py", line 72, in scan_miners
    await asyncio.gather(*tasks)
          │       │       └ [<coroutine object scan_subnet at 0xffff8124df40>, <coroutine object scan_subnet at 0xffff8124e130>, <coroutine object scan_s...
          │       └ <function gather at 0xffff84438a40>
          └ <module 'asyncio' from '/usr/local/lib/python3.12/asyncio/__init__.py'>

  File "/home/warp/scanner.py", line 47, in scan_subnet
    data = await miner.get_data()
                 │     └ <function MinerProtocol.get_data at 0xffff8259bce0>
                 └ L3+ (Stock): 10.1.2.2

  File "/home/warp/.local/lib/python3.12/site-packages/pyasic/miners/base.py", line 537, in get_data
    gathered_data = await self._get_data(
                          │    └ <function MinerProtocol._get_data at 0xffff8259bc40>
                          └ L3+ (Stock): 10.1.2.2
  File "/home/warp/.local/lib/python3.12/site-packages/pyasic/miners/base.py", line 499, in _get_data
    raise e
  File "/home/warp/.local/lib/python3.12/site-packages/pyasic/miners/base.py", line 497, in _get_data
    miner_data[data_name] = await function(**args_to_send)
    │          │                  │          └ {'rpc_stats': None}
    │          │                  └ <bound method AntminerOld._get_hashboards of L3+ (Stock): 10.1.2.2>
    │          └ 'hashboards'
    └ {'mac': 'C8:DF:84:C1:67:88', 'api_ver': None, 'fw_ver': None, 'hostname': None, 'hashrate': None, 'expected_hashrate': None}
  File "/home/warp/.local/lib/python3.12/site-packages/pyasic/miners/backends/antminer.py", line 602, in _get_hashboards
    boards = rpc_stats["STATS"]
             └ {}

KeyError: 'STATS'`

b-rowan, 2024-11-22:
Should be fixed in 0.64.1

smalldaemon, 2024-11-23:
File "/home/warp/scanner.py", line 59, in scan_subnet
    data = await miner.get_data()
                 │     └ <function MinerProtocol.get_data at 0xffff80294b80>
                 └ AntMiner (Stock): 10.1.8.36

  File "/home/warp/.local/lib/python3.12/site-packages/pyasic/miners/base.py", line 532, in get_data
    for i in range(self.expected_hashboards)
                   │    └ None
                   └ AntMiner (Stock): 10.1.8.36

TypeError: 'NoneType' object cannot be interpreted as an integer

smalldaemon, 2024-11-24:
Also a large number of L7 scanning errors have appeared

```Failed to call expected_hashrate on L7 (Stock): 10.12.10.1 while getting data.
Failed to call expected_hashrate on L7 (Stock): 10.12.10.2 while getting data.
Failed to call expected_hashrate on L7 (Stock): 10.12.10.3 while getting data.
Failed to call expected_hashrate on L7 (Stock): 10.12.10.4 while getting data.
Failed to call expected_hashrate on L7 (Stock): 10.12.10.5 while getting data.
Failed to call expected_hashrate on L7 (Stock): 10.12.10.6 while getting data.
Failed to call expected_hashrate on L7 (Stock): 10.12.10.7 while getting data.
Failed to call expected_hashrate on L7 (Stock): 10.12.10.9 while getting data.
Failed to call expected_hashrate on L7 (Stock): 10.12.10.10 while getting data.
Failed to call expected_hashrate on L7 (Stock): 10.12.10.12 while getting data.
Failed to call expected_hashrate on L7 (Stock): 10.12.10.13 while getting data.
Failed to call expected_hashrate on L7 (Stock): 10.12.10.15 while getting data.```

b-rowan, 2024-11-25:
> ```
>     data = await miner.get_data()
>                  │     └ <function MinerProtocol.get_data at 0xffff80294b80>
>                  └ AntMiner (Stock): 10.1.8.36
>
>   File [...]

This is because a miner is being scanned that pyasic is unable to find the type of, it is returning a None type.

smalldaemon, 2024-11-26:
> 10.1.8.36

It's a regular L3+ like the others - I just checked it out

b-rowan, 2024-11-26:
Can you send the output of the `/cgi-bin/get_system_info.cgi` endpoint and the `version` RPC command?

smalldaemon, 2024-11-26:
```{
"minertype":"Antminer L3+",
"nettype":"DHCP",
"netdevice":"eth0",
"macaddr":"C4:F3:12:9D:61:28",
"hostname":"antMiner",
"ipaddress":"10.1.8.36",
"netmask":"255.255.255.0",
"gateway":"",
"dnsservers":"",
"curtime":"03:24:57",
"uptime":"4:48",
"loadaverage":"1.23, 1.23, 1.14",
"mem_total":"251180",
"mem_used":"58976",
"mem_free":"192204",
"mem_buffers":"0",
"mem_cached":"0",
"system_mode":"GNU/Linux",
"ant_hwv":"1.0.1.3",
"system_kernel_version":"Linux 3.8.13 #22 SMP Tue Dec 2 15:26:11 CST 2014",
"system_filesystem_version":"Mon Nov 18 09:30:34 UTC 2024",
"cgminer_version":"4.9.0 rwglr"
}```

b-rowan, 2024-11-28:
Might have just been a temporary bug where the miner may not have responded with the correct type I guess.

closed 2024-12-10 by b-rowan