UpstreamData/pyasic #242: Problem with send_config() Bos miner version 2024-01-15-0-e8253ec1-23.12.1-plus

Awote, 2024-11-27:
Hi, i tried to change password on miner's pool like this:
```from pyasic import get_miner, AnyMiner
from pyasic.data import MinerData
import asyncio

async def test_pool() -> None:
    m: AnyMiner = await get_miner("10.1.50.122")
    m_d: MinerData = await m.get_data()
    config = m_d.config
    p_g = await m.web.get_pool_groups()
    config.pools.groups[0].pools[0].password = "1"
    await m.send_config(config)
    pass

asyncio.run(test_pool())```

And i get this error
```Exception has occurred: APIError
gRPC command failed - <bound method PerformanceServiceStub.set_performance_mode of <pyasic.web.braiins_os.boser.BOSMinerGRPCStub object at 0x00000219DF878A10>>
grpclib.exceptions.GRPCError: (<Status.UNAUTHENTICATED: 16>, 'Missing or invalid authentication token', None)

During handling of the above exception, another exception occurred:

grpclib.exceptions.GRPCError: (<Status.INVALID_ARGUMENT: 3>, 'power target [0W] is not in allowed range [Some(1700), Some(6500)]', None)

The above exception was the direct cause of the following exception:

  File "C:\Users\AIPROMAP-IPSCANNER\Desktop\t.py", line 12, in test_pool
    await m.send_config(config)
  File "C:\Users\AIPROMAP-IPSCANNER\Desktop\t.py", line 16, in <module>
    asyncio.run(test_pool())
pyasic.errors.APIError: gRPC command failed - <bound method PerformanceServiceStub.set_performance_mode of <pyasic.web.braiins_os.boser.BOSMinerGRPCStub object at 0x00000219DF878A10>>```

MB i do smth wrong?

b-rowan, 2024-11-27:
Seems like th power target isn't getting set correctly in the config, can you send the result of `await miner.get_config()`?

Awote, 2024-11-27:
Yes

```
MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://btc.foundryusapool.com:3333', user='soldierlakeimmersion.64cfd9473419', password=''), Pool(url='stratum+tcp://btc.foundryusapool.com:25', user='soldierlakeimmersion.64cfd9473419', password=''), Pool(url='stratum+tcp://btc.foundryusapool.com:443', user='soldierlakeimmersion.64cfd9473419', password='')], quota=1, name='test')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModePowerTune(mode='power_tuning', power=None, algo=StandardTuneAlgo(mode='standard'), scaling=ScalingConfig(step=None, minimum=None, shutdown=None)))
```

b-rowan, 2024-11-27:
Yeah, failing to find the power mode data.  Whats the result of `await miner.web.get_miner_configuration()`?

Awote, 2024-11-27:
```{
  "poolGroups": [
    {
      "uid": "0",
      "name": "foremanGroup",
      "quota": {
        "value": 1
      },
      "pools": [
        {
          "uid": "1",
          "url": "stratum+tcp://btc.foundryusapool.com:3333",
          "user": "soldierlakeimmersion.64cfd9473419",
          "password": "",
          "enabled": true
        },
        {
          "uid": "2",
          "url": "stratum+tcp://btc.foundryusapool.com:25",
          "user": "soldierlakeimmersion.64cfd9473419",
          "password": "",
          "enabled": true
        },
        {
          "uid": "3",
          "url": "stratum+tcp://btc.foundryusapool.com:443",
          "user": "soldierlakeimmersion.64cfd9473419",
          "password": "",
          "enabled": true
        }
      ]
    }
  ],
  "tuner": {
    "enabled": true,
    "tunerMode": 1
  },
  "dps": {
    "enabled": true
  }
}
```

b-rowan, 2024-11-27:
Have you set a power limit on this miner, or its on default?

Awote, 2024-11-27:
[image]

Awote, 2024-11-27:
[image]

b-rowan, 2024-11-27:
It seems like the betterproto files I have compiled don't exactly match up with the result from that version, are you able to update that miner to latest and try again?  Seems like latest should be `24.09`.

b-rowan, 2024-11-27:
Correct.  pyasic doesn't support changing independent parts of the config, since most miner types require the entire config to be sent all at once.

b-rowan, 2024-11-27:
If for whatever reason you can't upgrade it, just grab the `data.wattage_limit` and set the power limit in the config to that.

closed 2024-12-04 by b-rowan

Awote, 2024-12-05:
```data.wattag_limit``` 
in miner's data is None