UpstreamData/pyasic #249: Pydantic Validation errors

b-rowan, 2024-12-09:
Upgraded and seeing a lot of errors e.g.
> pyasic.errors.APIError: Failed to call config on S19 XP (BOS+): <IP> while getting data.
> pydantic_core._pydantic_core.ValidationError: 1 validation error for MiningModeHashrateTune

_Originally posted by @eandersson in https://github.com/UpstreamData/pyasic/issues/238#issuecomment-2525362193_

eandersson, 2024-12-10:
https://github.com/UpstreamData/pyasic/blob/master/pyasic/config/mining/scaling.py#L113

```
Traceback (most recent call last):
  File "/home/user/pyasic/.venv/lib/python3.12/site-packages/pyasic/miners/base.py", line 524, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/pyasic/.venv/lib/python3.12/site-packages/pyasic/miners/backends/braiins_os.py", line 800, in get_config
    return MinerConfig.from_boser(grpc_conf)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/pyasic/.venv/lib/python3.12/site-packages/pyasic/config/__init__.py", line 232, in from_boser
    mining_mode=MiningModeConfig.from_boser(grpc_miner_conf),
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/pyasic/.venv/lib/python3.12/site-packages/pyasic/config/mining/__init__.py", line 622, in from_boser
    return cls.hashrate_tuning(
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/user/pyasic/.venv/lib/python3.12/site-packages/pyasic/config/base.py", line 71, in __call__
    return self.value(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/pyasic/.venv/lib/python3.12/site-packages/pydantic/main.py", line 212, in __init__
    validated_self = self.__pydantic_validator__.validate_python(data, self_instance=self)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
pydantic_core._pydantic_core.ValidationError: 1 validation error for MiningModeHashrateTune
scaling
  Input should be a valid dictionary or instance of ScalingConfig [type=model_type, input_value=None, input_type=NoneType]
    For further information visit https://errors.pydantic.dev/2.9/v/model_type

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/user/PycharmProjects/monitor/monitor/miners.py", line 193, in <module>
    asyncio.run(M.status_thread())
  File "/usr/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/user/PycharmProjects/monitor/monitor/miners.py", line 162, in status_thread
    result = await self.check_miners()
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/PycharmProjects/monitor/monitor/miners.py", line 111, in check_miners
    data = await self.get_miner_data(ip)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/PycharmProjects/monitor/monitor/miners.py", line 99, in get_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/pyasic/.venv/lib/python3.12/site-packages/pyasic/miners/base.py", line 567, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/pyasic/.venv/lib/python3.12/site-packages/pyasic/miners/base.py", line 526, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call config on S19 XP (BOS+): 192.168.0.200 while getting data.
```

b-rowan, 2024-12-10:
Ok, should be a pretty easy fix, 1 sec.

b-rowan, 2024-12-10:
Try with 0.65.1.

eandersson, 2024-12-10:
Yep that worked. Thanks!