UpstreamData/pyasic #250: Partially supported miner found: Antminer L9

Combinator78, 2024-12-10:
C:\vekus\agent2\local_libs\pyasic\miners\factory.py:875: UserWarning: Partially supported miner found: Antminer L9, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(
Traceback (most recent call last):
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 497, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\backends\bmminer.py", line 128, in _get_hashrate
    return self.algo.hashrate(
           ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'hashrate'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\vekus\agent2\test.py", line 17, in <module>
    asyncio.run(gather_miner_data())
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\test.py", line 11, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 540, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 499, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call hashrate on AntMiner (Stock): 172.19.151.100 while getting data.

b-rowan, 2024-12-10:
Try with 0.65.4?

Combinator78, 2024-12-10:
It works, thank you!

ip='172.19.151.68' device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.L9: 'L9'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.scrypt.ScryptAlgo'>) mac='7A:87:E0:F2:15:51' api_ver='3.1' fw_ver='Wed Aug 21 19:20:02 CST 2024' hostname='Antminer' expected_hashrate=16.5 GH/s expected_hashboards=3 expected_chips=330 expected_fans=4 env_temp=None wattage=None voltage=None fans=[Fan(speed=3420), Fan(speed=3390), Fan(speed=3420), Fan(speed=3440)] fan_psu=None hashboards=[HashBoard(slot=0, hashrate=5.42 GH/s, temp=33.0, chip_temp=48.0, chips=110, expected_chips=110, serial_number='DGAHYS8BDAAAA0669', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=5.43 GH/s, temp=34.5, chip_temp=50.5, chips=110, expected_chips=110, serial_number='DGAHYS8BDAAAA0C5H', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=5.43 GH/s, temp=34.5, chip_temp=50.5, chips=110, expected_chips=110, serial_number='DGAHYS8BDAAAA0C3R', missing=False, tuned=None, active=None, voltage=None)] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='scrypt.privatpools.com:54430', user='worker.671158fe62e78', password='SliL9050F'), Pool(url='', user='SliL9050F', password=''), Pool(url='', user='', password='')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=False errors=[] is_mining=True uptime=90103 pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='scrypt.privatpools.com', port=54430, pubkey=None), accepted=10376, rejected=255, get_failures=1, remote_failures=0, active=True, alive=True, index=0, user='worker.671158fe62e78', pool_rejected_percent=2.398645470792964, pool_stale_percent=0.009406452826639074), PoolMetrics(url=None, accepted=0, rejected=0, get_failures=0, remote_failures=0, active=True, alive=False, index=1, user='SliL9050F', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=None, accepted=0, rejected=0, get_failures=0, remote_failures=0, active=True, alive=False, index=2, user='', pool_rejected_percent=0, pool_stale_percent=0)] hashrate=16.28 GH/s wattage_limit=None total_chips=330 nominal=True percent_expected_chips=100 percent_expected_hashrate=99 percent_expected_wattage=None temperature_avg=34 efficiency=None datetime='2024-12-10T21:18:47.301725+03:00' timestamp=1733854727 make='AntMiner' model='L9' firmware='Stock' algo='Scrypt'
unit=GH/s rate=16.28

closed 2024-12-10 by b-rowan