UpstreamData/pyasic #251: Partially supported miner found: AVALONMINER 1126PRO

Combinator78, 2024-12-10:
C:\vekus\agent2\local_libs\pyasic\miners\factory.py:875: UserWarning: Partially supported miner found: AVALONMINER 1126PRO, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(
Traceback (most recent call last):
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 497, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\backends\avalonminer.py", line 187, in _get_hashrate
    return self.algo.hashrate(
           ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'hashrate'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\vekus\agent2\test.py", line 17, in <module>
    asyncio.run(gather_miner_data())
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\test.py", line 12, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 540, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 499, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call hashrate on AvalonMiner (Stock): 172.20.177.11 while getting data.

b-rowan, 2024-12-10:
Should be in with 0.65.6

Combinator78, 2024-12-10:
Unfortunately, I can't check - the client took these devices. But I have confidence in you, and I think you fixed it.

Combinator78, 2024-12-10:
**AvalonMiner 1246-N-90**

PS C:\vekus\agent2> py test.py
Traceback (most recent call last):
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 524, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\backends\cgminer.py", line 159, in _get_pools
    pool_url = PoolUrl.from_str(url) if url else None
               ^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\data\pools.py", line 43, in from_str
    return cls(scheme=scheme, host=host, port=port, pubkey=pubkey)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\site-packages\pydantic\main.py", line 214, in __init__
    validated_self = self.__pydantic_validator__.validate_python(data, self_instance=self)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
pydantic_core._pydantic_core.ValidationError: 1 validation error for PoolUrl
port
  Input should be a valid integer [type=int_type, input_value=None, input_type=NoneType]
    For further information visit https://errors.pydantic.dev/2.10/v/int_type

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\vekus\agent2\test.py", line 17, in <module>
    asyncio.run(gather_miner_data())
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\test.py", line 12, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 567, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 526, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call pools on Avalon 1246 (Stock): 172.20.160.71 while getting data.
PS C:\vekus\agent2>

b-rowan, 2024-12-10:
Ok, chip count is up now.  Should be good to go with 0.65.7.

Combinator78, 2024-12-10:
Everything is fine!

ip='172.19.155.21' device_info=DeviceInfo(make=<MinerMake.AVALONMINER: 'AvalonMiner'>, model=<AvalonminerModels.Avalon1126Pro: 'Avalon 1126 Pro'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.sha256.SHA256Algo'>) mac='B4:A2:EB:36:E8:67' api_ver='3.7' fw_ver='4.11.1' hostname=None expected_hashrate=0.0 TH/s expected_hashboards=3 expected_chips=360 expected_fans=4 env_temp=34.0 wattage=None voltage=None fans=[Fan(speed=6689), Fan(speed=207), Fan(speed=6769), Fan(speed=6707)] fan_psu=None hashboards=[HashBoard(slot=0, hashrate=5.44396 TH/s, temp=56, chip_temp=70, chips=120, expected_chips=120, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=6.532760000000001 TH/s, temp=55, chip_temp=74, chips=120, expected_chips=120, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=None, temp=None, chip_temp=None, chips=None, expected_chips=120, serial_number=None, missing=True, tuned=None, active=None, voltage=None)] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://ss.antpool.com:3333', user='Piter40.9x45', password='x'), Pool(url='stratum+tcp://ss.antpool.com:3333', user='Piter40.9x45', password='x'), Pool(url='stratum+tcp://ss.antpool.com:3333', user='Piter40.9x45', password='x')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=False errors=[] is_mining=True uptime=40 pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ss.antpool.com', port=3333, pubkey=None), accepted=1, rejected=0, get_failures=0, remote_failures=0, active=True, alive=True, index=0, user='Piter40.9x45', pool_rejected_percent=0.0, pool_stale_percent=0.0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ss.antpool.com', port=3333, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=1, user='Piter40.9x45', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ss.antpool.com', port=3333, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=2, user='Piter40.9x45', pool_rejected_percent=0, pool_stale_percent=0)] hashrate=11.97672 TH/s wattage_limit=3200 total_chips=240 nominal=False percent_expected_chips=67 percent_expected_hashrate=0 percent_expected_wattage=None temperature_avg=56 efficiency=None datetime='2024-12-10T22:32:31.794878+03:00' timestamp=1733859151 make='AvalonMiner' model='Avalon 1126 Pro' firmware='Stock' algo='SHA256'
unit=TH/s rate=11.97672

closed 2024-12-10 by b-rowan