UpstreamData/pyasic #253: on L3+ L7 does not detect expected_hashrate

Combinator78, 2024-12-10:
PS C:\vekus\agent2> py test.py
Traceback (most recent call last):
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 524, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\backends\antminer.py", line 322, in _get_expected_hashrate
    rate=float(expected_rate), unit=self.algo.unit.from_str(rate_unit)
                                    ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\enum.py", line 786, in __getattr__
    raise AttributeError(name) from None
AttributeError: from_str

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\vekus\agent2\test.py", line 17, in <module>
    asyncio.run(gather_miner_data())
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\test.py", line 12, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 567, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 526, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call expected_hashrate on L7 (Stock): 172.20.162.132 while getting data.
PS C:\vekus\agent2>

b-rowan, 2024-12-10:
Ok, try with 0.65.5.

Combinator78, 2024-12-10:
I checked, in the old version of the library the L3+ device also did not return expected_hashrate. it is not critical, you can leave it like that if it is not possible to get this parameter from this device

b-rowan, 2024-12-10:
> I checked, in the old version of the library the L3+ device also did not return expected_hashrate. it is not critical, you can leave it like that if it is not possible to get this parameter from this device

I will add it if you know where to source it, not sure I've ever seen expected hashrate values on older antminers.  The modern web UI (S19 style) exposes it in the API, but I'm not sure about these ones.

Combinator78, 2024-12-10:
Yeah, I don't think it matters. Let's close the ticket.

closed 2024-12-10 by b-rowan