UpstreamData/pyasic #254: Partially supported miner found: Antminer S19k Pro, type: MinerTypes.VNISH

Combinator78, 2024-12-10:
C:\vekus\agent2\local_libs\pyasic\miners\factory.py:877: UserWarning: Partially supported miner found: Antminer S19k Pro, type: MinerTypes.VNISH, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(
Traceback (most recent call last):
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 524, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\backends\vnish.py", line 216, in _get_hashrate
    return self.algo.hashrate(
           ^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'hashrate'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\vekus\agent2\test.py", line 17, in <module>
    asyncio.run(gather_miner_data())
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\test.py", line 12, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 567, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 526, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call hashrate on Unknown (VNish): 172.19.168.201 while getting data.

Combinator78, 2024-12-10:
Traceback (most recent call last):
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 524, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\backends\vnish.py", line 273, in get_config
    self.config = MinerConfig.from_vnish(web_settings, web_presets)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\config\__init__.py", line 254, in from_vnish
    mining_mode=MiningModeConfig.from_vnish(web_settings, web_presets),
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\config\mining\__init__.py", line 598, in from_vnish
    return MiningModePreset.from_vnish(mode_settings, web_presets)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\config\mining\__init__.py", line 367, in from_vnish
    available_presets=[MiningPreset.from_vnish(p) for p in web_presets],
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\config\mining\__init__.py", line 367, in <listcomp>
    available_presets=[MiningPreset.from_vnish(p) for p in web_presets],
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\config\mining\presets.py", line 28, in from_vnish
    return cls(
           ^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\site-packages\pydantic\main.py", line 214, in __init__
    validated_self = self.__pydantic_validator__.validate_python(data, self_instance=self)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
pydantic_core._pydantic_core.ValidationError: 1 validation error for MiningPreset
hashrate
  Input should be a valid integer, unable to parse string as an integer [type=int_parsing, input_value='120  LC', input_type=str]
    For further information visit https://errors.pydantic.dev/2.10/v/int_parsing

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\vekus\agent2\test.py", line 18, in <module>
    asyncio.run(gather_miner_data())
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Vekus\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\test.py", line 12, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 567, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\local_libs\pyasic\miners\base.py", line 526, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call config on S19 (VNish): 10.1.5.234 while getting data.
PS C:\vekus\agent2>

b-rowan, 2024-12-10:
> it works on the old version
> 
> MinerData(ip='10.1.5.234', datetime='2024-12-10T22:37:29.728691+03:00', timestamp=1733859449, device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.S19: 'S19'>, firmware=<MinerFirmware.VNISH: 'VNish'>, algo=SHA256Algo), make='AntMiner', model='S19', firmware='VNish', algo='SHA256', mac='02:C6:34:DE:00:E1', api_ver=None, fw_ver='1.2.1', hostname=None, hashrate=110.81417300000001 TH/s, expected_hashrate=112.73099 TH/s, expected_hashboards=3, expected_chips=228, expected_fans=4, percent_expected_chips=166, percent_expected_hashrate=98, percent_expected_wattage=83, temperature_avg=75, env_temp=None, wattage=2740, wattage_limit=3310, voltage=None, fans=[Fan(speed=2700), Fan(speed=2670), Fan(speed=2730), Fan(speed=2730)], fan_psu=None, hashboards=[HashBoard(slot=0, hashrate=35.99403 TH/s, temp=75, chip_temp=60, chips=126, expected_chips=76, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=36.480940000000004 TH/s, temp=75, chip_temp=60, chips=126, expected_chips=76, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=38.339203000000005 TH/s, temp=76, chip_temp=61, chips=126, expected_chips=76, serial_number=None, missing=False, tuned=None, active=None, voltage=None)], total_chips=378, nominal=False, config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='ss.antpool.com:3333', user='DenD1985.01', password='123'), Pool(url='btc.poolbinance.com:1800', user='DenD1985.01', password='123'), Pool(url='bs.poolbinance.com:3333', user='DenD1985.01', password='123')], quota=1, name='RJ3EB3')]), fan_mode=FanModeNormal(mode='normal', minimum_fans=4, minimum_speed=10), temperature=TemperatureConfig(target=75, hot=None, danger=None), mining_mode=MiningModePowerTune(mode='power_tuning', power=3310, algo=StandardTuneAlgo(mode='standard'), scaling=None)), fault_light=None, errors=[], is_mining=True, uptime=1689767, efficiency=25, pools=[]) 110.81417300000001 TH/s

This is due to a change with how hashrate is handled, this is a quick fix.  Ill get it fixed shortly.

b-rowan, 2024-12-10:
are you able to get the result of `miner.web.autotune_presets()`?

Combinator78, 2024-12-10:
Please tell me how to do this

Combinator78, 2024-12-10:
Traceback (most recent call last):
  File "C:\vekus\agent2\test.py", line 18, in <module>
    asyncio.run(gather_miner_data())
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User1\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\User1\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User1\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\vekus\agent2\test.py", line 12, in gather_miner_data
    miner_data = await miner.web.autotune_presets()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'AntminerOldWebAPI' object has no attribute 'autotune_presets'
PS C:\vekus\agent2>

b-rowan, 2024-12-10:
> Traceback (most recent call last): File "C:\vekus\agent2\test.py", line 18, in asyncio.run(gather_miner_data()) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "C:\Users\User1\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run return runner.run(main) ^^^^^^^^^^^^^^^^ File "C:\Users\User1\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run return self._loop.run_until_complete(task) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "C:\Users\User1\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete return future.result() ^^^^^^^^^^^^^^^ File "C:\vekus\agent2\test.py", line 12, in gather_miner_data miner_data = await miner.web.autotune_presets() ^^^^^^^^^^^^^^^^^^^^^^^^^^ AttributeError: 'AntminerOldWebAPI' object has no attribute 'autotune_presets' PS C:\vekus\agent2>

Is that on the right miner?

Combinator78, 2024-12-10:
it looks like i did something wrong. Can you tell me the correct code?

Combinator78, 2024-12-10:
ip='10.1.5.234' device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.S19: 'S19'>, firmware=<MinerFirmware.VNISH: 'VNish'>, algo=<class 'pyasic.device.algorithm.sha256.SHA256Algo'>) mac='02:C6:34:DE:00:E1' api_ver=None fw_ver='1.2.1' hostname=None expected_hashrate=112.73099 TH/s expected_hashboards=3 expected_chips=228 expected_fans=4 env_temp=None wattage=2737 voltage=None fans=[Fan(speed=2730), Fan(speed=2760), Fan(speed=2910), Fan(speed=2820)] fan_psu=None hashboards=[HashBoard(slot=0, hashrate=38.76615 TH/s, temp=76, chip_temp=61, chips=126, expected_chips=76, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=36.792277 TH/s, temp=75, chip_temp=60, chips=126, expected_chips=76, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=36.90773 TH/s, temp=76, chip_temp=61, chips=126, expected_chips=76, serial_number=None, missing=False, tuned=None, active=None, voltage=None)] config=None fault_light=None errors=[] is_mining=True uptime=1690782 pools=[PoolMetrics(url=None, accepted=84080, rejected=21, get_failures=None, remote_failures=None, active=False, alive=True, index=0, user='DenD1985.01', pool_rejected_percent=0.02496997657578388, pool_stale_percent=0), PoolMetrics(url=None, accepted=355, rejected=0, get_failures=None, remote_failures=None, active=True, alive=True, index=1, user='DenD1985.01', pool_rejected_percent=0.0, pool_stale_percent=0), PoolMetrics(url=None, accepted=25, rejected=0, get_failures=None, remote_failures=None, active=True, alive=True, index=2, user='DenD1985.01', pool_rejected_percent=0.0, pool_stale_percent=0), PoolMetrics(url=None, accepted=48538, rejected=63, get_failures=None, remote_failures=None, active=True, alive=True, index=3, user='DevFee', pool_rejected_percent=0.12962696240818092, pool_stale_percent=0)] hashrate=112.466157 TH/s wattage_limit=3310 total_chips=378 nominal=False percent_expected_chips=166 percent_expected_hashrate=100 percent_expected_wattage=83 temperature_avg=76 efficiency=24 datetime='2024-12-10T22:54:24.612273+03:00' timestamp=1733860464 make='AntMiner' model='S19' firmware='VNish' algo='SHA256'
unit=TH/s rate=112.466157
PS C:\vekus\agent2>

b-rowan, 2024-12-10:
```
import pyasic

import asyncio

IP = "10.1.5.234"

async def main():
    miner = await pyasic.get_miner(IP)
    print(await miner.web.autotune_presets())

if __name__ == "__main__":
    asyncio.run(main())

```

And if that doesn't work -

```
import pyasic
from pyasic.miners.antminer import VNishS19

import asyncio

IP = "10.1.5.234"

async def main():
    miner = VNishS19(IP)
    print(await miner.web.autotune_presets())

if __name__ == "__main__":
    asyncio.run(main())

```

Combinator78, 2024-12-10:
The first code doesn't work either.

b-rowan, 2024-12-10:
> ```
> import pyasic
>
> import asyncio
>
> IP = "10.1.5.234"
>
> async def main():
>     miner = await pyasic.get_miner(IP)
>     print(await miner.web.autotune_presets())
>
>
> if __name__ == "__main__": [...]

^

Combinator78, 2024-12-10:
PS C:\vekus\agent2> py test.py
[{'name': 'disabled', 'pretty': 'Disabled', 'status': 'untuned', 'modded_psu_required': False}, {'name': '1630', 'pretty': '1630 watt ~ 67 TH', 'status': 'untuned', 'modded_psu_required': False}, {'name': '1780', 'pretty': '1780 watt ~ 71 TH', 'status': 'untuned', 'modded_psu_required': False}, {'name': '1920', 'pretty': '1920 watt ~ 76 TH', 'status': 'untuned', 'modded_psu_required': False}, {'name': '2103', 'pretty': '2103 watt ~ 81 TH', 'status': 'untuned', 'modded_psu_required': False}, {'name': '2270', 'pretty': '2270 watt ~ 86 TH', 'status': 'untuned', 'modded_psu_required': False}, {'name': '2475', 'pretty': '2475 watt ~ 91 TH', 'status': 'untuned', 'modded_psu_required': False}, {'name': '2710', 'pretty': '2710 watt ~ 96 TH', 'status': 'untuned', 'modded_psu_required': False}, {'name': '2900', 'pretty': '2900 watt ~ 101 TH', 'status': 'untuned', 'modded_psu_required': False}, {'name': '3095', 'pretty': '3095 watt ~ 105 TH', 'status': 'untuned', 'modded_psu_required': False}, {'name': '3310', 'pretty': '3310 watt ~ 110 TH', 'status': 'tuned', 'modded_psu_required': False}, {'name': '3600', 'pretty': '3600 watt ~ 115 TH', 'status': 'untuned', 'modded_psu_required': False}, {'name': '3900', 'pretty': '3900 watt ~ 120 TH LC', 'status': 'untuned', 'modded_psu_required': False}, {'name': '4300', 'pretty': '4300 watt ~ 125 TH LC', 'status': 'untuned', 'modded_psu_required': False}, {'name': '4700', 'pretty': '4700 watt ~ 130 TH LC', 'status': 'untuned', 'modded_psu_required': False}]
PS C:\vekus\agent2>

Combinator78, 2024-12-10:
Sorry, I'm confused in the files here.

b-rowan, 2024-12-10:
Should be fixed with 0.65.8.

Combinator78, 2024-12-10:
One works, I'll check the second one now.

ip='10.1.5.234' device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.S19: 'S19'>, firmware=<MinerFirmware.VNISH: 'VNish'>, algo=<class 'pyasic.device.algorithm.sha256.SHA256Algo'>) mac='02:C6:34:DE:00:E1' api_ver=None fw_ver='1.2.1' hostname=None expected_hashrate=112.73099 TH/s expected_hashboards=3 expected_chips=228 expected_fans=4 env_temp=None wattage=2740 voltage=None fans=[Fan(speed=2850), Fan(speed=2880), Fan(speed=2940), Fan(speed=2850)] fan_psu=None hashboards=[HashBoard(slot=0, hashrate=37.09851999999999 TH/s, temp=76, chip_temp=61, chips=126, expected_chips=76, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=35.678082 TH/s, temp=75, chip_temp=60, chips=126, expected_chips=76, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=36.56441 TH/s, temp=76, chip_temp=61, chips=126, expected_chips=76, serial_number=None, missing=False, tuned=None, active=None, voltage=None)] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://ss.antpool.com:3333', user='DenD1985.01', password='123'), Pool(url='stratum+tcp://btc.poolbinance.com:1800', user='DenD1985.01', password='123'), Pool(url='stratum+tcp://bs.poolbinance.com:3333', user='DenD1985.01', password='123')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=4, minimum_speed=10), temperature=TemperatureConfig(target=75, hot=None, danger=None), mining_mode=MiningModePreset(mode='preset', active_preset=MiningPreset(name='3310', power=3310, hashrate=110, tuned=True, modded_psu=False), available_presets=[MiningPreset(name='disabled', power=None, hashrate=None, tuned=False, modded_psu=False), MiningPreset(name='1630', power=1630, hashrate=67, tuned=False, modded_psu=False), MiningPreset(name='1780', power=1780, hashrate=71, tuned=False, modded_psu=False), MiningPreset(name='1920', power=1920, hashrate=76, tuned=False, modded_psu=False), MiningPreset(name='2103', power=2103, hashrate=81, tuned=False, modded_psu=False), MiningPreset(name='2270', power=2270, hashrate=86, tuned=False, modded_psu=False), MiningPreset(name='2475', power=2475, hashrate=91, tuned=False, modded_psu=False), MiningPreset(name='2710', power=2710, hashrate=96, tuned=False, modded_psu=False), MiningPreset(name='2900', power=2900, hashrate=101, tuned=False, modded_psu=False), MiningPreset(name='3095', power=3095, hashrate=105, tuned=False, modded_psu=False), MiningPreset(name='3310', power=3310, hashrate=110, tuned=True, modded_psu=False), MiningPreset(name='3600', power=3600, hashrate=115, tuned=False, modded_psu=False), MiningPreset(name='3900', power=3900, hashrate=120, tuned=False, modded_psu=False), MiningPreset(name='4300', power=4300, hashrate=125, tuned=False, modded_psu=False), MiningPreset(name='4700', power=4700, hashrate=130, tuned=False, modded_psu=False)])) fault_light=None errors=[] is_mining=True uptime=1691647 pools=[PoolMetrics(url=None, accepted=84123, rejected=21, get_failures=None, remote_failures=None, active=False, alive=True, index=0, user='DenD1985.01', pool_rejected_percent=0.024957216200798632, pool_stale_percent=0), PoolMetrics(url=None, accepted=355, rejected=0, get_failures=None, remote_failures=None, active=True, alive=True, index=1, user='DenD1985.01', pool_rejected_percent=0.0, pool_stale_percent=0), PoolMetrics(url=None, accepted=25, rejected=0, get_failures=None, remote_failures=None, active=True, alive=True, index=2, user='DenD1985.01', pool_rejected_percent=0.0, pool_stale_percent=0), PoolMetrics(url=None, accepted=48546, rejected=63, get_failures=None, remote_failures=None, active=True, alive=True, index=3, user='DevFee', pool_rejected_percent=0.12960562858729863, pool_stale_percent=0)] hashrate=109.341012 TH/s wattage_limit=3310 total_chips=378 nominal=False percent_expected_chips=166 percent_expected_hashrate=97 percent_expected_wattage=83 temperature_avg=76 efficiency=25 datetime='2024-12-10T23:08:44.776211+03:00' timestamp=1733861324 make='AntMiner' model='S19' firmware='VNish' algo='SHA256'
unit=TH/s rate=109.341012

Combinator78, 2024-12-10:
The second one works too. Thank you, you are the best!

ip='172.19.168.201' device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.S19kPro: 'S19k Pro'>, firmware=<MinerFirmware.VNISH: 'VNish'>, algo=<class 'pyasic.device.algorithm.sha256.SHA256Algo'>) mac='02:C0:4B:64:70:FC' api_ver=None fw_ver='1.2.6-alpha2' hostname=None expected_hashrate=34.76319 TH/s expected_hashboards=3 expected_chips=231 expected_fans=4 env_temp=None wattage=0 voltage=None fans=[Fan(speed=1830), Fan(speed=1860), Fan(speed=2220), Fan(speed=2250)] fan_psu=None hashboards=[HashBoard(slot=0, hashrate=None, temp=26, chip_temp=11, chips=77, expected_chips=77, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=None, temp=None, chip_temp=None, chips=77, expected_chips=77, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=34.003809999999994 TH/s, temp=70, chip_temp=55, chips=77, expected_chips=77, serial_number=None, missing=False, tuned=None, active=None, voltage=None)] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://bch.viabtc.io:3333', user='tehno2023.F5', password='123'), Pool(url='stratum+tcp://bch.viabtc.io:25', user='tehno2023.F5', password='123'), Pool(url='stratum+tcp://bch.viabtc.io:44', user='tehno2023.F5', password='123')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=4, minimum_speed=10), temperature=TemperatureConfig(target=70, hot=None, danger=None), mining_mode=MiningModeManual(mode='manual', global_freq=505.0, global_volt=1300.0, boards={0: ManualBoardSettings(freq=0.0, volt=0.0), 1: ManualBoardSettings(freq=0.0, volt=0.0), 2: ManualBoardSettings(freq=0.0, volt=0.0)})) fault_light=None errors=[] is_mining=True uptime=366283 pools=[PoolMetrics(url=None, accepted=82967, rejected=6, get_failures=None, remote_failures=None, active=False, alive=True, index=0, user='tehno2023.F5', pool_rejected_percent=0.0072312680028442985, pool_stale_percent=0), PoolMetrics(url=None, accepted=0, rejected=0, get_failures=None, remote_failures=None, active=True, alive=True, index=1, user='tehno2023.F5', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=None, accepted=0, rejected=0, get_failures=None, remote_failures=None, active=False, alive=False, index=2, user='tehno2023.F5', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=None, accepted=3115, rejected=6, get_failures=None, remote_failures=None, active=True, alive=True, index=3, user='DevFee', pool_rejected_percent=0.19224607497596924, pool_stale_percent=0)] hashrate=34.003809999999994 TH/s wattage_limit=None total_chips=231 nominal=True percent_expected_chips=100 percent_expected_hashrate=98 percent_expected_wattage=None temperature_avg=48 efficiency=0 datetime='2024-12-10T23:16:09.433722+03:00' timestamp=1733861769 make='AntMiner' model='S19k Pro' firmware='VNish' algo='SHA256'
unit=TH/s rate=34.003809999999994

closed 2024-12-10 by b-rowan