UpstreamData/pyasic #258: Partially supported miner found: Antminer S19j XP

Combinator78, 2024-12-24:
factory.py:879: UserWarning: Partially supported miner found: Antminer S19j XP, type: MinerTypes.ANTMINER, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(
ip='172.16.23.36' device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=None) mac=None api_ver='3.1' fw_ver='Wed Nov 15 17:11:05 CST 2023' hostname=None expected_hashrate=None expected_hashboards=None expected_chips=0 expected_fans=None env_temp=None wattage=None voltage=None fans=[] fan_psu=None hashboards=[] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://gate.emcd.network:3333', user='destructivespiraldea.Zs19jxp151M5', password='x'), Pool(url='stratum+tcp://eu.emcd.network:3333', user='destructivespiraldea.Zs19jxp151M5', password='x'), Pool(url='stratum+tcp://kz.emcd.network:3333', user='destructivespiraldea.Zs19jxp151M5', password='x')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=None errors=[] is_mining=True uptime=3239 pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='gate.emcd.network', port=3333, pubkey=None), accepted=142, rejected=0, get_failures=0, remote_failures=0, active=True, alive=True, index=0, user='destructivespiraldea.Zs19jxp151M5', pool_rejected_percent=0.0, pool_stale_percent=0.0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='eu.emcd.network', port=3333, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=1, user='destructivespiraldea.Zs19jxp151M5', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='kz.emcd.network', port=3333, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=2, user='destructivespiraldea.Zs19jxp151M5', pool_rejected_percent=0, pool_stale_percent=0)] hashrate=None wattage_limit=None total_chips=None nominal=None percent_expected_chips=None percent_expected_hashrate=None percent_expected_wattage=None temperature_avg=None efficiency=None datetime='2024-12-24T10:06:30.054274+03:00' timestamp=1735023990 make='AntMiner' model=None firmware='Stock' algo=None

b-rowan, 2024-12-24:
Added in 0.67.4, just need chip count now.  It will warn again when you run it on the newest version and should tell you the chip count in data.

Combinator78, 2024-12-24:
Please upload the version 0.67.4

Combinator78, 2024-12-24:
2024-12-24 06:11:05 Chain[0]: find 110 asic, times 0
2024-12-24 06:11:07 Chain[1]: find 110 asic, times 0
2024-12-24 06:11:09 Chain[2]: find 110 asic, times 0

b-rowan, 2024-12-24:
Ok, should be good with 0.67.5, building now.

closed 2024-12-24 by b-rowan

Combinator78, 2024-12-24:
it works, thanks