UpstreamData/pyasic #269: Failed to get config on some Bitmain running BOS+

womj, 2025-01-08:
**Describe the bug**
Some T21 run BOS+ are failing scans due to being unable to scan config

**To Reproduce**

package versions
Python 3.12.7
```
aiofiles                       24.1.0      File support for asyncio.
annotated-types                0.7.0       Reusable constraint types to use...
anyio                          4.6.2.post1 High level compatibility layer f...
asyncssh                       2.18.0      AsyncSSH: Asynchronous SSHv2 cli...
aws-msk-iam-sasl-signer-python 1.0.1       Amazon MSK Library in Python for...
betterproto                    2.0.0b7     A better Protobuf / gRPC generat...
boto3                          1.35.76     The AWS SDK for Python
botocore                       1.35.76     Low-level, data-driven core of b...
certifi                        2024.8.30   Python package for providing Moz...
cffi                           1.17.1      Foreign Function Interface for P...
click                          8.1.7       Composable command line interfac...
confluent-kafka                2.6.1       Confluent's Python client for Ap...
cryptography                   44.0.0      cryptography is a package which ...
grpclib                        0.4.7       Pure-Python gRPC implementation ...
h11                            0.14.0      A pure-Python, bring-your-own-I/...
h2                             4.1.0       HTTP/2 State-Machine based proto...
hpack                          4.0.0       Pure-Python HPACK header compres...
httpcore                       1.0.7       A minimal low-level HTTP client.
httpx                          0.27.2      The next generation HTTP client.
hyperframe                     6.0.1       HTTP/2 framing layer for Python
idna                           3.10        Internationalized Domain Names i...
jmespath                       1.0.1       JSON Matching Expressions
multidict                      6.1.0       multidict implementation
passlib                        1.7.4       comprehensive password hashing f...
pika                           1.3.2       Pika Python AMQP Client Library
pyaml                          24.9.0      PyYAML-based module to produce a...
pyasic                         0.68.54     A simplified and standardized in...
pycparser                      2.22        C parser in Python
pydantic                       2.10.2      Data validation using Python typ...
pydantic-core                  2.27.1      Core functionality for Pydantic ...
python-dateutil                2.9.0.post0 Extensions to the standard Pytho...
python-dotenv                  1.0.1       Read key-value pairs from a .env...
pyyaml                         6.0.2       YAML parser and emitter for Python
redis                          5.2.0       Python client for Redis database...
s3transfer                     0.10.4      An Amazon S3 Transfer Manager
setuptools                     75.6.0      Easily download, build, install,...
six                            1.16.0      Python 2 and 3 compatibility uti...
sniffio                        1.3.1       Sniff out which async library yo...
tomli-w                        1.1.0       A lil' TOML writer
typing-extensions              4.12.2      Backported and Experimental Type...
urllib3                        2.2.3       HTTP library with thread-safe co...
```

```
python src/scan_single.py 10.46.3.50
BOSMinerT21
Error: Failed to call config on T21 (BOS+): 10.46.3.50 while getting data.
```

Note these miners are able to be scanned using `Python 3.12`
```
Name: pyasic
Version: 0.57.6
```

I get the same error on some S19j Pro BOS+
```
python src/scan_single.py 10.11.2.61
BOSMinerS19jPro
Error: Failed to call config on S19j Pro (BOS+): 10.11.2.61 while getting data.
```

**Expected behavior**
I expect to get scan data for the miner

**Desktop (please complete the following information):**
LSB Version:	core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID:	Pop
Description:	Pop!_OS 22.04 LTS
Release:	22.04
Codename:	jammy

**Miner Information (If applicable):**
 - Manufacturer: Bitmain
 - Type: T21
 - Firmware Type: BOS+
 - Firmware Version: 2024-12-03-0-f4c282cd-24.09.1-plus

**Additional context**
Thank you :smile:

b-rowan, 2025-01-08:
Thanks for the nice bug report!  Ill see what I can do.

Do you happen to have any more information on why its failing to call the config method?  You can try calling `await miner.get_config()` directly and maybe it can give a more useful traceback...

b-rowan, 2025-01-08:
The biggest issue here is that I don't have a copy of this miner to test with, so it can be hard for me to replicate these and be able to find the issue...

womj, 2025-01-09:
If you want we could have a call about potentially giving remote access to part of our fleet as part of supporting your guys development? @b-rowan

b-rowan, 2025-01-09:
I'm not opposed to it, you can send me an email, or reach out on telegram (handle is b_rowan) if you prefer.  There is a good possibility someone I know has one of these that I could test on too, I can ask around.

b-rowan, 2025-02-03:
Give latest a shot, I have a S21 pro that I was testing with now and things should work better now possibly.

womj, 2025-02-26:
new version works

closed 2025-02-26 by womj