UpstreamData/pyasic #270: Vnish "Partially supported miner" warning caused by devfee hotfix

wilfredallyn, 2025-01-14:
After installing the [vnish hotfix](https://t.me/AsicTo/1/82606) to address the [devfee issue](https://t.me/AsicTo/82794), I'm getting this warning

```
/Documents/pyasic/pyasic/miners/factory.py:1045: UserWarning: Partially supported miner found: Antminer S19j Pro A, type: MinerTypes.VNISH, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
```

This issue causes problems with [hass-miner](https://github.com/Schnitzel/hass-miner). When I add the miner to home assistant, it creates this error in the logs

```
2025-01-14 10:20:08.316 WARNING (MainThread) [py.warnings] /home/vscode/.local/ha-venv/lib/python3.13/site-packages/pyasic/miners/factory.py:1045: UserWarning: Partially supported miner found: Antminer S19j Pro A, type: MinerTypes.VNISH, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(

2025-01-14 10:20:08.372 DEBUG (MainThread) [custom_components.miner.coordinator] Found miner: Unknown (VNish): 192.168.1.184
2025-01-14 10:20:08.490 ERROR (MainThread) [custom_components.miner.coordinator] Failed to call fans on Unknown (VNish): 192.168.1.184 while getting data.
Traceback (most recent call last):
  File "/home/vscode/.local/ha-venv/lib/python3.13/site-packages/pyasic/miners/base.py", line 525, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/vscode/.local/ha-venv/lib/python3.13/site-packages/pyasic/miners/backends/bmminer.py", line 211, in _get_fans
    fans = [Fan() for _ in range(self.expected_fans)]
                           ~~~~~^^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object cannot be interpreted as an integer

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/workspaces/core/config/custom_components/miner/coordinator.py", line 91, in _async_update_data
    miner_data = await self.miner.get_data(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<13 lines>...
    )
    ^
  File "/home/vscode/.local/ha-venv/lib/python3.13/site-packages/pyasic/miners/base.py", line 568, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
        allow_warning=allow_warning, include=include, exclude=exclude
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/home/vscode/.local/ha-venv/lib/python3.13/site-packages/pyasic/miners/base.py", line 527, in _get_data
    raise APIError(
        f"Failed to call {data_name} on {self} while getting data."
    ) from e
pyasic.errors.APIError: Failed to call fans on Unknown (VNish): 192.168.1.184 while getting data.
2025-01-14 10:20:08.495 DEBUG (MainThread) [custom_components.miner.coordinator] Finished fetching s19jp data in 0.178 seconds (success: False)
```

Should any logic be added to `MinerFactory` to handle this type of situation? If you point me in the right correction, I can look into fixing it.

b-rowan, 2025-01-14:
Just need to combine them in `miners.antminer.vnish.x19`, make sure they're imported in that init (it's all star inputs all the way up from there), then add the model string in all caps to the MINER_CLASSES dict in factory.

b-rowan, 2025-01-14:
This commit is a good example of what needs to be done (https://github.com/UpstreamData/pyasic/commit/8e15b00e704f526e06728b9d55ab95b250dee04e).  If you ignore the docs files, you can see that the miner model is defined, then imported to combine it with the stock Antminer backend, then imported a layer up, then added to the factory.  The only difference here is (I think) that the model hardware has already been defined, so it's just combining the hardware info with the Vnish backend and loading it in the factory.

PR #273 (2025-01-15): feature: add support for Vnish S19j Pro A
Fixes  #270

> Just need to combine them in miners.antminer.vnish.x19, make sure they're imported in that init (it's all star inputs all the way up from there), then add the model string in all caps to the MINER_CLASSES dict in factory.

I realized that I only needed to update MINER_CLASSES dict in factory and didn't need to add anything to `miners.antminer.vnish.x19` since is the miner is a `VNishS19jPro`

I saw that  [`ANTMINER S19J PRO BB`](https://github.com/UpstreamData/pyasic/commit/c19945bb82dd99b64cefcddfa1329520d4ff47df) had been added similar [... cut]
closed 2025-01-15 by b-rowan