UpstreamData/pyasic #271: Antminer KS5 and KS5pro are defined as IceRiver

Combinator78, 2025-01-15:
**Antminer KS5**
ip='172.20.169.2' device_info=DeviceInfo(make=<MinerMake.ICERIVER: 'IceRiver'>, model=<IceRiverModels.KS5: 'KS5'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.kheavyhash.KHeavyHashAlgo'>) mac='DC:03:40:92:6D:39' api_ver='3.1' fw_ver='Sat Jul 27 15:33:14 CST 2024' hostname='Antminer' sticker_hashrate=None expected_hashrate=20.0 TH/s expected_hashboards=3 expected_chips=276 expected_fans=4 env_temp=None wattage=None voltage=None fans=[Fan(speed=2470), Fan(speed=2470), Fan(speed=2500), Fan(speed=2540)] fan_psu=None hashboards=[HashBoard(slot=0, hashrate=6.5281899999999995 TH/s, temp=50.5, chip_temp=71.5, chips=92, expected_chips=92, serial_number='JYZZYRRBDJCBG0AM5', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=6.66546 TH/s, temp=52.0, chip_temp=70.5, chips=92, expected_chips=92, serial_number='JYZZYRRBDJCBG0AM6', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=6.71921 TH/s, temp=52.0, chip_temp=72.0, chips=92, expected_chips=92, serial_number='JYZZYRRBDJCBG0AF4', missing=False, tuned=None, active=None, voltage=None)] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://mining.viabtc.io:3015', user='KASPAPA.F20TKS5x018', password='123'), Pool(url='stratum+tcp://mining.viabtc.io:315', user='KASPAPA.F20TKS5x018', password='123'), Pool(url='stratum+tcp://mining.viabtc.io:3015', user='KASPAPA.F20TKS5x018', password='123')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=False errors=[] is_mining=True uptime=4217208 pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='mining.viabtc.io', port=3015, pubkey=None), accepted=1185494, rejected=8280, get_failures=23, remote_failures=0, active=True, alive=True, index=0, user='KASPAPA.F20TKS5x018', pool_rejected_percent=0.6935986208444814, pool_stale_percent=0.001926662835679115), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='mining.viabtc.io', port=315, pubkey=None), accepted=1946, rejected=56, get_failures=48, remote_failures=0, active=True, alive=True, index=1, user='KASPAPA.F20TKS5x018', pool_rejected_percent=2.797202797202797, pool_stale_percent=2.3976023976023977), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='mining.viabtc.io', port=3015, pubkey=None), accepted=120, rejected=48, get_failures=53, remote_failures=0, active=True, alive=True, index=2, user='KASPAPA.F20TKS5x018', pool_rejected_percent=28.57142857142857, pool_stale_percent=31.547619047619047)] hashrate=19.912860000000002 TH/s wattage_limit=None total_chips=276 nominal=True percent_expected_chips=100 percent_expected_hashrate=100 percent_expected_wattage=None temperature_avg=52 efficiency=None datetime='2025-01-15T10:54:30.902328+03:00' timestamp=1736927670 make='IceRiver' model='KS5' firmware='Stock' algo='KHeavyHash'
unit=TH/s rate=19.912860000000002

**Antminer KS5pro**
ip='172.20.169.98' device_info=DeviceInfo(make=<MinerMake.ICERIVER: 'IceRiver'>, model=<IceRiverModels.KS5: 'KS5'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.kheavyhash.KHeavyHashAlgo'>) mac='12:FF:8C:C7:DF:DE' api_ver='3.1' fw_ver='Sat Nov 16 20:49:39 CST 2024' hostname='Antminer' sticker_hashrate=None expected_hashrate=21.0 TH/s expected_hashboards=3 expected_chips=276 expected_fans=4 env_temp=None wattage=None voltage=None fans=[Fan(speed=3030), Fan(speed=3020), Fan(speed=3000), Fan(speed=3000)] fan_psu=None hashboards=[HashBoard(slot=0, hashrate=7.14328 TH/s, temp=45.5, chip_temp=61.5, chips=92, expected_chips=92, serial_number='JYZZYRRBDAAJH00N7', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=7.1658599999999995 TH/s, temp=49.0, chip_temp=67.0, chips=92, expected_chips=92, serial_number='JYZZYRRBDAAJH048J', missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=7.15017 TH/s, temp=44.5, chip_temp=61.5, chips=92, expected_chips=92, serial_number='JYZZYRRBDAAJH04CV', missing=False, tuned=None, active=None, voltage=None)] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://mining.viabtc.io:3015', user='KASPAPA.F21TKS5prox007', password='123'), Pool(url='stratum+tcp://mining.viabtc.io:315', user='KASPAPA.F21TKS5prox007', password='123'), Pool(url='', user='', password='')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=False errors=[] is_mining=True uptime=1351082 pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='mining.viabtc.io', port=3015, pubkey=None), accepted=393104, rejected=481, get_failures=2, remote_failures=0, active=True, alive=True, index=0, user='KASPAPA.F21TKS5prox007', pool_rejected_percent=0.12220994194392572, pool_stale_percent=0.0005081494467522898), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='mining.viabtc.io', port=315, pubkey=None), accepted=288, rejected=0, get_failures=14, remote_failures=0, active=True, alive=True, index=1, user='KASPAPA.F21TKS5prox007', pool_rejected_percent=0.0, pool_stale_percent=4.861111111111112), PoolMetrics(url=None, accepted=0, rejected=0, get_failures=0, remote_failures=0, active=True, alive=False, index=2, user='', pool_rejected_percent=0, pool_stale_percent=0)] hashrate=21.45931 TH/s wattage_limit=None total_chips=276 nominal=True percent_expected_chips=100 percent_expected_hashrate=102 percent_expected_wattage=None temperature_avg=46 efficiency=None datetime='2025-01-15T10:57:29.930037+03:00' timestamp=1736927849 make='IceRiver' model='KS5' firmware='Stock' algo='KHeavyHash'
unit=TH/s rate=21.45931

b-rowan, 2025-01-15:
Is there any way in the API to differentiate the KS5 and KS5 pro? They seem to be reporting the same values for the type...

Combinator78, 2025-01-16:
I think so, they both report Antminer KS5.

b-rowan, 2025-01-16:
Not entirely sure that I will be able to distinguish them then...  They should be properly identified in latest, but the submodel may be wrong...

closed 2025-01-16 by b-rowan