UpstreamData/pyasic #278: Support for Lucky Miner LV08

cannockwolf, 2025-01-17:
Is anybody looking to / able to add the Lucky Miner LV08 to the integration, my bitaxe works but not this yet it seems to use the same Web OS.

b-rowan, 2025-01-17:
Ok, looking into this a bit more, I think the main things will be the index file (the data served at http://ip/), which you can get with `curl http://ip/` in command prompt or terminal, and the data from the `/api/system/info` endpoint.

cannockwolf, 2025-01-18:
i got the attached file from the curl command

and this from the link

> {
	"power":	99.80615234375,
	"voltage":	12140.625,
	"current":	20750,
	"temp":	55,
	"vrTemp":	52,
	"hashRate":	3587.3619047440029,
	"bestDiff":	"618M",
	"bestSessionDiff":	"67.8M",
	"isUsingFallbackStratum":	0,
	"freeHeap":	8421064,
	"coreVoltage":	1200,
	"coreVoltageActual":	1182,
	"frequency":	485,
	"ssid":	"Sith Lord",
	"macAddr":	"E4:B0:63:B3:25:BC",
	"hostname":	"espressif",
	"wifiStatus":	"Connected!",
	"sharesAccepted":	57757,
	"sharesRejected":	66,
	"uptimeSeconds":	148893,
	"asicCount":	9,
	"asicDetected":	9,
	"smallCoreCount":	894,
	"ASICModel":	"BM1366",
	"minerModel":	"LV08",
	"stratumURL":	"solo.solohash.co.uk",
	"fallbackStratumURL":	"solo.ckpool.org",
	"stratumPort":	3333,
	"fallbackStratumPort":	3333,
	"stratumUser":	"bc1qckuqlc2jaj9sr8wfxpfk2ethplcvehwdcrpfzy.cannockwolf",
	"fallbackStratumUser":	"bc1qckuqlc2jaj9sr8wfxpfk2ethplcvehwdcrpfzy.cannockwolf",
	"version":	"1.0.0",
	"boardVersion":	"402",
	"runningPartition":	"factory",
	"flipscreen":	1,
	"overheat_mode":	0,
	"invertscreen":	0,
	"invertfanpolarity":	1,
	"autofanspeed":	0,
	"fanspeed":	100,
	"fanrpm":	3160,
	"fanSpeed":	3160,
	"allowUpdate":	0,
	"initMiner":	"lucky"
}

cannockwolf, 2025-01-18:
[luckyminercurl.txt](https://github.com/user-attachments/files/18464340/luckyminercurl.txt)

b-rowan, 2025-01-18:
Perfect, let me see what I can do.

b-rowan, 2025-01-20:
Done.  Should be working in 0.70.2, now just need to update hass-miner.

closed 2025-01-20 by b-rowan

cannockwolf, 2025-01-21:
I have updated the HA version to v1.1.24 and it still says "Unable to connect to Miner, is IP correct?" 

I have definitely got the correct IP