UpstreamData/pyasic #284: Support for Innosilicon A6 LTC Master

JabeBRD, 2025-01-28:
Hello.
Would it be possible to add support for older miner Innosilicon A6 LTC Master?

b-rowan, 2025-02-03:
Should be.  Can you send the result of `await pyasic.miner_factory.get_miner_model_innosilicon(ip)`?

JabeBRD, 2025-02-03:
Yes, if i get little help...
Im using this in Home Assistant HASS-Miner integration. But i think i cant run that command in HA integration?
Need to install this in virtual machine?

JabeBRD, 2025-02-03:
No, it just says that it cant connect.
This is from miner system overview
```
Miner Type | 17
Hw Version | G19
MAC Addr | 50:6C:BE:09:D4:0E
Build Date | 2018-06-22 14:32:26
Platform Version | 05ce00409af97f6a75ee5de76f761964a406de7f.0
Rootfs Version | 9948c6da21ea76466d0ff30e2e7ab753db2d7159
```

JabeBRD, 2025-02-03:
I managed to install `pyasic` in windows11 powershell using `poetry`, but i have no clue how to run `pyasic` from `poetry` enviroment.
After it has installed, what should i type in? I have never used python before... i can find directories and files in powershell but thats about it.

b-rowan, 2025-02-03:
> I managed to install `pyasic` in windows11 powershell using `poetry`, but i have no clue how to run `pyasic` from `poetry` enviroment. After it has installed, what should i type in? I have never used python before... i can find directories and files in powershell but thats about it.

Are you able to use the browser devtools to get outputs of endpoints while logged in?  Looking for `/overview` and `/api/type` outputs, that might be easier...

b-rowan, 2025-02-03:
Ok, thats a completely new web UI...  That will make things quite difficult.