UpstreamData/pyasic #292: Does not change the configuration of pools in Whatsminer

Combinator78, 2025-02-07:
stderr: C:\Vekus\agent\./local_libs\pyasic\rpc\base.py:228 in _send_bytes
[WARNING][02/07/25 11:36:53](root) - BTMinerRPCAPI: 10.1.6.41 - ([Hidden] Send Bytes) - Read timeout expired.

stderr: C:\Vekus\agent\./local_libs\pyasic\rpc\base.py:228 in _send_bytes
[WARNING][02/07/25 11:37:04](root) - BTMinerRPCAPI: 10.1.6.41 - ([Hidden] Send Bytes) - Read timeout expired.

commandData in configure_miner() {'id': 13269, 'ip': '10.1.6.41', 'owner_id': 488, 'pool_1_url': 'btc.viabtc.io:3333', 'pool_2_url': 'stratum+tcp://btc.ss.poolin.com:443', 'pool_3_url': 'stratum+tcp://btc.ss.poolin.com:700', 'pool_1_user': 'VMC2.DOLG.13269', 'pool_2_user': 'nikitosart1994.M50122t35', 'pool_3_user': 'nikitosart1994.M50122t35', 'pool_1_password': '', 'pool_2_password': '', 'pool_3_password': ''}

config from get_config() pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://btc.ss.poolin.com:443', user='nikitosart1994.M50122t35', password='x'), Pool(url='stratum+tcp://btc.ss.poolin.com:700', user='nikitosart1994.M50122t35', password='x'), Pool(url='stratum+tcp://btc.ss.poolin.com:1883', user='nikitosart1994.M50122t35', password='x')], quota=1, name=None)]) fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0) temperature=TemperatureConfig(target=None, hot=None, danger=None) mining_mode=MiningModePowerTune(mode='power_tuning', power=3500, algo=StandardTuneAlgo(mode='standard'), scaling=None)

updatedConfig pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='btc.viabtc.io:3333', user='VMC2.DOLG.13269', password='x'), Pool(url='stratum+tcp://btc.ss.poolin.com:443', user='nikitosart1994.M50122t35', password='x'), Pool(url='stratum+tcp://btc.ss.poolin.com:700', user='nikitosart1994.M50122t35', password='x')], quota=1, name='default')]) fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0) temperature=TemperatureConfig(target=None, hot=None, danger=None) mining_mode=MiningModePowerTune(mode='power_tuning', power=3500, algo=StandardTuneAlgo(mode='standard'), scaling=None)

closed 2025-02-07 by Combinator78

b-rowan, 2025-02-07:
Temporary bug I assume?

Combinator78, 2025-02-07:
I think so, after some time the settings changed. I'll keep an eye on it some more.