UpstreamData/pyasic #294: Issue with Device Identification for S19 on Stock Firmware

olegudff, 2025-02-13:
Hello,

I'm writing to report a bug related to the identification of S19 devices when using the library. Below are the details:

---

**Describe the Bug**
When polling Bitmain S19 devices running stock firmware, the device is periodically misidentified as "BHB42XXX" instead of "S19". This misidentification prevents me from retrieving data and managing the device properly.

---

**To Reproduce**
Steps to reproduce the behavior:

- Use a Bitmain S19 device with stock firmware (Firmware Version: Mon Dec 26 17:10:01 CST 2022).
- Start polling the device with the library.
- After some time, observe that the device is identified as "BHB42XXX" instead of "S19".
- Notice the following log message:
```
Partially supported miner found: Antminer BHB42XXX, type: MinerTypes.ANTMINER, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
```

---

**Expected Behavior**
The device should consistently be recognized as "S19", allowing full access for data retrieval and management.

---

**Miner Information**

- Manufacturer: Bitmain
- Type: S19
- Firmware Type: Stock
- Firmware Version: Mon Dec 26 17:10:01 CST 2022

---

**Additional Context**
Temporarily, the issue can be resolved by reflashing the device with the same firmware, after which it is correctly identified as S19. However, after an undefined period, the device reverts to being recognized as "BHB42XXX". Additionally, if the device is manually woken from sleep, it temporarily returns to being identified as S19, but the problem reoccurs over time. This problem appears to be widespread.

Could you please advise on how I can manage the S19 device even if it is identified as BHB42XXX? Is there a workaround or a fix planned for this bug?

Thank you for your time and support.

b-rowan, 2025-02-13:
I think this is an issue with the miner itself, but I'm not sure what exactly would cause this.  I'm guessing its failing to read from the EEPROM or something.  The closest thing ive seen is this - https://github.com/UpstreamData/pyasic/blob/4d7a13433b98762e289424467f95118f8390e983/pyasic/miners/factory.py#L121-L122

I'm not sure if I can actually implement a fix for this until there is a way to verify that this happens on only 1 X19 model, otherwise I would rather have it warn since it is a miner issue.

Also, you may want to try updating the firmware on that miner (or even better, swap to some aftermarket firmware) and see if the problem persists.

olegudff, 2025-02-17:
Indeed, this issue seems to be related to the device and its firmware rather than the pyasic library.

I've noticed that the problem temporarily goes away when updating to the stock firmware, only to reappear later. I haven't pinpointed the exact cause or pattern yet—although I often put the devices to sleep for power control. Flashing VNISH or a similar custom firmware completely resolves the issue, but that's not always a viable solution since not every device can be flashed with custom firmware.

So, am I correct in understanding that I can fork the project and try to add `"ANTMINER BHB42XXX": BMMinerX19` in `miners/factory.py`?

b-rowan, 2025-02-17:
> So, am I correct in understanding that I can fork the project and try to add "ANTMINER BHB42XXX": BMMinerX19 in miners/factory.py?

You absolutely could, that's why it's open source.  My concern with upstreaming it is the risk of mis-identifying devices...  is there any chance that updating the stock firmware to the latest version will fix it?

olegudff, 2025-02-17:
I've tried different firmware versions, including the latest one.

The issue becomes widespread if you first put the device into sleep mode and then try to wake it up. I'm going to experiment further.

olegudff, 2025-02-17:
Once I have the experimental results, I'll post them here again. Thank you!
Please do not close the issue for at least a week.

s0kil, 2025-03-10:
A good compromise is to use the 6060 API (specifically `/productName` endpoint), it's fast (no auth) and the port is open even if BMMiner is unavailable.

https://gist.github.com/s0kil/abfafa77a7c208d44f1156279428329a