UpstreamData/pyasic #295: Bitmain S21 Hydro Miners Inlet and Outlet Temperatures

cryptographicturk, 2025-02-13:
A critical measure of temperature on hydro miners is the inlet vs outlet temperature. When the differential between these 2 numbers reaches a threshold they shut down. The information is reported in the stats.cgi / cgminer STATS as temp_chip. Currently PyAsic averages the 2 numbers, but ideally we need these numbers exposed.

The following is exposed for each hashboard in the chain:

`"STATS": [
    "chain" : [
         {
              "index": <board index>
              "temp_chip": [
                   <inlet temperature>,
                   <outlet temperature>,
                   0,
                   0
         }
    ]
]
`

I would like these inlet and outlet temperatures to be exposed in the hashboard record.

b-rowan, 2025-02-13:
This seems like a good idea, but I'm not sure how to handle this in comparison to other miners, any ideas on how to format this data?

cryptographicturk, 2025-02-14:
I think adding `inlet_temp` and `outlet_temp` to the hashboards makes the most sense. I took a look at an air cooled T21 as well as an S19J Pro, and you can see the same parameters are used, just differently. The first 2 are inlet, and the second 2 are outlet. It's inconvenient that Bitmain is differentiating this by model.

ildartaipov, 2025-03-26:
Additionally, ```S21+ hydro``` transmits temperature data as follows:
```
{
...
    "STATS": [
        {
...
            "chain": [
                {
...
                    "temp_pic": [
                        <chip temperature>,
                        <PCB[0]>,
                        <PCB[1]>,
                        <PCB[2]>
                    ],
                    "temp_pcb": [
                        <inlet temperature>,
                        <PCB[3]>,
                        <outlet temperature>,
                        <PCB[4]>
                    ],
                    "temp_chip": [
                        0,
                        0,
                        0,
                        0
                    ],
                   ...
                }
                ...
            ]
        }
    ]
}
```