UpstreamData/pyasic #297: L3+ errors in data retrieval

Combinator78, 2025-02-17:
in version v0.71.12 devices are no longer detected L3+

Traceback (most recent call last):
  File "C:\vekus\agent\test.py", line 17, in <module>
    asyncio.run(gather_miner_data())
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\vekus\agent\test.py", line 12, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent\local_libs\pyasic\miners\base.py", line 566, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent\local_libs\pyasic\miners\base.py", line 489, in _get_data
    await asyncio.gather(rpc_command_task, web_command_task)
  File "C:\vekus\agent\local_libs\pyasic\web\mskminer.py", line 39, in multicommand
    await asyncio.gather(*[t for t in tasks.values()])
  File "C:\vekus\agent\local_libs\pyasic\web\mskminer.py", line 72, in info_v1
    return await self.send_command("info_v1")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent\local_libs\pyasic\web\mskminer.py", line 65, in send_command
    raise APIError(f"Command failed: {command}")
pyasic.errors.APIError: Command failed: info_v1

b-rowan, 2025-03-03:
Looks like its mis-identifying as an MSKMiner base...

Combinator78, 2025-04-01:
Is there anything that can be done about this? It still doesn't work.
Traceback (most recent call last):
  File "C:\vekus\agent\test.py", line 17, in <module>
    asyncio.run(gather_miner_data())
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\vekus\agent\test.py", line 12, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent\local_libs\pyasic\miners\base.py", line 566, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent\local_libs\pyasic\miners\base.py", line 489, in _get_data
    await asyncio.gather(rpc_command_task, web_command_task)
  File "C:\vekus\agent\local_libs\pyasic\web\mskminer.py", line 39, in multicommand
    await asyncio.gather(*[t for t in tasks.values()])
  File "C:\vekus\agent\local_libs\pyasic\web\mskminer.py", line 72, in info_v1
    return await self.send_command("info_v1")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent\local_libs\pyasic\web\mskminer.py", line 65, in send_command
    raise APIError(f"Command failed: {command}")
pyasic.errors.APIError: Command failed: info_v1
PS C:\vekus\agent>

b-rowan, 2025-04-01:
Try with 0.72.4

Combinator78, 2025-04-01:
ip='172.16.26.16' device_info=DeviceInfo(make=<MinerMake.ANTMINER: 'AntMiner'>, model=<AntminerModels.L3Plus: 'L3+'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.scrypt.ScryptAlgo'>) mac='90:70:65:B8:9A:89' api_ver='3.1' fw_ver='4.9.0 rwglr' hostname=None sticker_hashrate=None expected_hashrate=None expected_hashboards=4 expected_chips=288 expected_fans=2 env_temp=None wattage=None voltage=None fans=[Fan(speed=0), Fan(speed=0)] fan_psu=None hashboards=[HashBoard(slot=0, hashrate=128.58 GH/s, temp=52, chip_temp=43, chips=72, expected_chips=72, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=1, hashrate=129.19 GH/s, temp=48, chip_temp=39, chips=72, expected_chips=72, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=2, hashrate=128.7 GH/s, temp=45, chip_temp=39, chips=72, expected_chips=72, serial_number=None, missing=False, tuned=None, active=None, voltage=None), HashBoard(slot=3, hashrate=128.71 GH/s, temp=47, chip_temp=41, chips=72, expected_chips=72, serial_number=None, missing=False, tuned=None, active=None, voltage=None)] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='ltc.trustpool.ru:443', user='first1vadim.A001', password='123'), Pool(url='ltc.trustpool.ru:25', user='first1vadim.A001', password='123'), Pool(url='ltc.trustpool.ru:3333', user='first1vadim.A001', password='123')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=None errors=[] is_mining=True uptime=18170 pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=443, pubkey=None), accepted=2115, rejected=4, get_failures=0, remote_failures=0, active=True, alive=True, index=0, user='first1vadim.A001', pool_rejected_percent=0.18876828692779613, pool_stale_percent=0.0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=25, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=1, user='first1vadim.A001', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='ltc.trustpool.ru', port=3333, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=2, user='first1vadim.A001', pool_rejected_percent=0, pool_stale_percent=0), PoolMetrics(url=None, accepted=350, rejected=10, get_failures=6, remote_failures=0, active=True, alive=True, index=3, user='*', pool_rejected_percent=2.7777777777777777, pool_stale_percent=1.6666666666666667)] hashrate=515.18 GH/s wattage_limit=None total_chips=288 nominal=True percent_expected_chips=100 percent_expected_hashrate=None percent_expected_wattage=None temperature_avg=48 efficiency=None datetime='2025-04-01T22:50:46.501894+03:00' timestamp=1743537046 make='AntMiner' model='L3+' firmware='Stock' algo='Scrypt'
unit=GH/s rate=515.18
PS C:\vekus\agent> py test.py
Traceback (most recent call last):
  File "C:\vekus\agent\test.py", line 17, in <module>
    asyncio.run(gather_miner_data())
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\nomed\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "C:\vekus\agent\test.py", line 12, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent\local_libs\pyasic\miners\base.py", line 566, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent\local_libs\pyasic\miners\base.py", line 489, in _get_data
    await asyncio.gather(rpc_command_task, web_command_task)
  File "C:\vekus\agent\local_libs\pyasic\web\mskminer.py", line 39, in multicommand
    await asyncio.gather(*[t for t in tasks.values()])
  File "C:\vekus\agent\local_libs\pyasic\web\mskminer.py", line 72, in info_v1
    return await self.send_command("info_v1")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\vekus\agent\local_libs\pyasic\web\mskminer.py", line 65, in send_command
    raise APIError(f"Command failed: {command}")
pyasic.errors.APIError: Command failed: info_v1

closed 2025-08-16 by Combinator78