UpstreamData/pyasic #301: Latest pyasic version (0.71.12) fails with get_data on Avalon Nano 3

hanleyp, 2025-02-23:
Traceback (most recent call last):
  File "/Users/phanley/Source/miningprof/main.py", line 211, in <module>
    miners = asyncio.run(gather_miner_data(hash_price))
  File "/opt/homebrew/Cellar/python@3.13/3.13.0_1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ~~~~~~~~~~^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.0_1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.0_1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/asyncio/base_events.py", line 721, in run_until_complete
    return future.result()
           ~~~~~~~~~~~~~^^
  File "/Users/phanley/Source/miningprof/main.py", line 106, in gather_miner_data
    data = await current_miner.get_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/phanley/Source/miningprof/.venv/lib/python3.13/site-packages/pyasic/miners/base.py", line 568, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
        allow_warning=allow_warning, include=include, exclude=exclude
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/Users/phanley/Source/miningprof/.venv/lib/python3.13/site-packages/pyasic/miners/base.py", line 489, in _get_data
    await asyncio.gather(rpc_command_task, web_command_task)
  File "/Users/phanley/Source/miningprof/.venv/lib/python3.13/site-packages/pyasic/web/avalonminer.py", line 83, in multicommand
    all_data = await asyncio.gather(*tasks)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/phanley/Source/miningprof/.venv/lib/python3.13/site-packages/pyasic/web/avalonminer.py", line 99, in _handle_multicommand
    return json.loads(raw_data)
           ~~~~~~~~~~^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.0_1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ~~~~~~~~~~~~~~~~~~~~~~~^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.0_1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/json/decoder.py", line 344, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.13/3.13.0_1/Frameworks/Python.framework/Versions/3.13/lib/python3.13/json/decoder.py", line 362, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

hanleyp, 2025-02-23:
This was working until I upgraded.

hanleyp, 2025-02-23:
I went backwards and it works in v0.70.1.

b-rowan, 2025-03-03:
Looks like it might be an issue with the new `get_mac` function that was implemented for the Nano3?  What firmware version are you using?

b-rowan, 2025-03-03:
I think 276a476fab1b4b8b6c0d6da236a07d080ff422cb should fix it

hanleyp, 2025-03-04:
Here are the contents of raw_data:
`<!DOCTYPE html>
<html lang="en">

<head>
	<meta charset="UTF-8" />
	<meta http-equiv="X-UA-Compatible" content="IE=edge,chrome=1" />
	<meta name="renderer" content="webkit" />
	<meta http-equiv="Cache-Control" content="no-cache, must-revalidate" />
	<meta http-equiv="pragma" content="no-cache" />
	<title>Avalon Device</title>
	<style type="text/css">
		* {
			margin: 0;
			padding: 0;
			font-family: "Helvetica Neue", Helvetica, Arial, sans-serif;
		}

		li {
			list-style-type: none;
		}

		body {
			margin: 0;
			background-color: rgba(0, 0, 0, 1
			background-image: none;
			position: relative;
			text-align: left;
		}

		.bg {
			background-position: center;
			background-size: cover;
			background-repeat: no-repeat;
		}

		.btn {
			background-position: center;
			background-size: contain;
			background-repeat: no-repeat;
			cursor: pointer;
		}

		.login {
			position: fixed;
			margin-left: -175px;
			top: 50%;
			margin-top: -210px;
			z-index: 1001;
			left: 50%;
			width: 350px;
			height: 420px;
			background-color: rgba(215, 215, 215, 1
			border: 0;
			border-radius: 10px;
			-moz-box-shadow: 0 0 10px rgba(0, 0, 0, 0.349019607843137
			-webkit-box-shadow: 0 0 10px rgba(0, 0, 0, 0.349019607843137
			box-shadow: 0 0 10px rgba(0, 0, 0, 0.349019607843137
			font-family: "微软雅黑 Bold", "微软雅黑 Regular", "微软雅黑";
			font-weight: 700;
			font-style: normal;
			font-size: 16px;
			color: #999;
			text-align: left;
			line-height: 20px;
		}

		.login .logo {
			margin-left: 112px;
			margin-top: 52px;
[... 110 lines cut ...]
			document.cookie = name + "=" + (value || "") + expires + "; path=/";
		}
		function checkLogin() {
			if (loginform.username.value == "") {
				alert("Username is null!"
				return false;
			}
			if (loginform.passwd.value == "") {
				alert("Password is null!"
				return false;
			}
			if (loginform.passwd.value.length > 16) {
				alert("Password too long!"
				return false;
			}
			var hashpasswd = sha256(loginform.passwd.value).substring(0, 24
			hashpasswd = "ff0000ff" + hashpasswd;
			setCookie("auth", hashpasswd, 1
			loginform.passwd.value = hashpasswd;
			loginform.submit(
		}
	</script>
</body>
<script src="sha256.min.js"></script>
</html>`

hanleyp, 2025-03-04:
Firmware Version is 24071801_42c628d and as far as I know it hasn't been updated in a while.