UpstreamData/pyasic #303: errors in formatting get_data as_influxdb

audibleroot, 2025-03-03:
**Describe the bug**
influxdb 2.4 and 2.7 give errors when data is written to it after formatting using .as_influxdb using pyasic v0.71.12
1) I think timestamp is incorrect, it gives 1.740961172e+18 but should give 1740960522000000000 
I inserted int, to make timestamp code as below and now it works
timestamp = str(int(self.timestamp * 1e9))
2) it passes the following to influxdb for board 2 and 3 but not 1,
hashboard_2_expected_chips=126,hashboard_3_hashrate=unit=TH/s rate=16.782, 
which gives an error because its not a proper field syntax
3) its doesnt include any pool info

**To Reproduce**
Steps to reproduce the behavior:
    all_miner_data = await asyncio.gather(*[miner.get_data() for miner in miners])
    all_miner_data = [d.as_influxdb(measurement_name='serverata') for d in all_miner_data]

write_client = InfluxDBClient(url=url, token=token, org=org)
write_api = write_client.write_api(write_options=SYNCHRONOUS)
for item in all_miner_data:
          write_api.write(bucket=bucket, org=org, record=item)

**Expected behavior**
data gets written without errors

audibleroot, 2025-03-03:
it gives an error, 
influxdb_client.rest.ApiException: (400)
Reason: Bad Request
and i think because there is a comma missing between fields for algo and uptime
firmware=BOS+,algo=SHA256 uptime=94240,

b-rowan, 2025-03-03:
I don't see how that could be the case, the line protocol reference shows the same thing, since algo is a tag - https://docs.influxdata.com/influxdb/v2/reference/syntax/line-protocol/

audibleroot, 2025-03-03:
yes you are right, if everything before the space is a tag then it should be fine, here is the full message:
HTTP response body: {"code":"invalid","message":"unable to parse 'data,ip=10.10.10.1,mac=01:B1:1B:B2:11:DB,make=AntMiner,model=S19j Pro,firmware=BOS+,algo=SHA256 uptime=94240,expected_hashrate=100.0,hashrate=54.47210713094742,hashboards.0.hashrate=19.591815187673962,hashboards.0.temp=43,hashboards.0.chip_temp=58,hashboards.0.chips=126,hashboards.0.expected_chips=126,hashboards.1.hashrate=16.656482993722054,hashboards.1.temp=45,hashboards.1.chip_temp=60,hashboards.1.chips=126,hashboards.1.expected_chips=126,hashboards.2.hashrate=18.2238089495514,hashboards.2.temp=47,hashboards.2.chip_temp=62,hashboards.2.chips=126,hashboards.2.expected_chips=126,temperature_avg=45,wattage=1547,wattage_limit=1650,fans.0.speed=194742,fans.1.speed=237884,fans.2.speed=4230,fans.3.speed=4260,expected_fans=4,total_chips=378,expected_chips=378,efficiency=28,is_mining=True,pools.0.url=\"xxx\",pools.0.accepted=21975,pools.0.rejected=43,pools.0.active=True,pools.0.alive=True,pools.0.user=\"d.d2\",pools.1.url=\"xxx\",pools.1.accepted=0,pools.1.rejected=0,pools.1.active=False,pools.1.alive=True,pools.1.user=\"d.d2\",pools.2.url=\"xxx\",pools.2.accepted=0,pools.2.rejected=0,pools.2.active=False,pools.2.alive=True,pools.2.user=\"d.d2\" 1741036415000000000': invalid boolean"}

b-rowan, 2025-03-03:
Swapped boolean format to lower case, try the latest commit on that brinch, 09bc530335db68d529486cf30a2adf1e05d709b3

audibleroot, 2025-03-04:
again error is invalid boolean
booleans are lower case now though, eg. pools.1.active=false,pools.1.alive=true

but from my prev experience even if it complains about booleans it could be something else, for example when I had this line hashboard_3_hashrate=unit=TH/s rate=16.78208338483838,hashboard_3_temperature=43, which is missing a comma and has = sign which could be confusing it, it was also complaining about booleans.

maybe it doesnt like . in the field names? or maybe for pool urls and pool user it uses escape characters like this, sorry it wasnt clear up above
pools.2.url=\"xxx\",pools.2.accepted=0,pools.2.rejected=0,pools.2.active=false,pools.2.alive=true,pools.2.user=\"d.d2\"

audibleroot, 2025-03-05:
sorry i copy pasted wrong thing, I think it still gives an error because \" is not supported by influxdb
pools.1.user=\"d.d2\",pools.2.url=\"stratum+tcp://xx\",pools.2.accepted=0,pools.2.rejected=0,pools.2.active=false,pools.2.alivettrue,pools.2.user=\"d.d2\"

audibleroot, 2025-03-05:
oh i did copy and paste the right thing, but it gets removed 
basically this `pools.1.user=\"d.d2\",pools.2.url=\"stratum+tcp://xx\",pools.2.accepted=0,pools.2.rejected=0,pools.2.active=false,pools.2.alive=true,pools.2.user=\"d.d2\"`

b-rowan, 2025-03-05:
Quote from the docs -

"Always double quote string field values."

https://docs.influxdata.com/influxdb/v2/reference/syntax/line-protocol/

audibleroot, 2025-03-05:
yes but I think its supposed to be like this, without \

`pools.1.user="d.d2",pools.2.url="stratum+tcp://xx"

audibleroot, 2025-03-05:
this is how .as_influx formats data in 0.42 which works

b-rowan, 2025-03-05:
Give it another try on latest.  Tried formatting " " with "\ ".

audibleroot, 2025-03-05:
that works now. in the past you used underscores for field names, eg hashboard_3_chips. Will you keep hashboard.3.chips consistently into the future, or can be change back to hashboard_3_chips?

audibleroot, 2025-03-05:
a number of fields changed, before used to be hashboard, now hashboards, before fan_1 now fan.1.speed. I dont really mind, I ll just change but wanted to point out

b-rowan, 2025-03-05:
I think I can allow a user to pass that in.  Just a sec.

b-rowan, 2025-03-05:
Pass `level_delimiter="_"` to `as_influx` and you should be able to do it now.

audibleroot, 2025-03-05:
yep good idea. i tried it and getting this error
level_delimiter.join([key, i]),
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: sequence item 1: expected str instance, int found

audibleroot, 2025-03-05:
yep that fixed it

audibleroot, 2025-03-06:
for some reason, for a couple of S19 that are not mining, get_miner would return two commas:
`is_mining=false,,pools_0_url="stratum2+tcp://d2"`
then as_influxdb would keep those two commas, and also for some reason puts a \ into the fields, and one of these or both causes an error writing to influxdb
'HTTP response body: {"code":"invalid","message":"unable to parse 'data,ip=10.16.11.2,mac=82:30:3F:F7:71:F7,make=AntMiner,model=S19,firmware=BOS+,algo=SHA256 uptime=406473,expected_hashrate=95.0,hashrate=0.0,hashboards_0_temp=0,hashboards_0_chip_temp=15,hashboards_0_expected_chips=76,hashboards_1_temp=0,hashboards_1_chip_temp=15,hashboards_1_expected_chips=76,hashboards_2_temp=0,hashboards_2_chip_temp=15,hashboards_2_expected_chips=76,temperature_avg=0,wattage=200,wattage_limit=1800,fans_0_speed=1680,fans_1_speed=1680,fans_2_speed=720,expected_fans=4,expected_chips=228,efficiency=0,is_mining=false,,pools_0_url=\"d2\",pools_0_accepted=0,pools_0_rejected=0,pools_0_active=true,pools_0_alive=true,pools_0_user=\"d2\",pools_1_url=\"stratum+tcp://d2\",pools_1_accepted=0,pools_1_rejected=0,pools_1_active=false,pools_1_alive=true,pools_1_user=\"d2\",pools_2_url=\"stratum+tcp://d2\",pools_2_accepted=0,pools_2_rejected=0,pools_2_active=false,pools_2_alive=true,pools_2_user=\"d2\" 1741283367000000000': invalid field format"}'

b-rowan, 2025-03-06:
Seems like something with errors maybe?  Does `data.errors` contain anything?

audibleroot, 2025-03-06:
bos log shows this errors
```
01/03/2025 16:55:46 | Sensor hb1.73 probing failed: Unexpected response for temperature request: RxFrame { command: 3b, status: 0, payload: [] }
-- | --
  | 01/03/2025 16:55:47 | Sensor hb2.73 probing failed: Unexpected response for temperature request: RxFrame { command: 3b, status: 0, payload: [] }
``` 
and i ll try to see whats in data.errors

b-rowan, 2025-03-06:
I pushed a fix for the errors stuff, im pretty sure thats what was causing it.

audibleroot, 2025-03-06:
thank you that worked

audibleroot, 2025-03-08:
can ''fault_light': False,' also be one of the fields that as_influxdb handles?

audibleroot, 2025-03-08:
also percentage_expected field and values for hashrate, etc

audibleroot, 2025-03-09:
wattage, wattage_limit

b-rowan, 2025-03-10:
> also percentage_expected field and values for hashrate, etc

Shouldn't need this, you should be able to compute it directly in influxdb, just `hashrate/expected_hashrate`.

> can 'fault_light': False,' also be one of the fields that as_influxdb handles?

Yep, can add this.

audibleroot, 2025-03-11:
and is expected_hashrate what used to be called nominal hashrate?

PR #306 (2025-03-17): feature: update support for influxdb line protocol
InfluxDB line protocol support has been out of date for a while now, this should fix it.

Fixes: #303 

@audibleroot please test and see if this works for your case.
closed 2025-03-17 by b-rowan