UpstreamData/pyasic #305: Icereaver AL3 - UserWarning: Partially supported miner found: 10306, type: MinerTypes.ICERIVER

Combinator78, 2025-03-03:
C:\vekus\agent\local_libs\pyasic\miners\factory.py:1092: UserWarning: Partially supported miner found: 10306, type: MinerTypes.ICERIVER, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(
ip='172.16.26.39' device_info=DeviceInfo(make=<MinerMake.ICERIVER: 'IceRiver'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.base.GenericAlgo'>) mac='00:0A:71:10:09:C0' api_ver=None fw_ver=None hostname='ICERIVER' sticker_hashrate=None expected_hashrate=None expected_hashboards=None expected_chips=0 expected_fans=None env_temp=None wattage=None voltage=None fans=[Fan(speed=2682), Fan(speed=2682), Fan(speed=2682), Fan(speed=2682)] fan_psu=None hashboards=[] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://hash.viabtc.io:3016', user='ggringo14.alpha01', password='x'), Pool(url='stratum+tcp://hash.viabtc.io:3016', user='ggringo14.alpha01', password='x'), Pool(url='stratum+tcp://hash.viabtc.io:3016', user='ggringo14.alpha01', password='x')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=0 errors=[] is_mining=True uptime=280405 pools=[] hashrate=14.784 TH/s wattage_limit=None total_chips=None nominal=None percent_expected_chips=None percent_expected_hashrate=None percent_expected_wattage=None temperature_avg=None efficiency=None datetime='2025-03-03T15:41:10.530553+03:00' timestamp=1741005670 make='IceRiver' model=None firmware='Stock' algo='Generic (Unknown)'
unit=TH/s rate=14.784
PS C:\vekus\agent>

Combinator78, 2025-03-03:
There is no model name anywhere in the web interface, but it is an **iceriver AL3** model.
[image]

b-rowan, 2025-07-21:
Coming in next version release