UpstreamData/pyasic #311: Partially supported miner found: DG-Home1

MattD86, 2025-03-15:
can we get DG-Home1 added - DEVS if you need me to help let me know ?

Here is the errors:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/pyasic/miners/base.py", line 525, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pyasic/miners/backends/elphapex.py", line 214, in _get_hashboards
    for idx in range(self.expected_hashboards)
               ~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object cannot be interpreted as an integer

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/miner/coordinator.py", line 97, in _async_update_data
    miner_data = await self.miner.get_data(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<13 lines>...
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pyasic/miners/base.py", line 568, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
        allow_warning=allow_warning, include=include, exclude=exclude
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/pyasic/miners/base.py", line 527, in _get_data
    raise APIError(
        f"Failed to call {data_name} on {self} while getting data."
    ) from e
pyasic.errors.APIError: Failed to call hashboards on Elphapex (Stock): 192.168.12.111 while getting data.
2025-03-15 11:59:18.982 DEBUG (MainThread) [custom_components.miner.coordinator] Finished fetching DG-Home1_1 data in 0.595 seconds (success: False)

b-rowan, 2025-03-17:
Hey, can you try `await miner.rpc.devdetails()` and `await miner.rpc.version()`?  Just need to check the device type that is shown in the API then i can add it.

MattD86, 2025-03-18:
Hi, How do I get this information from home assistant or from API, I can run powershell on windows or command line, or is their something I can do in home assistant?

Keen to help, but not sure how to access this api information @b-rowan let me know what I need to do to get this information

purcell-lab, 2025-07-06:
http://192.168.86.83/cgi-bin/stats.cgi

```json
{
  "STATUS": {
    "STATUS": "S",
    "when": 120235,
    "timestamp": 1751782188,
    "api_version": "1.0.0",
    "Msg": "stats"
  },
  "INFO": {
    "miner_version": "DG-Home1_V1.0.5",
    "CompileTime": "",
    "dev_sn": "15HY251114N301525C12B",
    "type": "DG-Home1",
    "hw_version": "DG-Home1_HW_V1.0"
  },
  "STATS": [
    {
      "rate_unit": "MH/s",
      "elapsed": 119914,
      "rate_30m": 0,
      "rate_5s": 1459.78,
      "hwp_total": 0.11550000000000001,
      "rate_ideal": 2160,
      "chain": [
        {
          "freq_avg": 0,
          "index": 0,
          "sn": "",
          "temp_chip": [
            "",
            "",
            "",
            ""
          ],
          "rate_15m": 0,
          "hw": 0,
          "temp_pcb": [
            47,
            46,
            67,
            66
          ],
          "failrate": 0,
          "asic": "ooooooooo oooooooo oooooooo oooooooo oooo",
          "rate_real": 0,
          "asic_num": 0,
          "temp_pic": [
            47,
            46,
            67,
            66
          ],
          "rate_ideal": 0,
          "hashrate": 0
        },
        {
          "freq_avg": 0,
          "index": 1,
          "sn": "",
          "temp_chip": [
            "",
            "",
            "",
            ""
          ],
          "rate_15m": 0,
          "hw": 0,
          "temp_pcb": [
            47,
            46,
            67,
            66
          ],
          "failrate": 0,
          "asic": "ooooooooo oooooooo oooooooo oooooooo oooo",
          "rate_real": 0,
          "asic_num": 0,
          "temp_pic": [
            47,
            46,
            67,
            66
          ],
          "rate_ideal": 0,
          "hashrate": 0
        },
        {
          "freq_avg": 0,
          "index": 2,
          "sn": "",
          "temp_chip": [
            "",
            "",
            "",
            ""
          ],
          "rate_15m": 0,
          "hw": 0,
          "temp_pcb": [
            47,
            46,
            67,
            66
          ],
          "failrate": 0,
          "asic": "ooooooooo oooooooo oooooooo oooooooo oooo",
          "rate_real": 0,
          "asic_num": 0,
          "temp_pic": [
            47,
            46,
            67,
            66
          ],
          "rate_ideal": 0,
          "hashrate": 0
        },
        {
          "freq_avg": 64000,
          "index": 3,
          "sn": "11HY251116N300885H11LD22",
          "temp_chip": [
            "50812",
            "49125",
            "46812",
            "50750"
          ],
          "eeprom_loaded": true,
          "rate_15m": 2066,
          "hw": 120,
          "temp_pcb": [
            47,
            46,
            67,
            66
          ],
          "failrate": 0.050000000000000003,
          "asic": "ooooooooo oooooooo oooooooo oooooooo oooo",
          "rate_real": 2152.98,
          "asic_num": 120,
          "temp_pic": [
            47,
            46,
            67,
            66
          ],
          "rate_ideal": 2160,
          "hashrate": 1827.6500000000001
        }
      ],
      "rate_15m": 2066,
      "chain_num": 4,
      "fan": [
        "1260",
        "1200",
        "1200",
        "3300"
      ],
      "rate_avg": 2152.98,
      "fan_num": 4
    }
  ]
}
```

b-rowan, 2025-07-21:
Coming in next version release

closed 2025-07-21 by b-rowan

purcell-lab, 2025-10-18:
DG Home 1 partially supported. with 0.77

Please reopen issue.

```logs
$ ./venv/bin/python3 ./get-miners.py 
None None
/home/mark/venv/lib/python3.11/site-packages/pyasic/miners/factory.py:1173: UserWarning: Partially supported miner found: DG-Home1, type: MinerTypes.ELPHAPEX, please open an issue with miner data and this model on GitHub (https://github.com/UpstreamData/pyasic/issues).
  warnings.warn(
[Elphapex (Stock): 192.168.86.83]
```

commit 77f59f6d (2025-07-21): feature: add support for Elphapex DG1 Home

Fixes: #311