UpstreamData/pyasic #314: WhatsMiner M50_VH75 Model, hashrate and other data are not determined

Combinator78, 2025-03-19:
105 chips
ip='172.16.26.65' device_info=DeviceInfo(make=<MinerMake.WHATSMINER: 'WhatsMiner'>, model=None, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.base.GenericAlgo'>) mac='CA:0A:18:00:01:C7' api_ver=None fw_ver='20250120.16.Rel' hostname='WhatsMiner' sticker_hashrate=None expected_hashrate=None expected_hashboards=None expected_chips=0 expected_fans=None env_temp=None wattage=None voltage=None fans=[] fan_psu=7504 hashboards=[] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://gate.emcd.network:3333', user='volkavto1.Z16', password='x'), Pool(url='stratum+tcp://eu.emcd.network:3333', user='volkavto1.Z16', password='x'), Pool(url='stratum+tcp://kz.emcd.network:3333', user='volkavto1.Z16', password='x')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=False errors=[] is_mining=True uptime=369328 pools=[PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='gate.emcd.network', port=3333, pubkey=None), accepted=20875, rejected=40, get_failures=3, remote_failures=0, active=True, alive=True, index=1, user='volkavto1.Z16', pool_rejected_percent=0.19125029882859193, pool_stale_percent=0.014343772412144393), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='eu.emcd.network', port=3333, pubkey=None), accepted=24, rejected=1, get_failures=0, remote_failures=0, active=False, alive=True, index=2, user='volkavto1.Z16', pool_rejected_percent=4.0, pool_stale_percent=0.0), PoolMetrics(url=PoolUrl(scheme=<Scheme.STRATUM_V1: 'stratum+tcp'>, host='kz.emcd.network', port=3333, pubkey=None), accepted=0, rejected=0, get_failures=0, remote_failures=0, active=False, alive=True, index=3, user='volkavto1.Z16', pool_rejected_percent=0, pool_stale_percent=0)] hashrate=None wattage_limit=None total_chips=None nominal=None percent_expected_chips=None percent_expected_hashrate=None percent_expected_wattage=None temperature_avg=None efficiency=None datetime='2025-03-19T18:00:17.873305+03:00' timestamp=1742396417 make='WhatsMiner' model=None firmware='Stock' algo='Generic (Unknown)'
None
[image]

b-rowan, 2025-03-19:
Miner is supported, but it might be an issue with their API or something?

Combinator78, 2025-03-19:
The file does not contain a description of the VH75 model
pyasic/miners/whatsminer/btminer/M5X/M50.py

b-rowan, 2025-03-19:
The last digit is ignored, so it is an M50VH70.

eandersson, 2025-03-20:
It looks like their V2 API is broken in the latest firmware, `devdetails` returns.
> {'STATUS': [{'STATUS': 'E', 'Msg': 'Invalid command'}], 'id': 1}

I suspect we should be migrating to their V3 API.

b-rowan, 2025-03-20:
I'm pretty hesitant to just migrate everything to the V3 API, because backwards compatibility would be really nice.  I might do some more complaining about it to their devs...

eandersson, 2025-03-20:
> I'm pretty hesitant to just migrate everything to the V3 API, because backwards compatibility would be really nice. I might do some more complaining about it to their devs...

Yea - I mean right now it looks like their API is just broken.

b-rowan, 2025-03-27:
Oh good.  Thanks.  I'll look into getting V3 supplier right after I yell at their dev team...

eandersson, 2025-03-28:
> Oh good. Thanks. I'll look into getting V3 support right after I yell at their dev team...

We could always try V2 first, and then fallback to V3 if you want to keep compatibility with older firmware's.

Combinator78, 2025-08-16:
ip='10.1.23.11' device_info=DeviceInfo(make=<MinerMake.WHATSMINER: 'WhatsMiner'>, model=<WhatsminerModels.M61VK10: 'M61 VK10'>, firmware=<MinerFirmware.STOCK: 'Stock'>, algo=<class 'pyasic.device.algorithm.sha256.SHA256Algo'>) mac='CE:0B:0C:00:48:D8' api_ver='3.0.1' fw_ver='20250120.10.REL' hostname='WhatsMiner' sticker_hashrate=None **expected_hashrate=-0.003** expected_hashboards=3 expected_chips=540 expected_fans=2 env_temp=25.3 wattage=4120 voltage=None fans=[Fan(speed=4252), Fan(speed=4252)] fan_psu=[Fan(speed=7024)] hashboards=[HashBoard(slot=0, hashrate=66.888 TH/s, inlet_temp=56.5, outlet_temp=86.5, temp=56.5, chip_temp=None, chips=180, expected_chips=180, serial_number=None, missing=False, tuned=True, active=True, voltage=None), HashBoard(slot=1, hashrate=66.777 TH/s, inlet_temp=56.4, outlet_temp=84.9, temp=56.4, chip_temp=None, chips=180, expected_chips=180, serial_number=None, missing=False, tuned=True, active=True, voltage=None), HashBoard(slot=2, hashrate=70.862 TH/s, inlet_temp=62.1, outlet_temp=96.1, temp=62.1, chip_temp=None, chips=180, expected_chips=180, serial_number=None, missing=False, tuned=True, active=True, voltage=None)] config=MinerConfig(pools=PoolConfig(groups=[PoolGroup(pools=[Pool(url='stratum+tcp://btc.ru.viabtc.me:3333', user='Sasha0109.006', password='x'), Pool(url='stratum+tcp://btc.ru.powhashing.com:3333', user='Sasha0109.006', password='x'), Pool(url='stratum+tcp://ru.btc.viabtc.io:443', user='Sasha0109.006', password='x')], quota=1, name=None)]), fan_mode=FanModeNormal(mode='normal', minimum_fans=1, minimum_speed=0), temperature=TemperatureConfig(target=None, hot=None, danger=None), mining_mode=MiningModeNormal(mode='normal')) fault_light=False errors=[] is_mining=True uptime=92104 pools=[] hashrate=204.52700000000002 TH/s wattage_limit=None total_chips=540 nominal=True percent_expected_chips=100 percent_expected_hashrate=-6817567 percent_expected_wattage=None temperature_avg=58 efficiency=20 efficiency_fract=20.14 datetime='2025-08-16T15:59:43.521056+03:00' timestamp=1755349183 make='WhatsMiner' model='M61 VK10' firmware='Stock' algo='SHA256'
unit=TH/s rate=204.52700000000002

Everything works, except for the incorrect expected_hashrate=-0.003 parameter

And the scanning is a little slower. It is treated by changing the "factory_get_timeout":6 parameter in the pyasic.settings module

Combinator78, 2025-08-16:
stderr IN executePythonScan: C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py:1771 in default_exception_handler
[ERROR][08/16/25 16:15:56](asyncio) - Task was destroyed but it is pending!
task: <Task pending name='Task-50210' coro=<BTMinerV3RPCAPI._read_loop() running at C:\Vekus\agent\local_libs\pyasic\rpc\btminer.py:1169> wait_for=<Future pending cb=[Task.task_wakeup()]>>

stderr IN executePythonScan: C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py:1771 in default_exception_handler
[ERROR][08/16/25 16:15:56](asyncio) - Task was destroyed but it is pending!
task: <Task pending name='Task-50212' coro=<BTMinerV3RPCAPI._read_loop() running at C:\Vekus\agent\local_libs\pyasic\rpc\btminer.py:1169> wait_for=<Future pending cb=[Task.task_wakeup()]>>
C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py:1771 in default_exception_handler
[ERROR][08/16/25 16:15:56](asyncio) - Task was destroyed but it is pending!
task: <Task pending name='Task-50216' coro=<BTMinerV3RPCAPI._read_loop() running at C:\Vekus\agent\local_libs\pyasic\rpc\btminer.py:1169> wait_for=<Future pending cb=[Task.task_wakeup()]>>
C:\Users\User\AppData\Local\Programs\Python\Python311\Lib\asyncio\base_events.py:1771 in default_exception_handler
[ERROR][08/16/25 16:15:56](asyncio) - Task was destroyed but it is pending!
task: <Task pending name='Task-50217' coro=<BTMinerV3RPCAPI._read_loop() running at C:\Vekus\agent\local_libs\pyasic\rpc\btminer.py:1169> wait_for=<Future pending cb=[Task.task_wakeup()]>>

b-rowan, 2025-08-16:
> Everything works, except for the incorrect expected_hashrate=-0.003 parameter

I've noticed this recently even on the ones I've tested on, Ill replace it with None in this case I think.

closed 2025-08-16 by b-rowan