UpstreamData/pyasic #320: Data gathering from Braiins Miner S19 failed (when using custom password)

Antonmuellerp2b, 2025-03-24:
### Bug Report: Custom Passwords Not Working with pyasic (Braiins Firmware on Antminer S19)

**Description**
When trying to retrieve miner data using `pyasic` on an Antminer S19 running Braiins Firmware, authentication fails unless the default password "root" is set on the device/miner. Setting a custom password on the miner and updating `settings` in the python script has no effect, resulting in an `UNAUTHENTICATED` gRPC error.

To be clear: Changing password in the script via settings.update does not change the behaviour at all. 
Unless the default password "root" is set on the miner itself the same error occurs no matter if settings.update was used or not. 

**To Reproduce**
Steps to reproduce the behavior:
1. Set a custom password on the miner (e.g., `mypass`).
2. Run the following script:

```python
import asyncio
from pyasic import get_miner, settings

print(settings.get("default_bosminer_web_password"))
print(settings.get("default_bosminer_ssh_password"))

settings.update("default_bosminer_web_password", "mypass")
settings.update("default_bosminer_ssh_password", "mypass")

print(settings.get("default_bosminer_web_password"))
print(settings.get("default_bosminer_ssh_password"))

async def gather_miner_data():
    miner = await get_miner("172.20.1.93")
    if miner is not None:
        miner_data = await miner.get_data()
        print(miner_data)
        print(miner_data.hashrate)  # hashrate of the miner in TH/s

if __name__ == "__main__":
    asyncio.run(gather_miner_data())
```

**Expected Behavior**
The script should successfully retrieve miner data even with a custom password.

**Actual Behavior**
The script fails with the following error:

```
root
root
mypass
mypass
Traceback (most recent call last):
  File "/root/venv_pyasic/lib/python3.11/site-packages/pyasic/miners/base.py", line 525, in _get_data
    miner_data[data_name] = await function(**args_to_send)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/venv_pyasic/lib/python3.11/site-packages/pyasic/miners/backends/braiins_os.py", line 835, in _get_mac
    grpc_miner_details = await self.web.get_miner_details()
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/venv_pyasic/lib/python3.11/site-packages/pyasic/web/braiins_os/boser.py", line 443, in get_miner_details
    return await self.send_command(
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/venv_pyasic/lib/python3.11/site-packages/pyasic/web/braiins_os/boser.py", line 108, in send_command
    metadata.append(("authorization", await self.auth()))
                                      ^^^^^^^^^^^^^^^^^
  File "/root/venv_pyasic/lib/python3.11/site-packages/pyasic/web/braiins_os/boser.py", line 132, in auth
    await self._get_auth()
  File "/root/venv_pyasic/lib/python3.11/site-packages/pyasic/web/braiins_os/boser.py", line 145, in _get_auth
    await stream.recv_initial_metadata()
  File "/root/venv_pyasic/lib/python3.11/site-packages/grpclib/client.py", line 393, in recv_initial_metadata
    self._raise_for_grpc_status(status, message, details)
  File "/root/venv_pyasic/lib/python3.11/site-packages/grpclib/client.py", line 345, in _raise_for_grpc_status
    raise GRPCError(status, message, details)
grpclib.exceptions.GRPCError: (<Status.UNAUTHENTICATED: 16>, 'Bad credentials', None)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/root/read_1_miner.py", line 23, in <module>
    asyncio.run(gather_miner_data())
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/root/read_1_miner.py", line 18, in gather_miner_data
    miner_data = await miner.get_data()
                 ^^^^^^^^^^^^^^^^^^^^^^
  File "/root/venv_pyasic/lib/python3.11/site-packages/pyasic/miners/base.py", line 568, in get_data
    gathered_data = await self._get_data(
                    ^^^^^^^^^^^^^^^^^^^^^
  File "/root/venv_pyasic/lib/python3.11/site-packages/pyasic/miners/base.py", line 527, in _get_data
    raise APIError(
pyasic.errors.APIError: Failed to call mac on S19 (BOS+): 172.20.1.93 while getting data.
```

**Environment**
- OS: Debian
- Python version: [e.g., 3.11]
- pyasic version: [e.g., latest from pip]
- Miner: Antminer S19
- Miner Firmware: Braiins OS+ (Version: 2025-02-21-0-e05df053-25.01-plus)

**Additional Information**
- The issue persists regardless of whether `settings.update` is used.

Any insights or fixes would be appreciated!

b-rowan, 2025-03-24:
Definitely a hug, but curious if it's a bug with the settings handling or with the password handling.  Do you want to try setting `miner.web.pwd` to your custom password and seeing if that helps?

Antonmuellerp2b, 2025-03-27:
Hey, thanks for your reply,

> Do you want to try setting `miner.web.pwd` to your custom password and seeing if that helps?

how can I do this? I only get this error: 

`AttributeError: 'NoneType' object has no attribute 'web'.
`

I tried it like this: 

```
async def gather_miner_data():
    miner = await get_miner("172.20.1.93")
    miner.web.pwd="mypass"
    if miner is not None:
        miner_data = await miner.get_data()
        print(miner_data)
        print(miner_data.hashrate)  # hashrate of the miner in TH/s

```

b-rowan, 2025-03-27:
If the miner is None, then there is no miner being identified there, maybe it isn't responding at all or it's off.

Antonmuellerp2b, 2025-04-11:
I still don't know what else to try. Do you have any further ideas? I cannot leave the miner on root:root forever :/

jakubprogramming, 2025-04-14:
I have had the same exact experience. On Braiins firmware pyasic only works when default password is used. 
It's a shame we would love to use it for our larger installations, but obviously keeping miners running with default passwords is not safe in production environments.