UpstreamData/pyasic #325: whatsminer pool settings do not change

Combinator78, 2025-04-01:
On Whatsminer devices, changing the worker and pool does not work until you enter the web interface.
ОШИБКА stderr: C:\vekus\agent\./local_libs\pyasic\rpc\base.py:228 in _send_bytes
[WARNING][04/01/25 14:14:31](root) - BTMinerRPCAPI: 172.16.24.45 - ([Hidden] Send Bytes) - Read timeout expired.
C:\vekus\agent\./local_libs\pyasic\rpc\base.py:228 in _send_bytes
[WARNING][04/01/25 14:14:34](root) - BTMinerRPCAPI: 172.16.24.45 - ([Hidden] Send Bytes) - Read timeout expired.

After you enter the web interface of the device in the browser and then send the command to change the worker and pool again, the data is updated. But for the changes to take effect, you need to reboot the device.

b-rowan, 2025-04-01:
Sorry, a bit confused here.  You're saying you have to open the web interface of the miner in order to have the pools be updated?

b-rowan, 2025-08-16:
Should be fixed in 0.76.0 I think?