UpstreamData/pyasic #328: BOS miners are not accurately reporting "is_mining"

womj, 2025-04-09:
**Describe the bug**
BOS miners are not accurately reporting "is_mining"

**To Reproduce**
Steps to reproduce the behavior:
1. Get data on a bos miner that is stopped or started

**Expected behavior**
stopped miners should report is_mining false

**Desktop (please complete the following information):**
 - POPOS 22.04

**Miner Information (If applicable):**
 - Any BOS+ v 2024-12-03-0-f4c282cd-24.09.1-plus
**Additional context**
 narrowed the issue down to braiins_os.py line 1080

```
async def _is_mining(self, rpc_devdetails: dict = None) -> Optional[bool]:
        print("is mining", rpc_devdetails)
        if rpc_devdetails is None:
            try:
                rpc_devdetails = await self.rpc.send_command(
                    "devdetails", ignore_errors=True, allow_warning=False
                )
            except APIError:
                pass

        if rpc_devdetails is not None:
            try:
                return not rpc_devdetails["STATUS"][0]["Msg"] == "Unavailable"
            except LookupError:
                pass
```

rpc_devdetails returning as
```
Running

{'STATUS': [{'STATUS': 'S', 'When': 1744220558, 'Code': 69, 'Msg': 'Device Details', 'Description': 'BOSer boser-buildroot 0.1.0-e05df053'}], 'DEVDETAILS': [{'Chips': 108, 'Cores': 137808, 'DEVDETAILS': 0, 'Device Path': '', 'Driver': '', 'Frequency': 521, 'ID': 1, 'Kernel': '', 'Model': 'Antminer T21', 'Name': 'HashchainManager 1', 'Voltage': 14.62646722798786}, {'Chips': 108, 'Cores': 137808, 'DEVDETAILS': 1, 'Device Path': '', 'Driver': '', 'Frequency': 509, 'ID': 2, 'Kernel': '', 'Model': 'Antminer T21', 'Name': 'HashchainManager 2', 'Voltage': 14.62646722798786}, {'Chips': 108, 'Cores': 137808, 'DEVDETAILS': 2, 'Device Path': '', 'Driver': '', 'Frequency': 533, 'ID': 3, 'Kernel': '', 'Model': 'Antminer T21', 'Name': 'HashchainManager 3', 'Voltage': 14.62646722798786}], 'id': 1}

Stopped

is mining None

```

b-rowan, 2025-04-16:
Good to know.  Have to find  anew place to source that.  This AFAIK this was copied from the old bosminer implementation, but it looks like they fixed that...

b-rowan, 2025-12-12:
@markoceri 

I'm not particular either way, although it may be a good idea to check if the power is above a couple hundred watts, since there may be cases where it reads from the PSU, and the fans draw power even when off.