UpstreamData/pyasic #334: Support fractional (float) values for the efficiency parameter

SKART1, 2025-04-17:
**Is your feature request related to a problem? Please describe.**
Yes. The efficiency computed field in pyasic/data/__init__.py currently does this:
```
return round(self.wattage / float(self.hashrate))
```
Since round(x) with a single argument returns an int. This prevents accurate modeling of miner efficiency (for example, a true 0.80 J/TH is rounded up to 1, and 0.49 J/TH is rounded down to 0).

**Describe the solution you'd like**
Change the return type of efficiency from int to float | None.

Compute and return the ratio with sensible precision, for example:

```
return round(self.wattage / float(self.hashrate), 2)
```
so values like 0.80 and 1.25 are preserved.

Update the @computed_field decorator/type hints to reflect float.

**Describe alternatives you've considered**
Add second efficiency_fractional field to represent efficiency with extra precision

**Additional context**
Code lives here: pyasic/data/__init__.py

Downstream Home‑Assistant integration will automatically pick up the new float values.

SKART1, 2025-04-18:
#335

closed 2025-04-18 by b-rowan

SKART1, 2025-04-19:
Works like a charm now
[image]

@b-rowan  if possible - could you please merge also a fix? Sorry for forgetting retrun statement - python is not my "native" language

b-rowan, 2025-04-19:
I guess I didn't review it well enough haha.

commit ec88fbf6 (2025-04-18): Fix #334 efficiency calculation and add computed fields